Klipper3d/klipper #3467: BLTOUCH on Y Axis

Rizzist, 2020-10-24:
Hello! I have a homemade belt printer that swaps Z and Y axis, and tilts the new Y axis at an angle, like the Blackbelt, White Knight and CR30 3D printers. Due to the nature of the slicer, I am unable to swap Y and Z axis. Is it possible to use BLTOUCH on Y axis with Klipper? It is currently not possible with Marlin, Repetier or other firmwares. I don't know if its possible on Duet or Klipper, which is why I am asking here. Thank you!

Rizzist, 2020-10-24:
[klippy.log.zip](https://github.com/KevinOConnor/klipper/files/5433921/klippy.log.zip)
This is just a random klipper log from another issue. Hoping for a quick response and on where to look next if problem is not solved here. Thank you!

KevinOConnor, 2020-10-30:
Klipper only supports using probes when homing the Z axis.  I'm sure it's possible to implement, but an interested developer would need to take that on and submit the code.

-Kevin

closed 2020-11-28 by github-actions[bot]