Klipper3d/klipper #3468: Feature Request: Sort vsdlist by creation date

dushyantahuja, 2020-10-25:
Hi, I've seen that this was rejected - https://github.com/KevinOConnor/klipper/pull/1387/commits/c7cf8c5095a8bb744a0223eb299400efeaa40d1f
However, is it possible to relook at this - I have a lot of gcode files, and need to scroll down on the LCD menu to find the latest one that has been uploaded. Sorting by creation date would be really helpful, and make the menu more usable.

dushyantahuja, 2020-11-03:
@KevinOConnor Please let me know if this can be included - adds a slight complexity to the config file (only for those that want it), but would really help out on small LCDs.

KevinOConnor, 2020-11-04:
I have no preference here.  If someone wants to implement it, it's fine with me.  I'd say it should be done in the display code as part of specifying the menu items.

-Kevin

closed 2020-12-03 by github-actions[bot]