Klipper3d/klipper #3469: Issues communicating with ADXL345

opensource-alt, 2020-10-25:
Follow up of a discussion I started probaply at the wrong place...
https://github.com/KevinOConnor/klipper/issues/3027#issuecomment-716072466

I have (seems multiple) issues to have a working ADXL345 setup with klipper.
* maybe some 32 / 64 bits issues seen on Orange PI PC2
* maybe something related to the spi interface on RPi.

Beware: Orange PI PC 2 is GPIO pin compatible with a RPi *but* the wirering is spi1.0 and not the expected spi0.0..

klippy.log related files to be uploaded later today.

> @opensource-alt So, the same (32-bit) installation works on RPi 2, but does not work on RPi 3? And didn't work on 64 bit installation on the Orange Pi either? Well, @KevinOConnor wrote the low-level communication interface to ADXL345 (as well as Klipper SPI communication). In general, we'd need the full klippy.log of the failed attempt (and the result from the Octoprint console).
Yes: the same wires, same sdcard gives a working ADXL345 on my RPi 2 and a non working one on a RPi 3. The error is that the expected id of 0xe5 is not received. Most of the time it's 0 but not always.
Orange PI PC 2: it's another issue now: ioctl error. The "bad adress" suggestst something related to a memoery segment outside of the process adress space. I'm looking in the source code to see if I can find something. I found a leak in the file descriptor cache management in the daemon process. I'll open another issue with a suggested fix for that.

> 
> As a separate wild guess, you could check that your cable is not too long. I had problems when I made too long cable myself (granted, there could've been other issues with that cable of mine). But I do suspect that long cable can get noticeably high inductivity and capacitance, potentially influencing the step signal shape, and also becoming more susceptible to picking up noise.
I thought about this and checked with other shorter cables but in this case why it's working with my RPi 2? 

Philippe

opensource-alt, 2020-10-25:
Obtained on Orange PI PC 2 (latest 64 bits ubuntu): [klippy.log](https://github.com/KevinOConnor/klipper/files/5434654/klippy.log)

dmbutyugin, 2020-10-25:
A few notes / suggestions

> Beware: Orange PI PC 2 is GPIO pin compatible with a RPi _but_ the wirering is spi1.0 and not the expected spi0.0..
> ....
> Orange PI PC 2: it's another issue now: ioctl error. The "bad adress" suggestst something related to a memoery segment outside of the process adress space. I'm looking in the source code to see if I can find something. I found a leak in the file descriptor cache management in the daemon process. I'll open another issue with a suggested fix for that.

For Orange PI PC 2, can you sync everything to the latest HEAD to include ed5ce9cb376cc40f0a4503a6d01ef85b21293424 commit? What I mean is, sync to HEAD and rebuild all mcus as well. Right now you have some modified code for the mcus, so could you make sure that the mcu code is unmodified?

The latest error is 'Unable to issue spi ioctl', which is initiated at src/linux/spidev.c:106. So it seems that if the descriptors cache is fixed, the code then cannot perform the spi operation. You can try to check what the errno is (it seems it is not logged in klippy.log).

> Yes: the same wires, same sdcard gives a working ADXL345 on my RPi 2 and a non working one on a RPi 3. The error is that the expected id of 0xe5 is not received. Most of the time it's 0 but not always.

We would need the exact error message, and the exact klippy.log of that attempt on RPi 3. Especially interesting is if the received output isn't 0. You can run the query command several times and attach the log.

> I thought about this and checked with other shorter cables but in this case why it's working with my RPi 2?

So, what was the outcome of that test? Does it work with a shorter cable on RPi3? Sorry, I did not understand.

opensource-alt, 2020-10-26:
Hello,

> For Orange PI PC 2, can you sync everything to the latest HEAD to include ed5ce9c commit? What I mean is, sync to HEAD and rebuild all mcus as well. Right now you have some modified code for the mcus, so could you make sure that the mcu code is unmodified?

This commit was one of the modifications I did on my local git (and it was the only one). So binary are already up to date.

> The latest error is 'Unable to issue spi ioctl', which is initiated at src/linux/spidev.c:106. So it seems that if the descriptors cache is fixed, the code then cannot perform the spi operation. You can try to check what the errno is (it seems it is not logged in klippy.log).

Correct: the error is not provided by klippy.log because it happens in klipper_mcu binary.
https://github.com/KevinOConnor/klipper/issues/3027#issuecomment-716057273 : "... Still the same error when querying the sensor: unable to obtain spi_transfert_response. The root cause in an issue calling io to function. Error 14, bad address. ..."
Using spidev_test C program: ioctl command is OK. Maybe an issue with address management (like truncate) but I didn't find anything yet.

> We would need the exact error message, and the exact klippy.log of that attempt on RPi 3. Especially interesting is if the received output isn't 0. You can run the query command several times and attach the log.

Error message is: Invalid adxl345 id (got 0 vs e5) most of the time. 
I received 1 time: Invalid adxl345 id (got 25 vs e5) and I already received some other (like random) values.
[klippy.log](https://github.com/KevinOConnor/klipper/files/5437140/klippy.log)
I'm not sure it's sofware related after I saw https://stackoverflow.com/questions/48321492/adxl345-device-id-and-offset-being-wrong-raspberry-pi.
I also tried to setup the max speed at different values (5000000 by default, tested 3900000 and 7800000) as seen on https://elinux.org/RPi_SPI#SPI0_.28available_on_P1_headers_on_all_RPi_versions.29 and https://www.takaitra.com/spi-device-raspberry-pi/  (non standardized speed values values could give non working spi device) but it didn't help.

> So, what was the outcome of that test? Does it work with a shorter cable on RPi3? Sorry, I did not understand.

Yes, that was really unclear, sorry. Shorter cable length (10-15cm) didn't help. 

Philippe

opensource-alt, 2020-10-28:
Hello,

New tests tonight: I setup an Orange PI PC (a 32 bits kernel / latest Ubuntu). It seems there's also some (now fixed) issues with the cable (bad connexions on the printer side) and I now have a working ADXL345.
I still have the unexpected sensor id values errors with the RPi 3. I'll check this part again.

Philippe

opensource-alt, 2020-10-29:
Hello,

New tests yesterday after an ideetified issue with connectors (bad contacts on the ADXL side).
* Orange PI PC (32 bits, latest armbian with up to date packages) still working out of the box
* RPi 3 still have issue reading stable values: most of the time, the ADXL345 has a bad id (0 and other values). spidev_test works & show not stable values (as 0xe5 never at the same place or missing in the output buffer. buffer contains not null values)
* Orange PI PC 2: still have the ioctl error (bad address). spidev_test  (also using ioctl calls) program works & shows stable values (as 0xe5 always at the same place for the output buffer).

Philipppe

opensource-alt, 2020-10-29:
Hello,

I maybe found something in the source code that could explain the "bad address" error obtained with 64 bits kernel (for the other "RPi 3" issue: it's maybe an issue related to hardware, I'll try with another RPi 3 if I can catch one).

In ./src/spicmds.c file: 2 functions gets the arguments with 32 bits integers.
```
void command_spi_transfer(uint32_t *args)
void command_spi_send(uint32_t *args)
```

In both of them:
`    uint8_t *data = (void*)(size_t)args[2];
`

and that's not correct with 64 bits memory adresses.

Philippe

dmbutyugin, 2020-10-30:
@opensource-alt I have RPi 3 running with ADXL345 without problems, so I suppose it can be some HW issue on RPi 3 in your case.

> ```
> void command_spi_transfer(uint32_t *args)
> void command_spi_send(uint32_t *args)
> ```
> 
> In both of them:
> ` uint8_t *data = (void*)(size_t)args[2];`
> 
> and that's not correct with 64 bits memory adresses.

@KevinOConnor that indeed looks like a bug then. To be fair, the truncation of the address on 64 bit platform happens even earlier, at `src/command.c:81`:
```
            *args++ = (size_t)p;
```
for `uint32_t *args`. Maybe it's just this code is usually run on the microcontrollers, and those are 32bits max, so this went unnoticed.

KevinOConnor, 2020-10-30:
> uint8_t *data = (void*)(size_t)args[2];

I agree that looks like a bug.

-Kevin

KevinOConnor, 2020-10-30:
FYI, I put up a work-linuxbuf-20201029 test branch for that issue.

`cd ~/klipper ; git fetch ; git checkout origin/work-linuxbuf-20201029 ; sudo service klipper stop ; make ; make flash ; sudo service klipper start`

-Kevin

KevinOConnor, 2020-10-30:
Thanks.  I committed that fix to the master branch (commit a5604320).

-Kevin

closed 2020-11-28 by github-actions[bot]