Klipper3d/klipper #3470: Choice 'item' for option 'type' is not a valid choice

BobaSchmidt76, 2020-10-25:
Hi,
after updating Klipper I get this message "Choice 'item' for option 'type' is not a valid choice"

What goes wrong?

Regards
Marko

[klippy.log](https://github.com/KevinOConnor/klipper/files/5434524/klippy.log)

BlackStump, 2020-10-25:
[did you check the config changes doc](https://github.com/KevinOConnor/klipper/blob/master/docs/Config_Changes.md)

>20200809: The menu system has been rewritten. If the menu has been customized then it will be necessary to update to the new configuration. See config/example-menu.cfg for configuration details and see klippy/extras/display/menu.cfg for examples.

BobaSchmidt76, 2020-10-25:
In the changes Log is nothing abot the Menu.
I´ve got Mesh Leveling editet months ago. But if I delete the Bed Mesh leveling comes "Option 'set_velocity_limit square_corner_velocity' is not valid in section 'printer'"

theopensourcerer, 2020-10-25:
@BobaSchmidt76 It says exactly that... "20200809: The menu system has been rewritten. " 

See the example-menu.cfg and you will see there is no type called "item"

https://github.com/KevinOConnor/klipper/blob/master/config/example-menu.cfg#L21

BlackStump, 2020-10-25:
This is not a valid entry in your printer.cfg
>set_velocity_limit square_corner_velocity: 3.5

BobaSchmidt76, 2020-10-25:
I´ve updated the Menu.cfg and the Display.cfg. But still the same error. Sorry my English is not so good.

[klippy.log](https://github.com/KevinOConnor/klipper/files/5434623/klippy.log)

theopensourcerer, 2020-10-25:
```
[menu __prepare meshBed zposition]
type = item
```

There is no type called "item".

BobaSchmidt76, 2020-10-25:
What should stand there instead?

BobaSchmidt76, 2020-10-25:
I reinstall klipper.

closed 2020-10-25 by BobaSchmidt76