Klipper3d/klipper #3473: ADXL345 accelerometer expected values?

schnello, 2020-10-25:
Hello.

I try to install an ADXL345 accelerometer at my klipper installation.

Config is:

```
[adxl345]
axes_map: x,y,z
cs_pin: PA15
spi_software_sclk_pin: PC10
spi_software_mosi_pin: PC12
spi_software_miso_pin: PC11
```

ACCELEROMETER_QUERY returns now:
Recv: // adxl345 values (x, y, z): -4628.738800, -2275.142800, 9728.196800

The sensor hangs still in the air.

So my question? Are this values now correct. I guess the values should be 0 because there is no movement or?

Hywelmartin, 2020-10-25:
gravity??

dmbutyugin, 2020-10-25:
Please do follow the guide. As it [mentions](https://github.com/KevinOConnor/klipper/blob/master/docs/Measuring_Resonances.md#checking-the-setup),
> In Octoprint, run ACCELEROMETER_QUERY. You should see the current measurements from the accelerometer, including the **free-fall acceleration**, ...

schnello, 2020-10-26:
Thanks a lot.

closed 2020-10-26 by schnello

schnello, 2020-10-26:
@dmbutyugin 
Only as add question:
Could be the sensor sensitive enough to detect and evaluate the perfect belt tension?

dmbutyugin, 2020-10-28:
@schnello No, I do not think the sensor is good for that. You can certainly measure the effect of tensioning the belt with the sensor, but you won't get to know whether the tension is best or not. Someone shared this article
http://benchtopmachineshop.blogspot.com/2019/04/printer-belt-tension.html
which describes a decent way to check the tension. However, I did not try that myself.

schnello, 2020-10-29:
@dmbutyugin 

Thanks for this link. Works perfect and this is the first guide what includes the distance.