Klipper3d/klipper #3474: Temperature conrolled fans -> watermark: hysteresis

modmt, 2020-10-25:
Hello all,

I want to controll the fan for the radiator of my water cooling system, so I connected it to a MOSFET and added the temperature controlled fan section in my config.

It works good so far, but I miss the option to cofigurate the hysteresis of the watermark control. That means I want to switch on the fan at 35°C and want it to go off when the water reaches 32°C.
For the heater section the is an option "hysteresis:3" for example, but that does not work in the fan section....
Is this also possible for the controlled fans?

Thank you in advance!
Matthias
[klippy.log](https://github.com/KevinOConnor/klipper/files/5435788/klippy.log)

KevinOConnor, 2020-10-28:
The name of the config option is `max_delta` both for heaters and temperature_fan.

-Kevin

modmt, 2020-10-28:
Perfect. Thank you Kevin!

closed 2020-10-28 by modmt