Klipper3d/klipper #3476: MCU_HOST gpio buttons not working.

al3ph, 2020-10-26:
This was working but has now stopped and I've no idea why, if I run a python script to monitor the buttons they work correctly, but klipper doesn't seem to trigger the macros when the buttons are clicked.

The buttons trigger on low, i.e. they connect the pin to ground on click, and are pulled high when not clicked.
The pins are grabbed by klipper, 

pi@BumbleBee:/tmp $ gpioinfo |grep klipper
	line   9:      unnamed    "klipper"   input  active-high [used]
	line  10:      unnamed    "klipper"   input  active-high [used]
	line  22:      unnamed    "klipper"   input  active-high [used]
	line  23:      unnamed    "klipper"   input  active-high [used]
	line  24:      unnamed    "klipper"   input  active-high [used]
	line  25:      unnamed    "klipper"   input  active-high [used]
	line  27:      unnamed    "klipper"   input  active-high [used]

[klippy.log](https://github.com/KevinOConnor/klipper/files/5440428/klippy.log)

al3ph, 2020-10-26:
// front_button1: RELEASED
--
QUERY_BUTTON button=front_button1

The state is always RELEASED, whether they are pressed or not.

cheers.

al3ph, 2020-10-26:
This is the python script I'm using to validate the buttons work.
```
from gpiozero import Button
from time import sleep

button1 = Button(27)
button2 = Button(22)
button3 = Button(23)
button4 = Button(24)
button5 = Button(10)
button6 = Button(9)
button7 = Button(25)

while True:
    if button1.is_pressed:
        print("Pressed 1")
    if button2.is_pressed:
        print("Pressed 2")
    if button3.is_pressed:
        print("Pressed 3")
    if button4.is_pressed:
        print("Pressed 4")
    if button5.is_pressed:
        print("Pressed 5")
    if button6.is_pressed:
        print("Pressed 6")
    if button7.is_pressed:
        print("Pressed 7")
    sleep(0.5)

```

al3ph, 2020-10-26:
I noticed that rpi GPIO pins have a degree of configuration, i.e you can configure the pull mode,  Up/Down/None.
I wonder if this might be the missing piece, to my problem.

al3ph, 2020-10-26:
Unfortunately all buttons trigger on firmware restart. Initial state wrong ?

al3ph, 2020-10-26:
Ah negating the pin works.

theopensourcerer, 2020-10-27:
Just in case, I believe you can enable the pull-up or pull-down in the klipper config... 

From example.cfg

"
# names may be preceded by an '!' to indicate that a reverse polarity
# should be used (eg, trigger on low instead of high). Input pins may
# be preceded by a '^' to indicate that a hardware pull-up resistor
# should be enabled for the pin. If the micro-controller supports
# pull-down resistors then an input pin may alternatively be preceded
# by a '~'.
"

al3ph, 2020-10-27:
Ah, that's exactly what I needed, not sure why I didn't spot that in the docs.. I think I was looking at the RPI example.. Cheers!

closed 2020-10-27 by al3ph