Klipper3d/klipper #3477: Upgraded to Creality 4.2.7 board, can not flash klipper.

jerkwagon, 2020-10-26:
was running old board, couldnt get the BL touch to work, was tinkering fried the board,  bought a new board, BL touch was broken from teh start, bought a new one, works, but i can not flash klipper onto the board. 

pi@octopi:~/klipper $ make flash FLASH_DEVICE=/dev/serial/by-id/usb-1a86_USB2.0-Serial-if00-port0
  Flashing out/klipper.bin to /dev/serial/by-id/usb-1a86_USB2.0-Serial-if00-port0

Failed to flash to /dev/serial/by-id/usb-1a86_USB2.0-Serial-if00-port0: Unable to find tty device

If the device is already in bootloader mode it can be flashed with the
following command:
  make flash FLASH_DEVICE=1eaf:0003
  OR
  make flash FLASH_DEVICE=1209:beba

If attempting to flash via 3.3V serial, then use:
  make serialflash FLASH_DEVICE=/dev/serial/by-id/usb-1a86_USB2.0-Serial-if00-port0

make: *** [src/stm32/Makefile:68: flash] Error 255
[image]
[klippy.log](https://github.com/KevinOConnor/klipper/files/5442021/klippy.log)

npcizzy, 2020-10-27:
you'll have to get the klipper.bin off of your pi and flash it via microsd. i've tried but it only leaves my printer unresponsive until I reflash marlin onto it.

jerkwagon, 2020-10-27:
@npcizzy  so theres no way to get klipper on the 4.2.7 yet?  i did copy/rename the bin but no luck as i still can not communicate with the printer via usb.

npcizzy, 2020-10-27:
@jerkwagon i've read multiple success posts about people getting klipper to work on these new creality boards but i have yet to done the same. the config does exist but i truly don't understand what i am doing wrong. i've attempted this on an mks sgen l 32 board but i get the same result. loss of communication from the boards.

jakep82, 2020-10-27:
Did you verify the device ID?  The flash is failing because it doesn't see the serial device.  What is the output of `ls /dev/serial/by-id/*` when the board is connected?

jakep82, 2020-10-27:
Actually looking at the instructions for the 4.2.2 board, you need to disable USB for communication and flash via SD card.  I don't know if the 4.2.7 is the same, but that's where I would start.

jerkwagon, 2020-10-28:
@jakep82 OK, how do i communitcate with the device after its been  flashed via SD?

I tried the SD, but i was unable to connect to the device, maybe ill try again.

Gilabite, 2020-10-28:
You flash it with the SD card and then use the USB cable to communicate with it after its been flashed. On some of the boards you cant flash through the USB cable and have to still use the SD card to do it.

KevinOConnor, 2020-10-28:
To use the v4.2.7 board, use the config found at https://github.com/KevinOConnor/klipper/blob/master/config/generic-creality-v4.2.7.cfg and be sure to carefully follow the directions at the top of that file.

-Kevin

jerkwagon, 2020-10-28:
pi@octopi:~/klipper $ ls /dev/serial/by-id/*
/dev/serial/by-id/usb-1a86_USB_Serial-if00-port0

The pi sees the printer,  but octoprint can not connect to it. I tried auto on both, but then added the usb_serial

[image]
[image]

Hywelmartin, 2020-10-28:
please read the installation guide

https://github.com/KevinOConnor/klipper/blob/master/docs/Installation.md#configuring-octoprint-to-use-klipper

jerkwagon, 2020-10-29:
@Hywelmartin ummm im sorry but can you be a bit more specific?  I have installed klipper a bunch of times,  just ol the old boards, so this one is new to me. I have followed that , the CFG is updated. yet nothing works. 

telling someone to read the guide when they already followed it it really not helpful in any way, Especially when the guy who wrote the entire thing commented and said theres a different way to install for that board.

jerkwagon, 2020-10-30:
@Gilabite  so i think it has something to do with the firmware itself?  when i loaded klipper onto the SD card then udated the firmware to klipperr. i pluged it into a pc and connected it to pronterface, but this came up

Connecting...
‚Ó
h‚yJ
`‚{w×
hŠ{oÇ
`‚sw×
hŠ{oÇ
`‚{w×

after putting stock 4.2.7 marlin it connects no problem. i will try one more thing,  to install on a blank SD card...

fingers crossed.

Gilabite, 2020-10-30:
I dont know for sure because I havent used marlin very much but Im not sure if pronterface will show anything properly for klipper, isnt it for marlin? Another idea if they do normally work together is that looks like baud rates being diff between the two.

jerkwagon, 2020-10-30:
make clean
make menuconfig 
followed the instructions
copied .bin to fresh SD

turn on the printer.

still wont connect to the pi and i have no idea how to test the printer any other way

@KevinOConnor  are you able to chime in?

jakep82, 2020-10-30:
Have you run `ls /dev/serial/by-id/*` after connecting the board to verify you have the correct serial device id in your config?

jakep82, 2020-10-30:
Also the last log you posted has the wrong pins for a 4.2.7 board.

jerkwagon, 2020-10-30:
@jakep82 

pi@octopi:~/klipper $ ls /dev/serial/by-id/*
/dev/serial/by-id/usb-1a86_USB_Serial-if00-port0

Config mcu is this
[mcu]
serial: /dev/serial/by-id/usb-1a86_USB_Serial-if00-port0
restart_method: command

here is my new klippy.  
[klippy oct 30th.log](https://github.com/KevinOConnor/klipper/files/5467779/klippy.oct.30th.log)

jerkwagon, 2020-10-30:
https://pastebin.com/HnhGeJk0

communitcation to printer.

jerkwagon, 2020-10-30:
So i have no idea what i did to fix it,  but it now connects and i can move the printer.   I will have to figure out how to get the display to work and what pins are needed for it to show up..  

Same with BL touch, i will try discord if no luck, submit a new ticket.

closed 2020-10-30 by jerkwagon

er9xuser, 2021-01-21:
I know it is old post. But tested today...on creality board v4.2.7. This is my config and LCD is working. Maybe someone should change default file: https://github.com/KevinOConnor/klipper/blob/master/config/generic-creality-v4.2.7.cfg

# This file contains pin mappings for the Creality "v4.2.7" board. To
# use this config, during "make menuconfig" select the STM32F103 with
# a "28KiB bootloader" and with "Use USB for communication" disabled.

# If you prefer a direct serial connection, in "make menuconfig"
# select "Enable extra low-level configuration options" and select the
# USART3 serial port, which is broken out on the 10 pin IDC cable used
# for the LCD module as follows:
# 3: Tx, 4: Rx, 9: GND, 10: VCC

# Flash this firmware by copying "out/klipper.bin" to a SD card and
# turning on the printer with the card inserted. The firmware
# filename must end in ".bin" and must not match the last filename
# that was flashed.

# See docs/Config_Reference.md for a description of parameters.

[stepper_x]
step_pin: PB9
dir_pin: PC2
enable_pin: !PC3
microsteps: 16
rotation_distance: 40
endstop_pin: ^PA5
position_endstop: 0
position_max: 235
homing_speed: 50

[stepper_y]
step_pin: PB7
dir_pin: PB8
enable_pin: !PC3
microsteps: 16
rotation_distance: 40
endstop_pin: ^PA6
position_endstop: 0
position_max: 235
homing_speed: 50

[stepper_z]
step_pin: PB5
dir_pin: !PB6
enable_pin: !PC3
microsteps: 16
rotation_distance: 8
endstop_pin: ^PA7
position_endstop: 0.0
position_max: 250

[extruder]
max_extrude_only_distance: 100.0
step_pin: PB3
dir_pin: PB4
enable_pin: !PC3
microsteps: 16
rotation_distance: 33.500
nozzle_diameter: 0.400
filament_diameter: 1.750
heater_pin: PA1
sensor_type: EPCOS 100K B57560G104F
sensor_pin: PC5
control: pid
pid_Kp: 21.527
pid_Ki: 1.063
pid_Kd: 108.982
min_temp: 0
max_temp: 250

[heater_bed]
heater_pin: PA2
sensor_type: EPCOS 100K B57560G104F
sensor_pin: PC4
control: pid
pid_Kp: 54.027
pid_Ki: 0.770
pid_Kd: 948.182
min_temp: 0
max_temp: 130

[fan]
pin: PA0

[mcu]
serial: /dev/serial/by-id/usb-1a86_USB_Serial-if00-port0
restart_method: command

[printer]
kinematics: cartesian
max_velocity: 300
max_accel: 3000
max_z_velocity: 5
max_z_accel: 100

[display]
lcd_type: st7920
cs_pin: PB12
sclk_pin: PB13
sid_pin: PB15
encoder_pins: ^PB14, ^PB10
click_pin: ^!PB2

ahinrichs, 2021-02-07:
I just tried an can confirm. With this config the LCD is working on my Creality v4.2.7 board.

WuuWuu87, 2021-07-08:
Anyone Figure out how to get the 4.2.7 board to connect ? i have tried 20 differnt usb cables and 2 differnt configs and still no luck with getting it to connect i still get a MCU cant connect error?

cartesian-theatrics, 2021-12-20:
I had to remove the [mcu rpi] section when connecting via. USB without Rasberry Pi (I'm just using an Intel machine running linux mint).

roshanabady, 2022-03-30:
I just got my 4.2.7 board flashed, and it is working successfully after a couple of restarts of all parts (Octoprint, Klipper server and Hosts). 
Besides setting the correct pins in the config file, I used the same 8GB sd card that came in with the printer.
I was using 64GB Samsung pro endurance with no luck, and the no-name 8GB sd did the trick for me.
I hope this works for anyone looking for a solution as well. 
Crealitiy's official page for the 4.2.7 board says, "Prepare an SD card less than 8GB."