Klipper3d/klipper #3478: Gcode buttons triggering when started

xbst, 2020-10-27:
Hi,
I'm getting a weird bug with gcode buttons. I have them wired so they short to GND when pressed, but they trigger every time printer is repowered (Y U no FIRMWARE_RESTART?, explained below)
Initially I had them wired to a Maple Mini but during troubleshooting I tried a endstop pin on my SKR 1.3 and the same happened.
Also even with nothing plugged in, the button gets triggered, so definitely no wiring issue.
Another thing I noticed, with a (or more) gcode button in printer.cfg, if I FIRMWARE_RESTART, I get a communication error. If I remove (comment) them, it works fine.

One of the buttons as an example:
...
[gcode_button pa15]
pin: ^stm:PA15
press_gcode: BED_MESH_CALIBRATE
...
[klippy.log](https://github.com/KevinOConnor/klipper/files/5445958/klippy.log)

jakep82, 2020-10-27:
The sample you've posted references an MCU named "stm".  I don't see an MCU in your log with that name.  Also I believe if you want the pins to trigger low, you need to use a ! in front of your pins.  So you would want:

```
[gcode_button pb3]
pin = ^!z:P1.27
press_gcode = G28

xbst, 2020-10-27:
Log was the SKR test
[klippy.log](https://github.com/KevinOConnor/klipper/files/5447953/klippy.log)
I inverted the pin, same results

jakep82, 2020-10-27:
Nothing in the log indicates you inverted the pin I mentioned, and in your final restart it was unable to connect to the stm MCU you added.  You need to get the MCU connected and working before anything related to buttons will work.

xbst, 2020-10-28:
MCU connection problem so far only happened when I have a gcode button in printer.cfg, so I assumed it didn't work since it happened again. But rebooting the system led to connectivity and buttons work now. 

However FIRMWARE_RESTART still leads to mcu connection issue, regardless of printer.cfg. I was wrong about the cause, but this issue still exists. Always with the STM32 MCU.

[klippy.log](https://github.com/KevinOConnor/klipper/files/5451313/klippy.log)

jakep82, 2020-10-28:
Try a different restart_method for that particular MCU.

https://github.com/KevinOConnor/klipper/blob/master/config/example.cfg#L290

xbst, 2020-10-28:
[klippy.log](https://github.com/KevinOConnor/klipper/files/5453200/klippy.log)
Tried all methods, by both doing a firmware_restart after editing, and tried doing a reboot then firmware_restart. In all attempts, I get errors.

xbst, 2020-10-28:
[klippy.log](https://github.com/KevinOConnor/klipper/files/5453227/klippy.log)
I uploaded the wrong log.

jakep82, 2020-10-28:
If it works properly after initial startup, I would expect `restart_method: rpi_usb` to work.  All that does is toggle USB power off and on to the device which should restart it.  If this doesn't work, then it's possible you have an issue with the bootloader on your board, or you've found a bug in Klipper.  I'm not able to help beyond this.

Arksine, 2020-10-28:
Some bootloaders have issues enumerating the USB device after a software reset.  It is very likely that is the issue here.  The board should re-enumerate after a power cycle or a hard reset.   If the board has a reset pin on one of its headers you could wire a button to it.

Alternatively you could try flashing another [bootloader](https://github.com/KevinOConnor/klipper/blob/master/docs/Bootloaders.md#stm32f103-micro-controllers-blue-pill-devices).  If you choose this route, keep in mind that you will lose the ability to flash using the sd card, and you will need to choose the correct bootloader offset when building Klipper.

xbst, 2020-10-28:
Maple Mini (clone) board I use does have a reset pin. Is it possible to configure klipper to reset the MCU with the pin when doing a restart? My aim is to be able to restart the firmware without doing a reboot or being near the printer to press a button.
I don't want to flash a new bootloader as flashing stm32duino was painful enough.

closed 2020-11-26 by github-actions[bot]