Klipper3d/klipper #3480: Turn dual zone heaters on automatically by model size

Gilabite, 2020-10-28:
Hello, I'm building a large CoreXY printer with a dual zone bed heater and would like to add a way to turn the zones on automatically based off the area of the model. I was thinking I could have klippy calculate the models area by parsing the gcode file and looking for min/max x/y calculate the area and store the value and have a macro that would check the value and turn the heaters on accordingly or something to that effect. Obviously would need a config setting to turn the area calculation on/off so it wouldn't do it for people that don't need this. And a config setting to store the size of the zones. It would be my first contribution and I was hoping @KevinOConnor could offer some advice on if this is something that would get accepted and what files I might look into to add the feature if its something worthwhile and if there is any functions already made that may help with the process. If someone else more experienced with klippers codebase wanted to add this that would be cool too, I would not be offended lol. Thanks.

Gilabite, 2020-10-28:
Don't have a log file Mr.GitIssueBot. Looks like Cura already calculates min/max values for all axis and puts it at the beginning of the gcode file, not sure if they all do this or not.

jhpadjustable, 2020-10-30:
I can think of three reasons that functionality doesn't belong in the controller:

1. In G-code, the ; declares a comment. Print servers and slicers extend comments as an out-of-band channel for metadata. In any case, most print servers strip comments from the controller's command stream to maximize command bandwidth, in which case the controller will never see them.
2. A controller's proper concern is to respond to dynamic, real-world observations as commanded. Resource allocation is a planning concern that is static across runs of a given job, and should have already been factored into the job instructions.
3. Simplify3D already set a precedent of bed heater zone control as a slicer responsibility.

To do it properly, you will inevitably need to use Cura's start and end G-code in the printer configuration. I see two obvious ways to do so.

* Parallelling your original plan, the slicer's start G-code would be modified to invoke some e.g. SET_BED_BOUNDING_BOX Klipper macro, which stashes the XY bounding box of the job in the controller. Another Klipper macro or two would override M140/M190 commands to stash the bed temperature set value, then perform a SET_HEATER_TEMPERATURE on each bed heater, computed according to the stashed values. To be absolutely proper, the bounding box macro would also need to recompute and set each heater value in case the bounding box changes.
* More simply, I would simply use the same Cura variables and conditionals on multiple SET_HEATER_TEMPERATURE commands/parameters in that start G-code block, and alleviate the controller of that concern entirely.

Gilabite, 2020-10-30:
Awesome, thank you sir. I was hoping someone had some kind of pointers, new to 3D printing but not coding. I was thinking of it being more of a klippy thing on the pi before hand not on the printers side, but Iam still taking your advice though. So am I understanding right that Simplify3D has automatic dual zone functionality already? Iam used to Cura and would probably still go that route but would give Simplify a shot if it already has the functionality especially to start off with till I got the Cura part going. Ive tried a bunch of slicers but cant recall if simplify was one, not that I would have noticed anything about dual zone heating at the time. Thanks again.

Gilabite, 2020-10-30:
Ok I have not used Simplify, no trial lol.

Gilabite, 2020-10-31:
So I got this running with a Cura post processing script to get the bounds of the mesh and have the start gcode call a klipper macro with the value, thank you very much @jhpadjustable for the pointers.