Klipper3d/klipper #3483: action_raise_error fails

lixxbox, 2020-10-30:
Hello,
i'm attempting to write some small gcode macros and would like to set an action_raise_error to not execute the rest of the macro in certain scenarios.

> action_raise_error(msg): Abort the current macro (and any calling macros) and write the given msg to the /tmp/printer pseudo-terminal. The first line of msg will be sent with a “!! “ prefix and subsequent lines will have a “// “ prefix.

it looks like i get an erro
[klippy.log](https://github.com/KevinOConnor/klipper/files/5464335/klippy.log)
r message telling me that the function action_raise_error could not be executed correctly:
```
Error evaluating 'gcode_macro lixxbox:gcode': CommandError: action_raise_error
test1
Traceback (most recent call last):
  File "/home/pi/klipper/klippy/extras/gcode_macro.py", line 61, in render
    return str(self.template.render(context))
  File "/home/pi/klippy-env/local/lib/python2.7/site-packages/jinja2/environment.py", line 1008, in render
    return self.environment.handle_exception(exc_info, True)
  File "/home/pi/klippy-env/local/lib/python2.7/site-packages/jinja2/environment.py", line 780, in handle_exception
    reraise(exc_type, exc_value, tb)
  File "<template>", line 2, in top-level template code
  File "/home/pi/klipper/klippy/extras/gcode_macro.py", line 89, in _action_raise_error
    raise self.printer.command_error(msg)
CommandError: action_raise_error
test1
Error evaluating 'gcode_macro lixxbox:gcode': CommandError: action_raise_error
test1
```

it does serve its purpose and breaks the macro, but perhaps it should be fixed.

Arksine, 2020-10-30:
It is behaving as intended,  `action_raise_error` raises an exception and Klipper logs that exception.

lixxbox, 2020-10-30:
ok, thanks for the feedback. then i guess i just had wrong expectations regarding the function.

i had hoped there would be a proper error message like `RESPOND TYPE=error MSG="<message>"` and it would additionally terminate the macro.

lixxbox, 2020-11-03:
i have found another solution for my specific problem.
as this is the intended behaviour, I close the issue.

closed 2020-11-03 by lixxbox

nofuturekid, 2021-01-02:
@lixxbox: Maybe you can share your solution? I'm currently working on macros too and fall into the same pitfall like you.