Klipper3d/klipper #3484: neopixel_update command not found

pheonix3330, 2020-10-30:
I updated to the latest klipper firmware and also updated the mcu firmware just in case. I made the color_order change listed in the klipper config changes but I am still getting an error when trying to boot. "command not found: neopixel_update". The only way I can get klipper to start is by commenting out the neopixel lines.

Here is what the neopixel section looks like in my config:
[neopixel fysetc_mini12864]
pin: P1.21
chain_count: 3
initial_RED: 1.0
initial_GREEN: 0.0
initial_BLUE: 0.0
color_order: RGB

Any ideas what could be happening?

KevinOConnor, 2020-10-30:
When upgrading the software, please be sure to always follow the directions in the FAQ: https://www.klipper3d.org/FAQ.html#how-do-i-upgrade-to-the-latest-software

In particular, you will sometimes need to reflash the firmware.

-Kevin

pheonix3330, 2020-10-30:
Thanks Kevin, I thought I had reflashed the mcu but I realize now I forgot to rename the klippy.bin file to firmware.bin before loading it onto the sd card for my skr 1.4. Thanks for the response.

closed 2020-10-30 by pheonix3330