Klipper3d/klipper #3485: Proposed style change for Klipper example configuration files.

mtw3d, 2020-10-31:
I am proposing that we define a preferred style for example config files in a way that makes configuration files as close to controller-neutral as possible. This change requires no code changes at all to Klipper itself, it is entirely done at the config level. But the result is that it makes swapping controllers an almost trivial process.

We accomplish this by taking full advantage of the existing `board_pins` alias capability in Klipper. As an example, here is my `board_pins` for an SKR E3 2.0:

    [board_pins mcu] 
    # To use these pins in the config, refer to them with by the pin name only, iow XSTEP
    # Use ^ to activate the pullup, ^XSTEP, or ! to invert the logic, !XSTEP
    aliases: XSTEP=PB13, XDIR=PB12, XEN=PB14, XSTOP=PC0,
             YSTEP=PB10, YDIR=PB2, YEN=PB11, YSTOP=PC1,
             ZSTEP=PB0, ZDIR=PC5, ZEN=PB1, ZSTOP=PC2,
             #Extruder
             ESTEP=PB3, EDIR=PB4, EEN=PD2, EHEAT=PC8, ETHERM=PA0,
             #Heated Bed
             BEDHEAT=PC9,
             BEDTHERM=PC3,
             #fans
             FAN0=PC6,
             FAN1=PC7,
             #bltouch
             BLTCTL=PA1,
             BLTSEN=PC14,
             # EXP1 header
             EXP1_1=PB5, EXP1_3=PA9,   EXP1_5=PA10, EXP1_7=PB8, EXP1_9=<GND>,
             EXP1_2=PA15, EXP1_4=<RST>, EXP1_6=PB9,  EXP1_8=PB15, EXP1_10=<5V>
             # Other Pins
             PS-ON=PC13, PT-DET=PC12, ESTOP=PC15, NEOPIX=PA8 

That means my `[stepper_x]` config looks like this:

    [stepper_x] # All pins on board XBRD Port X
    step_pin: XSTEP
    dir_pin: XDIR   
    enable_pin: !XEN 
    step_distance: .0025
    endstop_pin: ^XSTOP
    position_min: -100
    homing_speed: 60
    position_max: 260
    homing_retract_dist: 5
    homing_positive_dir: FALSE
    homing_retract_speed: 20

As you can see, with the possible exception of pullups, logic inversion, and step distance, this should work on nearly any controller. 

This means that to make a new example config for any new controller will only require creating the `board_pins` block, and modifying a bare minimum of settings in a baseline config, plus adding any new board-specific features. 

It also means swapping controllers in the future is much easier. So if I want to replace my controller a year from now, all I have to do to switch is swap out this simple code block, plus make any necessary changes to board-specific code, such as stepper driver configs and step distances. Everything else in my existing config (code macros, etc.) should work with little or no additional modification, as long as they aren't relying on board-specific features. Even if they are, the change will be easier than it is using the existing style of calling the pin numbers directly. 

As a side effect, it is actually easier to initially setup the example config, since all the pin definitions are in the same spot. It's much quicker to edit all the pins at once, then to have to do them as you come to them in the config file. 

So to be clear, here is all that I am proposing:

1. We decide on standard pin alias names for the most common pins. I propose basing them on the ones I use here, since they are clear and simple, but I am open to advice if there are issues with these names.
2. We encourage (but don't require?) all new example configs to use this style. Given how much easier it is to define the pins this way, I suspect most people will prefer it once they get used to it.
3. We recommend but don't require existing configs to be updated to use this new style. 

As a real-world example of the benefit, I use this style on a IDEX printer that uses two of the SKR E3 controllers. I am in the process of adding IDEX to another existing printer that uses a single RAMBo controller. To make the switch, I will be adding a single SKR E3 in addition to the existing controller. 

To convert my existing dual E3 config to work with one E3 and one RAMBo, I just had to swap out the second `board pins` block, remove the TMC2209 configs where relevant, and add the stepper current configs for the Rambo. Total conversion process was probably 10 minutes. In reality is was a bit more than that, since I had to create the `board_pins` block for the rambo, but even that only took maybe another 10-15 minutes, and if it if this was our standard style for the future, it would have just been a matter of copying and pasting the code block in. 

I can't see any downside to using this style at all, and it has quite a bit of upside. It's easier both for the person creating the config, as well as the end user. But I welcome any feedback.

vladbabii, 2020-10-31:
This would work well if we would require for any new boards an alias pin definition for the config, which users could get from the config example folder and just [include ..] in their printer config.

mtw3d, 2020-10-31:
I like the [include] idea. You could even completely isolate the controller specific code that way. For example have:

```
[stepper_x] 
step_pin: XSTEP
dir_pin: XDIR   
enable_pin: !XEN 
endstop_pin: ^XSTOP
position_min: -100
homing_speed: 60
position_max: 260
homing_retract_dist: 5
homing_positive_dir: FALSE
homing_retract_speed: 20
```
in the `printer.cfg`

and 

``` 
[stepper_x]
step_distance: .0025
```
(and any other controller-specific configurations) in the `controller.cfg`. Klipper is already smart enough to treat multiple `[stepper_x]` blocks as a single block, so this this is an easy change that would fully isolate any controller-specific code.

That does add a touch of extra complexity for the end-user though, since they would now have to know to look somewhere else for the extra code, so I am not sure if this ends up making things easier or not. I think it's probably a good idea, but it's not quite as obvious.

theopensourcerer, 2020-10-31:
This is a very nice idea. 

One potential code change might be to support the inversion and pull-up/pull-down nomenclature in the alias definitions... Just a thought.

mtw3d, 2020-10-31:
Yeah, that would be my preference as well. At least for now I wanted the suggestion to not require any code changes, but that would definitely be my preferred long-term behavior.

KevinOConnor, 2020-10-31:
FWIW, my high-level feedback is that I agree this could be a nice improvement.

Some comments:
* I think some minor code changes would be helpful.
  1. It might be nice to add support for something like `[include %config/pins-rambo.cfg]` such that `%config` would automatically resolve to the Klipper source code config/ directory.
  2. The `[board_pins]` config section really should take any number of `aliases_XXX` options so that pins can be listed more conveniently in the config.
  3. I now find the `[board_pins mcu]` config format to be cumbersome, and I now think it would be preferable to have an explicit `mcu: my_mcu1, my_mcu2` config option in the board_pins config section.  Specifically, the idea would be allow one to do something like `[include %config/pins-rambo.cfg] ... [board_pins rambo] mcu: my_zboard` to include a pin file and then specify the mcu(s) it applies to.
* If we were to go this route, I do think we'd really need to go the whole way and convert most of the existing example config files.  Having multiple different example config styles, I fear will be a greater confusion to new users, and I fear that would ultimately outweigh any gain from using board_pins.
  1. It's a lot of work converting all the configs.  If we want to go down this route I'd really like to see commitment from a few individuals to do all that work.  (I can help with any coding; the majority of the work is in the config conversion.)  To be clear - this is work - I think we'd really want to have someone go over the conversion of each config in detail to make sure no actual change is made to each pin, have someone else review every pin, find the person who originally submitted the config and ask them to review, and track when it's time to commit each config.
* I think we'd also want a new docs/Example_Config_Format.md guide describing the expected format of example configs.  If there is a desire to go down this route, this is probably where'd I'd recommend starting.

Cheers,
-Kevin

mtw3d, 2020-10-31:
I like all this feedback, Kevin. 

On the code changes, would any of these changes be blocking changes, things that the code needs to be in place before we can start using this system? It doesn't seem like it, but you have a better sense what is involved with those changes. 

It seems like the most logical way to proceed with the conversion process is to get a volunteer to undertake the conversion for a printer that they own. That lets them edit the config as needed, and then test the config before submitting it. This would ideally be the original config author, but anyone with the appropriate hardware could do it. I bet between asking on the Klipper IRC / Discord and a post on Reddit, we could find volunteers to take care of most of the configs. 

That said, I am happy to help out in any way that I can. I have somewhat limited time, but should be able to make a decent commitment. 

I'm not a tech writer, but I can probably write up the Example_Config_Format.md. Can you either point me to a model to aim for, or give me more of a general overview of what you think it should include? I'm assuming identifying the most common pin names (I'll go with the style in my example unless I hear otherwise), and a general overview of what should be in the controller config vs. what should be in the machine config? Anything else?

mtw3d, 2020-10-31:
Question: Can a give pin be given more than one alias? For example, on the RAMBo, we use a pin on a header block as the control pin for the BLTouch, but someone else might want to use that header block for some other purpose. Obviously they would only be able to use the header for one thing at a time. Can I set up an alias for (for example, though these are not the actual pins and it is not that same header) BLTSEN=PA1 and EXP1_1=PA1, then call it by whichever name makes sense for the end user?

Obviously if a user really did want to use both the BLT and the header, they would need to find some other pin that works for the BLT, and edit the config, but I don't think we will ever be able to eliminate all edge cases... The goal is just to make it as easy as possible for as many people as possible,

mtw3d, 2020-10-31:
I took a first stab at a example_config_format doc... This almost certainly is nothing at all like what you want, so any feedback you have is welcomed :-)

```
This document explains the proper formatting of Klipper example config files for printer controllers and printers. The goal of this document is to help maximize the separation of printer-specific configuration information from controller-specific configuration. This makes it much easier for users o change their electronics in the future with minimal frustration.

To manage this we do two things: 

1. We separate the configurations into two separate files.
2. We take advantage of klippers `board_pins` alias feature to define all the controller's pins in one place, using common, predefined pin names that can be called in a generic fashion. 

Pin Name Convention:

The goal here is not to create a common name for every possible pin. The goal here is to create common names for all commonly used pins, so if a user wants to switch from *controller_x* to *controller_y*, as few changes to their config as possible will be required. We are also creating a naming convention that will allow you to choose new names for your own pins that will, hopefully, make it easy to add new pins that others will intuitively use.

Although Klipper is not case-sensitive, Pin Names should always be cited in upper case. for ease of legibility.

These are the pin names we use for the most common pins:

For axis pins, use the axis letter and the first two-four letters of the usage name. Just enough to be clear:

XSTEP
XDIR
XEN
XSTOP

For other axes, substitute the appropriate first letter.

For the primary extruder, use

ESTEP
EDIR
EEN
EHEAT
ETHERM

If your controller supports multiple extruders, start numbering the additional extruders at 1. So the primary extruder (E0) step pin is ESTEP. E1 will be E1STEP. E2 would be E2STEP, etc. 

For the heated bed, use BEDHEAT and BEDTHERM.

Fans should be named according to the port used on the board: FAN0, FAN1, etc.

We recommend defining pin names for every pin on a board that has a specific purpose. This does not include pins that are just broken out to pin headers, unless that header is used for a particular purpose (such as the LCD headers).

Other pins can be named as you see fit, with the convention that it should be as short as possible while remaining clear. For example, we use BLTCTL and BLTSEN for the BLTouch Control and Sense pins, respetively. These names are both easy to understand, and short to type. Given the number of different 

For an example of all this put into practice, here is the `board_pins` for a BigTreeTech SKR e3 Mini v2.0:

[board_pins mcu] 
# To use these pins in the config, refer to them with by the pin name only, iow XSTEP
# Use ^ to activate the pullup, ^XSTEP, or ! to invert the logic, !XSTEP
aliases: XSTEP=PB13, XDIR=PB12, XEN=PB14, XSTOP=PC0,
         YSTEP=PB10, YDIR=PB2, YEN=PB11, YSTOP=PC1,
         ZSTEP=PB0, ZDIR=PC5, ZEN=PB1, ZSTOP=PC2,
         #Extruder
         ESTEP=PB3, EDIR=PB4, EEN=PD2, EHEAT=PC8, ETHERM=PA0,
         #Heated Bed
         BEDHEAT=PC9,
         BEDTHERM=PC3,
         #fans
         FAN0=PC6,
         FAN1=PC7,
         #bltouch
         BLTCTL=PA1,
         BLTSEN=PC14,
         # EXP1 header
         EXP1_1=PB5, EXP1_3=PA9,   EXP1_5=PA10, EXP1_7=PB8, EXP1_9=<GND>,
         EXP1_2=PA15, EXP1_4=<RST>, EXP1_6=PB9,  EXP1_8=PB15, EXP1_10=<5V>
         # Other Pins
         PS-ON=PC13, PT-DET=PC12, ESTOP=PC15, NEOPIX=PA8 
		 
Pins should be named according to their board-usage, even if you might be using them otherwise. For example I have a filament runout switch connected to my ZSTOP pin, but it is called the ZSTOP pin. This is important because it creates a degree of self-documentation. If you ever need to rewire it, you can just refer to the config to know exactly how you had it wired. 		 
		

What Else to Include in the Controller config?

In addition to the `board pins` aliases, we recommend including any controller-specific configs in the controller config file. For example, the SKR E3 Mini shown above has built-in TMC2209 Stepper drivers. The definitions for these blocks is mostly controller-specific, so they should be deinfed in the controller file. The Axes definitions themselves (`stepper_x`, etc) are machine specific, so they will not be defined in the controller_config. 

```

mtw3d, 2020-10-31:
I took a first stab at a example_config_format doc... This is probably not at all what you want, so I welcome any feedback.

This document explains the proper formatting of Klipper example config files for printer controllers and printers. The goal of this document is to help maximize the separation of printer-specific configuration information from controller-specific configuration. This makes it much easier for users o change their electronics in the future with minimal frustration.

To manage this we do two things: 

1. We separate the configurations into two separate files.
2. We take advantage of klippers `board_pins` alias feature to define all the controller's pins in one place, using common, predefined pin names that can be called in a generic fashion. 

Pin Name Convention:

The goal here is not to create a common name for every possible pin. The goal here is to create common names for all commonly used pins, so if a user wants to switch from *controller_x* to *controller_y*, as few changes to their config as possible will be required. We are also creating a naming convention that will allow you to choose new names for your own pins that will, hopefully, make it easy to add new pins that others will intuitively use.

Although Klipper is not case-sensitive, Pin Names should always be cited in upper case. for ease of legibility.

These are the pin names we use for the most common pins:

For axis pins, use the axis letter and the first two-four letters of the usage name. Just enough to be clear:

```
XSTEP
XDIR
XEN
XSTOP
```

For other axes, substitute the appropriate first letter.

For the primary extruder, use

```
ESTEP
EDIR
EEN
EHEAT
ETHERM
```

If your controller supports multiple extruders, start numbering the additional extruders at 1. So the primary extruder (E0) step pin is `ESTEP`. E1 will be `E1STEP`. E2 would be `E2STEP`, etc. 

For the heated bed, use `BEDHEAT` and `BEDTHERM`.

Fans should be named according to the port used on the board: `FAN0`, `FAN1`, etc.

We recommend defining pin names for every pin on a board that has a specific purpose. This does not include pins that are just broken out to pin headers, unless that header is used for a particular purpose (such as the LCD headers).

Other pins can be named as you see fit, with the convention that it should be as short as possible while remaining clear. For example, we use `BLTCTL` and `BLTSEN` for the BLTouch Control and Sense pins, respectively. These names are both easy to understand, and short to type. Given the number of different 

For an example of all this put into practice, here is the `board_pins` for a BigTreeTech SKR e3 Mini v2.0:

```[board_pins mcu] 
# To use these pins in the config, refer to them with by the pin name only, iow XSTEP
# Use ^ to activate the pullup, ^XSTEP, or ! to invert the logic, !XSTEP
aliases: XSTEP=PB13, XDIR=PB12, XEN=PB14, XSTOP=PC0,
         YSTEP=PB10, YDIR=PB2, YEN=PB11, YSTOP=PC1,
         ZSTEP=PB0, ZDIR=PC5, ZEN=PB1, ZSTOP=PC2,
         #Extruder
         ESTEP=PB3, EDIR=PB4, EEN=PD2, EHEAT=PC8, ETHERM=PA0,
         #Heated Bed
         BEDHEAT=PC9,
         BEDTHERM=PC3,
         #fans
         FAN0=PC6,
         FAN1=PC7,
         #bltouch
         BLTCTL=PA1,
         BLTSEN=PC14,
         # EXP1 header
         EXP1_1=PB5, EXP1_3=PA9,   EXP1_5=PA10, EXP1_7=PB8, EXP1_9=<GND>,
         EXP1_2=PA15, EXP1_4=<RST>, EXP1_6=PB9,  EXP1_8=PB15, EXP1_10=<5V>
         # Other Pins
         PS-ON=PC13, PT-DET=PC12, ESTOP=PC15, NEOPIX=PA8```
		 
Pins should be named according to their board-usage, even if you might be using them otherwise. For example I have a filament runout switch connected to my `ZSTOP` pin, but it is called the `ZSTOP` pin. This is important because it creates a degree of self-documentation. If you ever need to rewire it, you can just refer to the config to know exactly how you had it wired. 		 
		
What Else to Include in the Controller config?

In addition to the `board pins` aliases, we recommend including any controller-specific configs in the controller config file. For example, the SKR E3 Mini shown above has built-in TMC2209 Stepper drivers. The definitions for these blocks is mostly controller-specific, so they should be deinfed in the controller file. The Axes definitions themselves (`[stepper_x]`, etc) are machine specific, so they will not be defined in the controller_config.

Hywelmartin, 2020-11-01:
I don't see the added complexity render any real gains..

mtw3d, 2020-11-02:
@Hywelmartin What added complexity do you perceive? For the most people, this will make things simpler, not more complex. 

For end-users, it won't make a huge difference unless they ever change their electronics, then it will be a huge benefit. 

For people creating a board config file, it will greatly simplify the process by separating most machine config from the controller config. 

And for people creating a machine config, it is simpler because (if their controller already exists) they only need to define the machine specific configuration. 

I can see how this seems complex at first, but the only actual complexity is adds is an initial conversion process. That certainly is not nothing, but I suspect it will have a big return on whatever time we invest in it.

yschroeder, 2020-11-02:
Be aware, that you might end up in an "include hell", where it will become hard for a (new) user to understand from where a magical definition/setting is included or why his printer is doing things it does (because some setting is hidden in 3 layers of file includes).

Every additional abstraction between the board hardware and the klipper configuration makes it harder to track down what happens.

I think from a new user perspective 

```
[stepper_x] 
step_pin: PB13
```

is much easier to understand than

```
[include %config/pins-rambo.cfg]

[stepper_x] 
step_pin: XSTEP
```

Also: How often does the average user switch the board while he wants all of his config to stay intact by only changing the include with pin definitions at the top?

If I was to change my printer board, I would locate the example config for the new one and would change the (rather few) occurances of pin definitions. Most of the Klipper config when using a lot of its features are not related to pins anyway.

vladbabii, 2020-11-02:
What if we build a script to generate the files, based on existing configs, instead of doing it by hand? Most of the pins are pretty self-explanatory and could be converted directly, we can fill any extra ones by hand and have someone else double check...

@yschroeder the main benefit would be portable configs, because if i make a nice macro with some specific pins, then someone else has to re-map those instead of just include and restart. Board changes don't happen pretty often, but quite a lot of people have the same printer with different boards so for them it would be use config for printer X and include board Y.

That being said, from my point of view
1. a pin must be able to have multiple aliases (the official one like "xstep" and maybe a second and third one that matches what's written on the board or board manual
2. the include should be in the [mcu x] section, so with multiple boards x:xstep will work fine. replacing a board would be changing from [mcu x] with aliases:board/x.cfg to [mcu x] with aliases: board/y.cfg

KevinOConnor, 2020-11-02:
I have some random comments.  (Since I'm not planning to do much work on this, you shouldn't take what I say too seriously - I'm open to alternatives.)

> Can a give pin be given more than one alias?

Yes, but I'd say that should be limited as it can quickly become confusing.

> For example, on the RAMBo, we use a pin on a header block as the control pin for the BLTouch, but someone else might want to use that header block for some other purpose.

FWIW, I'd say the only time a pins-XYZ.cfg file should have a BLTOUCH_SENSOR type alias is if the board itself has a port named "BLTouch".  If someone is using a random pin for a feature then it's not possible to provide meaningful pin guidance - the user will have to make the choice for themselves and configure accordingly.

> The goal here is to create common names for all commonly used pins, so if a user wants to switch from controller_x to controller_y, as few changes to their config as possible will be required.

FWIW, I don't think this goal is achievable, so I'd recommend against attempting it.  If a user changes boards they will have to carefully review every declaration in their config.  I'd recommend the goal be to make it easier to configure a printer in general.

> Although Klipper is not case-sensitive, Pin Names should always be cited in upper case for ease of legibility.

FYI, Klipper pin names are case sensitive.  I agree it would make sense to always use uppercase for standard aliases.

> with the convention that it should be as short as possible while remaining clear

FWIW, I would prefer longer names.  For example, X_STEP, BLTOUCH_SENSOR, BED_THERMAL_SENSOR, etc.

> This does not include pins that are just broken out to pin headers

FWIW, I think it would be helpful to add all the ports of a board in a pins-XYZ.cfg file - as it is a real pain to find the schematic and lookup the pins when wiring something.

-Kevin

KevinOConnor, 2020-11-02:
> Be aware, that you might end up in an "include hell", where it will become hard for a (new) user to understand from where a magical definition/setting is included or why his printer is doing things it does (because some setting is hidden in 3 layers of file includes).

That is certainly a concern of mine.  I'd say it could be alleviated by adopting a convention where a `pins-XYZ.cfg` file never itself includes any other file (nor depends on definitions from any other file) and never defines an alias that depends on another alias.  If that convention is followed, I'd guess it would not be too daunting for users.

> I don't see the added complexity render any real gains..

This is also a concern of mine.  I do, however, see people struggle with the pin definitions and I wonder if it could be improved.

In particular, it can be a real pain to find a useful schematic for some boards, and when one does finally track it down it's likely a lot easier to define every pin in that schematic in a single location all at once.

I also see that many of the example configs have a lot of overlap that isn't necessarily clear to users.  For a random example, the `printer-lulzbot-taz6-2017.cfg` is really a rambo board, but it doesn't have the lcd header pins defined, so if someone had one of those printers and wanted to hookup a different LCD they'd be back at reading schematics.  Having more consistency here could be a benefit.

-Kevin

KevinOConnor, 2020-11-02:
> If I was to change my printer board, I would locate the example config for the new one and would change the (rather few) occurances of pin definitions. Most of the Klipper config when using a lot of its features are not related to pins anyway.

I agree (and I also agree with your other comments).

FWIW, though, I have found that defining standard names for the LCD pins (ie, EXP1_X and EXP2_X) has been very helpful.  There was a lot of confusion around these pins before.  So, I do wonder if there will be a similar advantage to using some standardized aliases for other pins as well.

-Kevin

Hywelmartin, 2020-11-02:
> In particular, it can be a real pain to find a useful schematic for some boards, and when one does finally track it down it's likely a lot easier to define every pin in that schematic in a single location all at once.

A link in every xyz.cfg for a decent pinout and/ or all named pins on the board somehow documented.. would be nice

vladbabii, 2020-11-02:
Why not have the "standard" aliases like x_step (random example) be defaults for stepper x step_pin ? This would cut down even more on the config file

mtw3d, 2020-11-02:
> Be aware, that you might end up in an "include hell", where it will become hard for a (new) user to understand from where a magical definition/setting is included or why his printer is doing things it does (because some setting is hidden in 3 layers of file includes).

I don't see this as a problem if people are using it as I envision it. This should *only* define the controller. I don't see any reason why anyone would want any includes in this file, it kind of defeats the purpose of the whole intention. 

For example, the printers that my company sells mostly use the Ultimachine Rambo. We use this board on several different machines with several different specs. The same board is also used on most of Lulzbot's machines, as well as many other 3d printer manufacturers. The Rambo settings will be identical, regardless of which of these machines the end user has, so why not put them into a specific file? FWIW, this is *basically* what Marlin does with their pins files, but by moving things like board-specific stepper configs into the same file, it makes it even more modular.

> I think from a new user perspective

>is much easier to understand than

I don't agree. The average new user just doesn't care about the specific pins, so "intuitiveness" of the pin names themselves isn't really an important goal. What is important is intuitiveness of setup, and I think this way is much simpler, both for the end user, and for anyone creating the configuration file. 

>Also: How often does the average user switch the board while he wants all of his config to stay intact by only changing the include with pin definitions at the top?

Not often, but wouldn't it be nice to make it as easy as possible, even if it isn't common? 

I admit my perspective on this is different. I sell printers, and I want to make it easy for my customers to update their printers. By doing this I can have a single configuration for my printers, and a separate configuration for the controllers. If a customer wants to update from a Rambo to Ultimachine's newer Archim, for example, they can just swap out the controller configuration file. From a customer support perspective, this is a massive benefit, since I only need one config for each printer, and one config for each controller. Then the customer can just choose the correct files for their needs and they're up and running.

But I proposed it here, rather than just implementing my own printer configurations this way, because I think it is a significantly simpler way to handle the configs.

mtw3d, 2020-11-02:
> Yes, but I'd say that should be limited as it can quickly become confusing.

I agree, it's not something I would do often. 

>FWIW, I'd say the only time a pins-XYZ.cfg file should have a BLTOUCH_SENSOR type alias is if the board itself has a port named "BLTouch".

Ok, this is reasonable. 

> FWIW, I don't think this goal is achievable, so I'd recommend against attempting it. If a user changes boards they will have to carefully review every declaration in their config. I'd recommend the goal be to make it easier to configure a printer in general.

I am mainly focused on the most common pins. We certainly cannot get every pin named, but we can get the most common ones that account for most basic tasks. 

> FYI, Klipper pin names are case sensitive. I agree it would make sense to always use uppercase for standard aliases.

Ah, my mistake. I assumed they were like Klipper commands, but that makes sense. 

That's fine... I prefer brevity, but not a big deal. 

> FWIW, I think it would be helpful to add all the ports of a board in a pins-XYZ.cfg file - as it is a real pain to find the schematic and lookup the pins when wiring something.

How do you define "ports" here? Consider the Ultimachine Rambo shown at the following link:

https://cdn.shopify.com/s/files/1/1223/5992/products/RAMBoV13L_1024x1024.jpg?v=1498583046

To me, all the pins that are commonly used should be defined. But if you look at that board, he breaks out *all* the pins to various headers that can be used for various alternate purposes. But 95% of users don't use any of them. It would add probably an hour of work to define all of them, for something that few users would benefit from. Basically what I would do there is define all the common ports (which all have shrouded connectors on the Rambo), but leave the others up to the user if they need them.

It's certainly an option, but just seems like an unnecessary extra complication for very little benefit. We could add a link to the board schematic or whatever docs have the pin breakouts for anyone needing to go deeper.

yschroeder, 2020-11-02:
What I could get behind is something like this: the klipper repository provides example configs for boards and separate example configs for printers. 

When a users wants to set up his machine he will copy two of these files to his home directory, the printer config will be the main config and the board config which will be included in the main config. The printer config has to be renamed to klipper.cfg and the board config has to be renamed to board.cfg (or similar). The example printer configs always include the file board.cfg with a comment next to it defining which board to use. 

That way the user learns that a full config consists of two files and he has both of them at hand to look into them and learn what they do. 

Other than that, I think the default aliases provided above seem valid. However, I would refrain from introducing multiple aliases for the same pin, except maybe the one marked in the silkscreen next to the physical pin on the board. 

As Kevin said, do not impose any defaults onto a user e.g. for bltouch. Maybe add a hint to the board file where it would be commonly connected in the form of commented out aliases. If the board has dedicated bltouch pins though, it is fine to create an alias for them.

mtw3d, 2020-11-02:
> I also see that many of the example configs have a lot of overlap that isn't necessarily clear to users. For a random example, the `printer-lulzbot-taz6-2017.cfg` is really a rambo board, but it [...]

This is *exactly* the sort of issue I am trying to prevent.

mtw3d, 2020-11-02:
> What I could get behind is something like this: the klipper repository provides example configs for boards and separate example configs for printers.
>
> When a users wants to set up his machine he [...]

This is exactly as I see it being implemented. This has the added benefit of working as is with no code changes. Just as the do now, they create their `printer.cfg`, but they also create their `controller.cfg` and populate both with the included example files. As long as the printer.cfg has `[include controller.cfg]`, this just works with no customization required.

KevinOConnor, 2020-11-05:
> Basically what I would do there is define all the common ports (which all have shrouded connectors on the Rambo), but leave the others up to the user if they need them.

Okay.

> When a users wants to set up his machine he will copy two of these files to his home directory

FWIW, it's hard enough to get new users to setup one config file.  I don't think it makes sense to ask new users to copy two files.  (That's why I mentioned `[include %config/pin-rambo.cfg]` as a possibility, as that wouldn't require users to copy additional files.)

Cheers,
-Kevin

Sineos, 2020-11-05:
Since my first time with Klipper are not too long ago (didn't hurt that much):
At the time I switched from Marlin and the original Ender 5 board to Klipper and updated the board to a BTT SKR 1.4. This meant that I had to compare 2 thankfully provided cfg-files and merge the information into one. Here the proposed system would have come in handy and reduced the efforts.

With some experience I have to say that I very much appreciate Klipper's approach of keeping a well structured and concise cfg file with easy access over respective Octoprint plugin. 
Factoring out common board specific pins into an include like Kevin mentioned is a one timer and does not change much in the subsequent way of working WRT optimizing / extending the rest of the configuration.

mtw3d, 2020-11-05:
> FWIW, it's hard enough to get new users to setup one config file. I don't think it makes sense to ask new users to copy two files. 

I certainly understand this, but I don't really think this is a *major* issue. Getting them to copy the one file is the hard part, getting them to copy the second one won't be twice as hard. 

That said, I agree that your suggestion is better (with caveats), but to make it robust, it would require some code changes I think, which I wanted to avoid.

> (That's why I mentioned `[include %config/pin-rambo.cfg]` as a possibility, as that wouldn't require users to copy additional files.)

I have two concerns with this method. First is the potential for breaking changes if the file were ever updated. One careless edit by a user on the other side of the planet, and a commit at the wrong time, just when I pull an update, and my printer stops working for reasons that I have no understanding of. This *should* never happen, but it seems likely that it will, sooner or later.

This can be fixed if `[include %config/pin-rambo.cfg]` automatically created a copy of the config,  but only if it didn't already exist. Ideally this would also include a gcode command to force it to pull the latest config version, but I could live without that, since the file should only rarely be updated. 

The other issue I see is how to set up multiple MCUs using this mechanism. If you added something like `[include as_mcu=MCU2 %config/pin-rambo.cfg]` that would handle the pin definitions, but it would break on any driver setup blocks that were in that file, which sort of defeats the purpose of the whole mechanism. 

I see two workarounds, one is easy enough but basically ignores this mechanism, and one is very elegant, but likely requires significant coding. 

1. (Easy enough) Manually create the second config as a normal config, edit the necessary blocks, and include just as a normal file.  
2. (Elegant) Make it so that any blocks defined in a file with `as_mcu=MCU2` all apply to board MCU2. So `board_pins` would apply, but also if:

```
   [tmc2209 stepper_y]
   uart_pin: PC11
   tx_pin: PC10
```

were included in that file, it would be interpretted as:

```
   [tmc2209 stepper_y]
   uart_pin: MCU2:PC11
   tx_pin: MCU2:PC10
```

I have no idea how hard this would be to implement, so I am not sure what the best solution is. Obviously  the second is ideal, but the first is workable. Since it will tend to be more advanced users who require multiple MCUs I think it's reasonable solution. 

But really, to me this seems to be opening a whole can of worms that is avoided by just manually copying the config files and editing as necessary. It would be really nice if this stuff worked as I see it, but I have no idea is the actual hassle saved probably is enough to justify the coding.

mtw3d, 2020-11-05:
> With some experience I have to say that I very much appreciate Klipper's approach of keeping a well structured and concise cfg file with easy access over respective Octoprint plugin.
> Factoring out [...]

This does have the downside of the controller config not being editable in the the Octoprint pugin's edit Config window. But that shouldn't be a major issue for most people. The settings in the controller config are things that most users will rarely edit. Mainly the microsteps required, and the stepper current. These tend to be set & forget for most users, at least after an initial setup phase.

mtw3d, 2020-11-28:
Sorry, I have been slammed with other stuff. I do plan to work on this more, but my schedule likely won't let me make major contributions before the end of the year. I don't necessarily need the issue to remain open to work on it, so you can leave it open or close it as you see fit.

Thanks for the change in #3559, that looks like it will be a big help.

KevinOConnor, 2020-12-23:
FYI, I made some comments on this at PR #3671 .

-Keivn

mtw3d, 2020-12-24:
Responding to the comments in that PR...

> Ultimately, I'm unsure if converting to this scheme is a good idea. It does make some things nicer - it's easier to compare configs between similar printers with different electronics. But, some things don't seem to work so well - I fear it wont be immediately apparent to new users what is and isn't an alias, and the block of alias definitions at the end of every config looks pretty intimidating.

If we do what I see as recommended-- make an alias for any pin that is broken out to a purpose-named header (For example motors, thermistors, fans, etc)-- the vast majority of new users will never need to use a pin that doesn't have an alias. By definition, anyone doing anything that is using a general-purpose header is doing more of a deep-dive into both the firmware and the electronics, so I don't see having it slightly more complicated as a big downside, compared to making it easier for nearly everyone else. 

> One thing interesting that I found is that a large number of common printers use very similar boards, but often do have subtle differences. The atmega1284p melzi boards are an example of this - basically every printer that uses the atmega1284p is using a melzi derivative, but some define different motor enable pins and many have different exp1/exp2 plugs.

This doesn't seem like an issue with this plan, though. It seems like the problem exists whether we change or not. The non-standard boards are still out there regardless of the naming scheme. Short of just not calling on those pins in the default config file (which we can do with this plan by not aliasing any non-standard pins), it is unavoidable.

It seems like the best solution to the issue is just adding comments to the config file. I don't see an easy way around that.

KevinOConnor, 2020-12-27:
FYI, I made some updates to the comments and config section ordering in PR #3671 .

> If we do what I see as recommended

FYI, if you are recommending a different config layout then it's probably best to make those changes to a bunch of configs, and then upload a PR with it.  I think having some examples will help with discussions.

Cheers,
-Kevin

closed 2021-01-24 by github-actions[bot]