Klipper3d/klipper #3487: BLTOUCH Raising plate too high on right side

gitovillegas, 2020-10-31:
Hi!

I'm having some trouble printing on a newly klippered ender 5+
It seems to be moving the nozzle too close to the z axis specifically on the right side.
I have tried re leveling my bed, cleraning the mesh, making new mesh, changing z offset etcetc
I usually probe with the bed heated up

Attached is the config. Any help is apprectiated!
[image]

[klipper config e5p.txt](https://github.com/KevinOConnor/klipper/files/5469468/klipper.config.e5p.txt)

gitovillegas, 2020-10-31:
attaching klippy log
[klippy.log](https://github.com/KevinOConnor/klipper/files/5469506/klippy.log)

KevinOConnor, 2020-11-09:
Alas, the best info we have on the bl-touch and bed mesh is in the documentation: https://www.klipper3d.org/Overview.html#configuration-and-tuning-guides

-Kevin

closed 2020-12-08 by github-actions[bot]