Klipper3d/klipper #3489: creality 4.2.7 board will not work at all.

jerkwagon, 2020-10-31:
so this board has been  giving me a huge headache i highly doubt klipper is even worth it at this point, over 15 hours of troubleshooting. thats $1500 of my time.  so i would say no..

fresh install of octopi-buster-lite-0.17.0 .zip
ran the setup from IP address. 
after setup,  ran the following in terminal. 

git clone https://github.com/KevinOConnor/klipper
./klipper/scripts/install-octopi.sh

cd ~/klipper/
make menuconfig

select the STM32F103 with
a "28KiB bootloader" and with
"Use USB for communication" disabled.

make

copied the the klipper.bin to sd, turned on the printer with SD, (even tried with a new name.bin) no idea if the firmware was pushed, i was told it should create a .DAT file, but it doesnt, only when i flash the offical firmware from creality. 

here is where the fun stuff starts, 

open up octoprint web.  

followed the directions 

_Navigate to the Settings tab (the wrench icon at the top of the page). Under “Serial Connection” in “Additional serial ports” add “/tmp/printer”. Then click “Save”.

Enter the Settings tab again and under “Serial Connection” change the “Serial Port” setting to “/tmp/printer”.

In the Settings tab, navigate to the “Behavior” sub-tab and select the “Cancel any ongoing prints but stay connected to the printer” option. Click “Save”.

From the main page, under the “Connection” section (at the top left of the page) make sure the “Serial Port” is set to “/tmp/printer” and click “Connect”. (If “/tmp/printer” is not an available selection then try reloading the page.)

_

now i though i got somewhere when the octoprint said, 
_State: Operational_

but heres thw dumb part, even with the printer unplugged i can get that message. 

this is the log from klipper. 

**9:07:21 Klipper state: Disconnect
19:10:23 Connected to host via /tmp/printer @115200bps
19:10:23 Firmware version: v0.9.1-22-g1de7f395
19:10:32 Unable to connect
Once the underlying issue is corrected, use the
"FIRMWARE_RESTART" command to reset the firmware, reload the
config, and restart the host software.
Error configuring printer
19:10:32 Unable to connect**

this is the log from the terminal 

**Recv: // Unable to connect
Recv: // Once the underlying issue is corrected, use the
Recv: // "FIRMWARE_RESTART" command to reset the firmware, reload the
Recv: // config, and restart the host software.
Recv: // Error configuring printer
Recv: !! Unable to connect**

hitting host/firmware does nothing.

this is the status with no printer connected via usb. 

_Changing monitoring state from "Offline" to "Opening serial connection"
Connecting to port /tmp/printer, baudrate 115200
Changing monitoring state from "Opening serial connection" to "Connecting"
Connected to: Serial<id=0x6f7736b0, open=True>(port='/tmp/printer', baudrate=115200, bytesize=8, parity='N', stopbits=1, timeout=10.0, xonxoff=False, rtscts=False, dsrdtr=False), starting monitor
Send: N0 M110 N0125
Recv: ok
Send: N0 M110 N0125
Changing monitoring state from "Connecting" to "Operational"
Recv: ok
Send: N0 M110 N0125
Recv: ok
Send: N1 M11539
Recv: ok FIRMWARE_VERSION:v0.9.1-22-g1de7f395 FIRMWARE_NAME:Klipper
Send: N2 M21*18
Recv: ok_

I have no clue what is happening or how to test/trouble shoot,  i have added teh following to the MCU neither work

pi@octopi:~/klipper $ ls /dev/serial/by-id/*
/dev/serial/by-id/usb-1a86_USB_Serial-if00-port0
pi@octopi:~/klipper $ ls /dev/serial/by-path/*
/dev/serial/by-path/platform-3f980000.usb-usb-0:1.5:1.0-port0

i am ready throw this out the window and take a loss on my time and money..

[klippy.log](https://github.com/KevinOConnor/klipper/files/5469710/klippy.log)

theophile, 2020-11-01:
You have it connected by USB but you disabled communication over USB?

jerkwagon, 2020-11-01:
> You have it connected by USB but you disabled communication over USB?

i think i did it both ways, but ill double check tomorrow.

jhpadjustable, 2020-11-01:
> (even tried with a new name.bin) no idea if the firmware was pushed, i was told it should create a .DAT file, but it doesnt, only when i flash the offical firmware from creality.

The 28k ARM bootloader looks for a file named firmware.bin. If found and flashed successfully, the bootloader will then rename it to firmware.cur. Any .dat file would have been created on the card by the Marlin firmware to emulate serial EEPROM functionality, not relevant to Klipper.

jerkwagon, 2020-11-01:
@jhpadjustable i dont think thats true, as i just flashed the board with file called "firm OG.bin" which was the stock 4.2.7 firmware for BL touch.  i have yet to get klipper to give me a .dat file from flashing only a blank screen.

Sepitus, 2020-11-01:
i dont know if it'll work for you but it worked for me. I'm running a new BTT SKR 1.4 with BTT TMC2209 drivers and i had to uninstall klipper from my raspi then re-install it and make klipper.bin again to put on the SKR 1.4 SD card. It worked!

jerkwagon, 2020-11-01:
Recv: ok FIRMWARE_VERSION:v0.9.1-22-g1de7f395 FIRMWARE_NAME:Klipper

So i dont really understand, if the octoklipper says, OK firmware, it must be talking to the printer.. but im still getting the error message unable to connect

can anyone else confirm that the firmware is correct and that this firmware will creat the .dat file?

BlackStump, 2020-11-01:
@jerkwagon  - you should focus on the klippy.log when commissioning a mcu and not even attempt anything with Octoprint until the klippy.log is happy ie no errors.  Octoprint / Octoklipper is not Klipper  and Octoprint,s default settings are Not klipper friendly.

@jhpadjustable  gave you good info but you choose to disagree
>The 28k ARM bootloader looks for a file named firmware.bin. If found and flashed successfully, the bootloader will then rename it to firmware.cur. Any .dat file would have been created on the card by the Marlin firmware to emulate serial EEPROM functionality, not relevant to Klipper.

I do not know that mcu , and I suspect that is the same for a lot of other volunteer's that give there time to help others like yourself. If people that are using that mcu successfully are not willing to chime in then you are left with generic info.

jhpadjustable, 2020-11-01:
Er, I actually don't have experience with that board, just the micro that's on it. They appear to have modified the usual 28k bootloader to load any *.bin file (Teaching Tech video confirms). It's possible they also customized the renaming behavior.

jerkwagon, 2020-11-02:
I have figured it out,  turns out you need to enable extra low level but do not select anything. i litterally spent over 20 hours trying to figure this out, along the way i have been lucky to have helpful people i am grateful for.   

I used the link kevin sent, then added the BL and LCD to it, i dont have th BL touch working yet, but i am really close to it.

linux-paul, 2020-11-06:
Aren't there TMC2225 on it?

OXERY, 2020-11-10:
Here is my configuration from an Ender 3, which I upgraded using Creality's 4.2.7 Board.
It seemed to work, but at a test print (CHEPs Level Test) it tried to move Z < 0 and the print was stopped. I think it has something to do with the probing speed. Is there a way lowering it? And to also level it a second, slower, time?

`# This file contains pin mappings for the Creality "v4.2.7" board. To
# use this config, during "make menuconfig" select the STM32F103 with
# a "28KiB bootloader" and with "Use USB for communication" disabled.

# If you prefer a direct serial connection, in "make menuconfig"
# select "Enable extra low-level configuration options" and select the
# USART3 serial port, which is broken out on the 10 pin IDC cable used
# for the LCD module as follows:
# 3: Tx, 4: Rx, 9: GND, 10: VCC

# Flash this firmware by copying "out/klipper.bin" to a SD card and
# turning on the printer with the card inserted. The firmware
# filename must end in ".bin" and must not match the last filename
# that was flashed.

# See the example.cfg file for a description of available parameters.
[bltouch]
sensor_pin: ^PB1
control_pin: PB0
pin_up_touch_mode_reports_triggered: False
probe_with_touch_mode: True
x_offset: -48
y_offset: 10
pin_move_time: 0.4

[safe_z_home]
home_xy_position: 155,120
speed: 80
z_hop: 10
z_hop_speed: 10

[bed_mesh]
speed: 50
horizontal_move_z: 5
mesh_min: 32, 60  # add the x and y offsets for both mesh_min and max. This is set for mine.
mesh_max: 172, 210
probe_count: 3,3
mesh_pps: 2,2

[display]
lcd_type: st7920
cs_pin: PB12
sclk_pin: PB13
sid_pin: PB15
encoder_pins: ^PB14, ^PB10
click_pin: ^!PB2

[gcode_macro G29]
gcode:
    BED_MESH_CALIBRATE
    BED_MESH_OUTPUT

[stepper_x]
step_pin: PB9
dir_pin: PC2
enable_pin: !PC3
step_distance: .0125
endstop_pin: ^PA5
position_endstop: 0
position_max: 235
homing_speed: 50

[stepper_y]
step_pin: PB7
dir_pin: PB8
enable_pin: !PC3
step_distance: .0125
endstop_pin: ^PA6
position_endstop: 0
position_max: 235
homing_speed: 50

[stepper_z]
step_pin: PB5
dir_pin: !PB6
enable_pin: !PC3
step_distance: .0025
endstop_pin: probe:z_virtual_endstop
position_endstop: 0.0
position_max: 250

[extruder]
max_extrude_only_distance: 100.0
step_pin: PB3
dir_pin: PB4
enable_pin: !PC3
step_distance: 0.010526
nozzle_diameter: 0.400
filament_diameter: 1.750
heater_pin: PA1
sensor_type: EPCOS 100K B57560G104F
sensor_pin: PC5
control: pid
pid_Kp: 21.527
pid_Ki: 1.063
pid_Kd: 108.982
min_temp: 0
max_temp: 250

[heater_bed]
heater_pin: PA2
sensor_type: EPCOS 100K B57560G104F
sensor_pin: PC4
control: pid
pid_Kp: 54.027
pid_Ki: 0.770
pid_Kd: 948.182
min_temp: 0
max_temp: 130

[fan]
pin: PA0

[mcu]
serial: /dev/serial/by-id/usb-1a86_USB_Serial-if00-port0
restart_method: command

[printer]
kinematics: cartesian
max_velocity: 300
max_accel: 3000
max_z_velocity: 5
max_z_accel: 100

#*# <---------------------- SAVE_CONFIG ---------------------->
#*# DO NOT EDIT THIS BLOCK OR BELOW. The contents are auto-generated.
#*#
#*# [bltouch]
#*# z_offset = 2.400
`

1N4148, 2020-11-10:
The default position_min is 0 if not specified

https://github.com/KevinOConnor/klipper/blob/afbed5d4c71fbd7465b970169be0b54649f7c20e/config/example.cfg#L41

OXERY, 2020-11-11:
> The default position_min is 0 if not specified
> 
> https://github.com/KevinOConnor/klipper/blob/afbed5d4c71fbd7465b970169be0b54649f7c20e/config/example.cfg#L41

Could you provide more context please?

1N4148, 2020-11-11:
If you don't specify the position_min per axis it's set to 0 by default. This means your axis cannot go beyond 0. To have your z-axis go below 0 you'll have to specify e.g. position_min: -1 for [stepper_z]

OXERY, 2020-11-11:
Thx, no I understand. In Marlin/TH3D the limits only have to be deactivated once while adjusting Z offset, therefore I did not really get the reason. Regarding BLTouch I noticed that there is no possibility to use the sensor via the Z-Switch. These two wires also have to be connected to the dedicated port. This makes changing firmware between Marlin/Klipper a little complex as the housing has to be opened each time..

1N4148, 2020-11-11:
Just change the pin defined in

[bltouch]
sensor_pin: ^PB1 (<- see your hw documentation for the correspondig port name)

to the pin where your BLTouch switch is connected to. That's the charme of Klipper: You can adapt the sw config to your hw config and restart. No need to rewire, no need to recompile & flash. 

My BLTouch sensors are connected to the legacy Z-switch port.

OXERY, 2020-11-12:
Thx, it worked. Changed sensor_pin to ^PA7

closed 2020-12-11 by github-actions[bot]