Klipper3d/klipper #3491: Is it possible to use Z-stop as a safety mechanism only, and not for levelling?

marcosscriven, 2020-11-01:
[klippy.log](https://github.com/KevinOConnor/klipper/files/5470763/klippy.log)

Now that https://github.com/KevinOConnor/klipper/issues/3453 is fixed, you'll see I've got the bltouch working for homing and levelling. It seems to be working well in practice too, with a better first layer. 

However, I'd like to be able to still use the z-endstop switch for two reasons:

1. As a safety mechanism/backup - in case something goes wrong homing z with the bltouch. The z-stop can be setup far enough below the lowest part of the bed.
2. So a 'rough' z=0 can be temporarily set, so that 'safe home' doesn't need to arbitrarily raise, not knowing if it's already at the maximum extent.

I've seen there's a `relative_reference_index`, however, that then seem to:

1. Limit the overall bed level accuracy back to the physical switch, when the bltouch is more accurate
2. Override the z-offset, which in my case seems to be pretty stable.

Hywelmartin, 2020-11-02:
https://github.com/KevinOConnor/klipper/blob/e0842e0e03b29dce260bf9e519d08f48f7e5ace7/config/example-extras.cfg#L1385

marcosscriven, 2020-11-02:
Thanks @Hywelmartin - that almost looks like what I want, but I’m concerned going via gcode is too high level. 

If a homing command is incomplete, but is causing the head to crash, would that not then prevent any motor stop gcode from being enacted, as it would be stuck in the buffer waiting?

jakep82, 2020-11-03:
You could simply wire the end stop switch to the reset pin on your controller board if it has one.

KevinOConnor, 2020-11-09:
This looks like a dup of #2942 .

-Kevin

closed 2020-12-08 by github-actions[bot]