Klipper3d/klipper #3492: Host Buffer 100% Failed Print, hotend PID integrator windup, firmware retract disregards retract speed

DroneMang, 2020-11-02:
Hi,

Was printing this:
[Part3-1.zip](https://github.com/KevinOConnor/klipper/files/5472484/Part3-1.zip)

And the Host Buffer is going to 100% multiple times.  When it does this, the print head stops, leaves a blob on the print, and moves on, problem is that when the nozzle comes back to that part I get a layer shift and missed steps when it hits the blob.  I tried it with the new arc welder plugin, and with a regular gcode file, both fail in the same way.  I believe the first part of the log file is a failed print for the arcwelder print, and the second part is just the regular gcode.  I am using S3D slicer.

Also, when my cooling fan turns on after the first 3 layers, the print fails due to hotend temp dropping too much. I have set the timeout higher to help this. I think the integrator in the PID has wound up and so it takes it a long time to adjust to the new setpoint. Perhaps you could reset the PID but keep same calibration values if the cooling fan speed changes? I have a large bed with the FT-6 so large prints cause this error.  Also I have a 50W 24V heater cartridge, so the PID values may not be what you would expect for a standard cartridge, I have run autotune.

I am using firmware retract and pressure advance. With a volcano hotend I found that setting the retract and unretract speeds very high (I have had 600 mm/s set in marlin) virtually eliminated ooze or gaps when starting a new line as the hotend moves immediately to the new point. I have tried setting high retract and unretract speeds in klipper, but as far as I can tell, no change to speed happens. Is there a max speed that is allowed?

Thanks!
[klippy7.zip](https://github.com/KevinOConnor/klipper/files/5472481/klippy7.zip)

jakep82, 2020-11-02:
Host buffer issues are usually caused by Octoprint. The only solution is using virtual SD which bypasses the gcode reader in Octoprint, but you'll lose some features.

https://github.com/OctoPrint/OctoPrint/issues/450

DroneMang, 2020-11-02:
OK. I'm trying buffer_time_high=15 for now, seems to be working, I'll also try the virtual SD card.

Can someone look into the PID?  I get this huge dip in temp when the cooling fan turns on.  Thanks!
[image]

jakep82, 2020-11-02:
Some [discussion](https://github.com/KevinOConnor/klipper/pull/3382) has happened regarding a look ahead option for the extruder, but the solution right now is to insulate your heater block.

DroneMang, 2020-11-02:
Well the print isn't failing with buffer_time_high=15, but pause resume takes forever. I think I could use a smaller value. Any idea how to determine the smallest value that doesn't fail?

DroneMang, 2020-11-02:
When I do an M20 to list the contents of the SD Card, there are no files there, I am using the below in the cfg, I have all my stls in folders in octoprint, is that why nothing shows?

[virtual_sdcard]
path: ~/.octoprint/uploads/

closed 2020-12-01 by DroneMang