Klipper3d/klipper #3494: I can't get this to work with my Qidi X-one 2 Printer.

clinde04, 2020-11-02:
Is there any known way to make klippy work with a Qidi tech X-one 2 printer? I have done all of the install instructions with no success.

jakep82, 2020-11-02:
I have this printer, and the control board has an encrypted bootloader.  I've seen reports of success cracking the bootloader on the Marlin github, but it wasn't worth the hassle for me to try and salvage a mediocre board with loud A4988 drivers, so I replaced it with something open source.  Maybe someone else will have more info, but as of right now I don't think you can run Klipper on this printer with the original control board.

https://github.com/MarlinFirmware/Marlin/issues/7264

clinde04, 2020-11-02:
Ok that sucks. I will try to find out if it is possible to crack the bootloader. Thanks for the fast response though!

KevinOConnor, 2020-11-02:
FYI, @cab404 did add support for the Chitu bootloader ( #3265 )  .  However, I don't have any knowledge on the Qidi X-one.

-Kevin

jakep82, 2020-11-02:
Glad to see Chitu support was added recently.  I hadn't looked into since I swapped the board in my printer about 6 months ago.

clinde04, 2020-11-02:
Ok, so I just edited my menuconfig and set the bootloader to the chitu v6 one (34kib). So when I do make flash FLASH_DEVICE=(my device thing from ls /dev/serial/by-id/*) it errors out saying no DFU capable USB devices found.

jakep82, 2020-11-02:
Try following the directions at the top of [this config file](https://github.com/KevinOConnor/klipper/blob/master/config/printer-tronxy-x5sa-v6-2019.cfg).  The pins on that config might be a good place to start, but you'll probably be spending some time working on the config since one doesn't currently exist for this printer.

clinde04, 2020-11-02:
Ok so I did that. The screen is currently black but it did the beep thing and I waited for it to stop beeping before restarting it with the sd card out. Do I simply have to update the pins or something for the screen to work or whats up with that? I'm about to see if I can print anything soon.

clinde04, 2020-11-02:
I get printer errors from octoprint. They say the following:

 Unable to connect
Once the underlying issue is corrected, use the
"FIRMWARE_RESTART" command to reset the firmware, reload the
config, and restart the host software.
Error configuring printer

jakep82, 2020-11-02:
The touch screen isn't supported and won't work.  

You'll need to post a log file for any more help, and you'll need to create your own config file.  Nobody has used Klipper on this exact printer and control board before.  Marlin lists pins for 4 different Chitu boards at the link below.  I don't know exactly which configuration will work (if any of them).  Try starting with a bare minimum config file that only contains X,Y, and Z and see if you can get them to move.

https://github.com/MarlinFirmware/Marlin/tree/2.0.x/Marlin/src/pins/stm32f1

clinde04, 2020-11-02:
ok thank you so much. will do!

clinde04, 2020-11-02:
Ok..... well now I tried to flash marlin on the printer and it just flashed and went dark and now only a soft fan blows in the printer... The steps I took are:

I used platformio and vscode and set the configuration.h and platformio.ini stuff all to point to the chitu v6 board. Then I built it with marlin auto build in vscode and then it gave me the update.cbd file. I put that on the SD card and put it into the printer then turned it on. It beeped 4 times then went dark and now only blows a fan softley.
[Configuration_adv.h.txt](https://github.com/KevinOConnor/klipper/files/5478102/Configuration_adv.h.txt)
[Configuration.h.txt](https://github.com/KevinOConnor/klipper/files/5478103/Configuration.h.txt)
Those are my config files and this is my platformio file: 
[platformio.ini.txt](https://github.com/KevinOConnor/klipper/files/5478109/platformio.ini.txt)

jakep82, 2020-11-02:
You'll need to ask Marlin questions on the Marlin Github.  I pointed you to the Marlin pins files because they might be useful in creating a Klipper config for this printer.  I'm not even sure which version of the Chitu board is in this printer, so you might not be able to make it work with anything but stock firmware.

clinde04, 2020-11-02:
Ok, i'll keep trying

closed 2020-11-02 by clinde04

bvaerewyck, 2021-09-22:
I got pretty frustrated with trying to set up my Qidi X-One2 with Octopi and I can't imagine Clipper.  How did it turn out?
 There are several versions of the Qidi X-One2 printer. The blue version and the red and blue version. I happened to have the red&blue with just the USB2-Type A Storage Slot.

I have also found they have three versions of motherboards as follows: (Current Motherboard) V4.8 with the USB Memory Stick - No UART pins. (Small Motherboard) V5.3 with the USB Memory Stick and UART pins. (Original Motherboard) V7.4 with the SD Card Reader with USB2-Type B port.

So I guessing you have either V7.4 using the USB2-Type B port or the V5.3 with the UART pins. Took a few days to see everyone posting that they could use the Qidi X-One2 with Octopi and that some people are frustrated that they cant find the UART pins or the board is just a blank on that part of the board. The V5.3 board is on Ebay from the manufacturer for those that are wanting to connect it their network. It is a pity that the WiFi pin headers are non functional. I also enjoy the Qidi support.

BlockArchitech, 2022-01-24:
So, I agree with @bvaerewyck 

This printer sucks to flash anything on. Personally, I have the 'easy' v7.4 board variant, but it's been a pain getting a working config going. I've tried @clinde04's config, however, it fails to work on the latest version of marlin. I tried to make my own, but it's pretty messed up.

Hopefully, somebody manages to get clipper on here because it's a pretty solid printer in my opinion.

bvaerewyck, 2022-01-24:
I got a board with uart pins off EBay.  No issues.  Connects to Octoprint.