Klipper3d/klipper #3496: Support for load cell based probe

mhier, 2020-11-02:
I am in the process of writing the support for my 3D printer (Renkforce RF1000) which has a special type of bed level probes based on a load cell to measure the force when the nozzle touches the build plate. I am wondering what the best approach is to integrate this. I find the modular approach of the Klipper firmware very appealing and I would like to embrace it as much as possible. This is what I got so far:

I have written a module to support the I2C-connected ADC for measuring the load cell signal:
https://github.com/RF1000community/klipper/blob/master/klippy/extras/ads1100.py

This already works, I can add the ADC as pin for a a fake temperature sensor to get it into the system. Then I see the debug log messages printing, and the query adc command works as well.

Next I was thinking to write another implementation of the PrinterProbe class and register it as a "probe" object, so it is found by the ProbePointsHelper. That part (finding the object) also seems to work.

Now I wonder I best interface the new PrinterProbe implementation with the ADC module without breaking the abstraction. My idea was to register a callback function (via setup_adc_callback) to retrieve new samples when arriving and process them inside the run_probe() function. This doesn't seem to work though, because the callback function is no longer serviced while staying inside run_probe() - presumably because both would run in the same thread.

Is there another generic method to obtain the ADC samples (preferably with a timing based on the ADC sampling time)? If possible I would like to avoid introducing a private interface between the new PrinterProbe implementation class and the ADC module, because this would prevent to use another ADC type with the probe implementation.

PS: I am intending to send pull requests with the printer support once it is "complete". Is there any policy how do to this? Individual pull requests for each feature/module, or better one big pull request with everything? The printer support also involves a new stepper driver module which will even require modifications to the SPI code (it expects an inverted CS signal... so far I just hacked this...).

KevinOConnor, 2020-11-03:
Interesting.  FYI, there was a discussion on load cells a few months ago (see #2977).

> I can add the ADC as pin for a a fake temperature sensor to get it into the system.

FYI, it isn't necessary to route the data back through the system as an ADC/temperature input.  You can just directly register the reactor timer, query the i2c values, and then handle the results.  Doing so will simplify the code and introduce less issues with blocking.

> Now I wonder I best interface the new PrinterProbe implementation

That is going to be some work.  The current probe system is based on the idea that the micro-controller can directly stop the steppers on an endstop/probe signal.  That's important for homing because we don't want any mcu<->host communication delays to result in the head crashing into the bed.

So, if you want to implement a similar system it would require implementing all the sensor reading and interpretation in the micro-controller C code.  Getting reliable timing over an i2c interface will also be a challenge.

In contrast, if you're not interested in homing with the sensor, and instead are just interested in doing probing, then it may be possible to do that in just the host code.  It would require moving, waiting for the move to complete, measuring the load, moving again, measuring, etc.  It would be slow, but may work okay.  In that case, you probably want to look at how klippy/extras/bltouch.py interacts with klippy/extras/probe.py - your code would also need to override how the low-level PrinterProbe._probe() code works.

-Kevin

mhier, 2020-11-03:
Thanks for the fast reply!

> Interesting. FYI, there was a discussion on load cells a few months ago (see #2977).

Indeed, thanks for the link. We have a different chip (actually two chips, one amplifier and one ADC), but the principles are the same. I take it there was no real outcome of the other thread though. I am in a bit different position: I do have a working system with a Repetier-based firmware available in source code, so I know what to do (and I have plenty ideas how to do better with the now-available computation power).

> FYI, it isn't necessary to route the data back through the system as an ADC/temperature input. You can just directly register the reactor timer, query the i2c values, and then handle the results. [...]

I know, still I like to avoid tying the module to the particular chip. This would be unfortunate, because people e.g. with a HX711as in #2977 (which looks like a much better choice than the RF1000's 16 bit 8 SPS ADC...) could not use the code. Maybe an intermediate solution is to add a new function to the ADC module interface, which is then required for all ADCs which shall be used with the new probe module.

> In contrast, if you're not interested in homing with the sensor, and instead are just interested in doing probing, then it may be possible to do that in just the host code. 

Sorry, this was unclear in my original post. I intend to do exactly that. Homing based on the sensor is anyway not a good idea, our ADC has a low sampling rate of only 8 SPS. Also load cell readings are subject to drifts (especially at high temperatures - some people need to do mesh bed leveling with a hot bed), which makes it impossible to use a simple threshold on the read out value. Instead we need a sophisticated algorithm moving the toolhead down and up iteratively until we found the contact point while applying a low force. A higher force will spoil the measurement long before doing damage to the hardware, because the load cells (and all other components) will bend slightly. - As I said, I do not have to reinvent this, as we have a working firmware doing exactly this.

> It would require moving, waiting for the move to complete, measuring the load, moving again, measuring, etc. It would be slow, but may work okay.

This is what the Repetier firmware is doing. I don't expect it to be (considerably) slower with Klipper in the end, simply because we are mostly limited by the 8 SPS readout speed.

> In that case, you probably want to look at how klippy/extras/bltouch.py interacts with klippy/extras/probe.py - your code would also need to override how the low-level PrinterProbe._probe() code works.

Thanks, this is what I will do!

Btw: Other features mentioned in #2977 are also already implemented (at least partially) in our Repetier-based firmware, and I intend to implement them for Klipper as well. We have e.g. a Z distance correction for the first layer based on the extrusion pressure, which does a great job to compensate for the thermal expansion of the extruder (which partially happens only after reaching the temperature). Also, since the RF1000 can double as a small CNC mill, I like to add stock part edge detection to set the origin and rotation easily. This works, because the load cells are somewhat sensitive also in X and Y (but algorithms need to be even more sophisticated).

mhier, 2020-11-04:
I have a first, somewhat working version. The scan for one point takes quite some time. This is expected. The problem now is, that Octoprint will lose connection during the scan. Do I have to call some function periodically to make sure other tasks are properly serviced?

KevinOConnor, 2020-11-05:
As long as some output is periodically produced (`gcmd.respond_info()` or `gcode.respond_info()`) then OctoPrint will not complain.

-Kevin

mhier, 2020-11-27:
I am still working on this. I personally don't need this ticket to stay open, but maybe it would be a good idea to keep this open so others know that there is something ongoing. A quick status update:

Bed mesh calibration works now quite well. The repeatability is in the order of 1/100 mm (peak-to-peak) resp. 2/1000 mm (RMS):
```
Recv: // probe accuracy results: maximum -2.268871, minimum -2.275891, range 0.007020, average -2.271472, median -2.270901, standard deviation 0.002020
```

The absolute values are slightly off though, and I have to investigate why. The 0 position is slightly above the bed (0.05 to 0.1mm). It is not a show stopper, as it can be easily corrected, but it is still unclear whether this depends on the copy of the printer model (one other tester had reported a bigger discrepancy). I guess I also need more testers...

Also I want to implement more advanced features as well. Even if they don't go into the first pull request, I want to make sure I have a working interface for these advanced features. Also some code cleaning is still necessary.

mhier, 2021-01-04:
Sorry, I missed the bot because I didn't check the mails in the holiday season. I am still working on this, and I actually have a question:

Closely connected to the load-cell probe and the bed mesh scan, I have implemented a feature (in a separate module called z_offset_scan) which allows to remeasure the overall Z offset at a single point. This is helpful if e.g. a mechanical change was done which affects the absolute distance between nozzle and bed (e.g. change position where Z endstop triggeres) without changing the relative levelling of the bed (e.g. bed screws are not changed etc.).

I am wondering, whether this feature already exists in Klipper - I couldn't find anything exactly like this in the documentation. It may be somewhat motivated by the slowness of the load cell based probe - a full bed mesh scan takes easily 10 or 20 minutes (depending on number of points and the starting distance from the bed). Still I could imagine it would be a nice feature to have for everyone.

Now my actual question: For this feature to work as expected under all conditions, I would need to modify the bed mesh scan module: The z_offset_scan module needs to be informed when the bed mesh scan is completing a new scan, since the Z offset needs to be set to 0 in that case (any Z offset correction previously determined by the z_offset_scan module is then properly corrected by the new mesh already). Also it would ideally require knowledge of the profile management, as the Z offset may need to be managed per profile (I still have to think of the implications in detail here...).

Would such change be ok? Of course the bed mesh module still need to work without the z_offset_scan module. Or maybe the entire feature can be integrated into the bed mesh module.

What do you think?

KevinOConnor, 2021-01-05:
> I have implemented a feature (in a separate module called z_offset_scan) which allows to remeasure the overall Z offset at a single point.

How does that differ from the existing PROBE_CALIBRATE command?

> The z_offset_scan module needs to be informed when the bed mesh scan is completing a new scan, since the Z offset needs to be set to 0 in that case

The bed mesh code works by altering the coordinates of the incoming stream of g-code commands.  When it does a scan, it always uses the coordinate system defined in the config file, and is thus not impacted by a mesh or other form of coordinate translation.  Also, the bed mesh code already has a mechanism to apply (or not apply) the probe's z offset - it's the relative_reference_index config option.

-Kevin

mhier, 2021-01-05:
Thanks for reopening :+1: 

> How does that differ from the existing PROBE_CALIBRATE command?

In my understanding (which might be incorrect), PROBE_CALIBRATE calibrates the absolute probe offset **before** performing a bed mesh scan, involving a MANUAL_PROBE like procedure. The Z offset scan still relies on a properly calibrated probe and is fully automated. It will update an **existing** bed mesh using a single measurement point. This is mostly a speed up, a Z offset scan can be replaced by a full bed mesh scan.

In addition, the Z offset scan also allows to measure the Z offset after homing (without updating the config). This can be used e.g. if the Z endstop is not precise (statistical error), or for milling (tool length sensor).

So if my understanding of the PROBE_CALIBRATE command, this is something completely different.

> The bed mesh code works by altering the coordinates of the incoming stream of g-code commands. When it does a scan, it always uses the coordinate system defined in the config file, and is thus not [...]

Yes, I know. The problem is the other way round: The z offset scan module also implements a coordinate transform (same mechanism), which alters the Z coordinate and then passes on the transformed coordinates to the bed mesh module. Also when performing the Z offset scan, it will subtract the bed mesh correction value for the used XY position to avoid that part of the correction being applied twice.

Maybe the best way to implement this would be to avoid adding the second coordinate transform and instead alter the bed mesh itself. This would nicely play along with switching mesh profiles and intrinsically provide the correct behaviour if a new mesh is scanned. It still probably will require to alter the bed mesh module, since it has to alter the mesh also on the fly (for the milling case resp. bad endstop case).

KevinOConnor, 2021-01-06:
I'm not really sure what you're trying to do.  Lets see if we can walk through it.

> In my understanding (which might be incorrect), PROBE_CALIBRATE calibrates the absolute probe offset before performing a bed mesh scan, 

The PROBE_CALIBRATE command is a tool for finding the probe's z_offset.  The z_offset is the distance, in millimetres, between the bed and nozzle when the probe "triggers".  This value is intended to be a physical constant which is not really related to bed mesh.

> The Z offset scan still relies on a properly calibrated probe and is fully automated. It will effectively update an existing bed mesh using a single measurement point.

Okay, that sounds more like Z_ENDSTOP_CALIBRATE.  That tool is intended to find the distance, in millimetres, between the nozzle and the bed when the z endstop "triggers".  It too is intended to be a physical constant.

> In addition, the Z offset scan also allows to measure the Z offset after homing (without updating the config). This can be used e.g. if the Z endstop is not precise (statistical error), or for milling (tool length sensor).

Are you saying that the probe is more accurate than the endstop switch, but you can't use the probe for Z homing?  If so, I'd probably handle that with a homing_override config section (and/or a series of macros).  That is, a procedure that does something like: 1) home using Z endstop, 2) probe using Z probe, 3) call `SET_KINEMATIC_POSITION Z=...` to update the printer with a more accurate Z position.

> The z offset scan module also implements a coordinate transform (same mechanism)

Okay, but I don't understand what that module is trying to accomplish.

-Kevin

mhier, 2021-01-07:
Ok, maybe the term "z_offset" has a different meaning for me, or in other words: I should find a different name :-)

> The PROBE_CALIBRATE command is a tool for finding the probe's z_offset. The z_offset is the distance, in millimetres, between the bed and nozzle when the probe "triggers". This value is intended to be a physical constant which is not really related to bed mesh.

With a load-cell based probe, this z_offset is always 0, since the nozzle is the tip of the probe. At least the deviation from 0 should be much less than it would be possible to measure with a paper test. If this is different in another hardware setup, this is a different story (I do not intend to cover this now).

> Okay, that sounds more like Z_ENDSTOP_CALIBRATE. That tool is intended to find the distance, in millimetres, between the nozzle and the bed when the z endstop "triggers". It too is intended to be a [...]

Yes, this is closely related. In my understanding the Z_ENDSTOP_CALIBRATE is:

1. manual
2. changes the Z value the printer assumes right after homing, hence it does not allow to correct the Z value without performing another homing operation
3. probably requires to perform a new bed mesh scan after the endstop calibration has been done.

These 3 points a the main difference.

> Are you saying that the probe is more accurate than the endstop switch, but you can't use the probe for Z homing? If so, I'd probably handle that with a homing_override config section (and/or a [...]

I still need to have some command which determines the new Z=0 position (contact point of bed and nozzle) to be able to write this kind of macro. Maybe this points me to a better solution for the implementation of the Z correction.

So I could change my new command (after finding a better name ;-)) such that it merely searches for the new Z=0 position and stops there. The user then has the option to call SET_KINEMATIC_POSITION Z=0 (or whatever Z coordinate he likes to have there). This would allow the "online" correction.

Is there an option to perform a Z_ENDSTOP_CALIBRATE using an automatic probe? If not, maybe the preferred way would be to implement that?

I think, then with these two solutions all use cases should be covered, and the implementation is much more "klippery" ;-)

KevinOConnor, 2021-01-24:
FYI, there was a recent discussion on this at https://github.com/KevinOConnor/klipper/issues/3741#issuecomment-762986313 .  @dmbutyugin - you may be interested in the comments at https://github.com/KevinOConnor/klipper/issues/3496#issuecomment-755971545 .

> With a load-cell based probe, this z_offset is always 0, since the nozzle is the tip of the probe. At least the deviation from 0 should be much less than it would be possible to measure with a paper test.

FYI, every contact probe I've seen required a negative z_offset - that is, it always depresses the bed a small amount prior to the signal being received and processed.

Last I looked, I was able to calibrate heights to around the 10micron range using the "paper test".  That should be accurate enough to detect and measure a negative z offset.

-Kevin

dmbutyugin, 2021-01-24:
Thanks, Kevin.

> FYI, every contact probe I've seen required a negative z_offset - that is, it always depresses the bed a small amount prior to the signal being received and processed.

In principle, I can second that. However, when I used a paper test with PROBE_CALIBRATE to calibrate the 'offset' of ADXL345 probe vs the nozzle, I was getting results around -0.11..-0.09 mm, which are awfully close to the thickness of the paper sheet I was using for calibration (I measured it to be equal to 0.1mm +- 0.01mm with a micrometer). It's always necessary to leave some room for thermal expansion (and z_offset of the probe is likely the most straightforward way to do that). But it would seem that in this case the 'probe' has very little offset of its own vs the tip of the nozzle.

> Is there an option to perform a Z_ENDSTOP_CALIBRATE using an automatic probe? If not, maybe the preferred way would be to implement that?

It also makes sense to have something like that for ADXL345-as-a-probe. The motivation is similar to the reasoning given in this ticket: due to the need to prepare the nozzle for calibration (clean the tip from the filament residue), it is less practical to use ADXL345 as a Z endstop, but the sensor might be precise enough to re-calibrate the bed level should something change in the system.

So, my proposal would be to really implement something like that in Klipper. I made an implementation of my own for ADXL345 as `BED_OFFSET_CALIBRATE` command, but that implementation is very crude - it simply recalculates the new Z endstop offset and allows the user to save it using `SAVE_CONFIG`. Ideally the implementation should allow online adjustment of the bed offset too. In principle, it would probably also be fine if that could be done using some macro (assuming the config values can be altered from within the macro too), e.g. if a PROBE command can return a value that can be used in a macro.

mhier, 2021-01-25:
> FYI, every contact probe I've seen required a negative z_offset - that is, it always depresses the bed a small amount prior to the signal being received and processed.

I am measuring the force in dependence of Z, starting from a rough estimate of the contact position into negative Z direction (for a small distance only, of course). Then I am making a linear fit, which allows me to extrapolate the position with zero force. Hence I do not expect a systematic offset for this probe, as long as the force is linear with the Z depth (it usually should be, unless e.g. one uses a custom build plate on top of the original heat bed and it is not perfectly attached - I have tried to minimise the impact of this as well, but it wouldn't be 0).

On the other hand, I have to say that one of my testers is seeing a small, negative Z offset (ca. 0.1mm, potentially less). He is using a relatively soft surface (Pertinax, phenolic paper), maybe this makes a difference. We are still investigating where this is coming from. He is anyway seeing some discrepancy between his different methods to measure this offset, so I am not really sure what is going on.

I have checked the Z offset (on my printer) with 3 methods: "paper test", measure thickness of a printed brim/skirt, and visual inspection of the air gap between nozzle and bed (lower nozzle until the light of a LED right behind the nozzle is completely blocked). For me all three methods perfectly match within their precision (measure brim/skirt is worst, visual air gap is best), and I do not see any significant discrepancy from what the load cell probe is measuring.

In any case, we could introduce a static offset to correct our load cell or accelerometer based probes, if this seems necessary. One would have to determine this offset with a "paper test" or similar approach (and keep in mind it will likely depend on the bed surface material). This has to be done only once. A `BED_OFFSET_CALIBRATE` will be used much more frequently, e.g. whenever one exchanges a nozzle or even the entire hotend one would quickly use this command to correct for the changed distance.

@dmbutyugin I assume we can use the same code for the `BED_OFFSET_CALIBRATE` command for our two probe modules, am I correct? Can you point me to your code, so I can test it with my load cell probe?

dmbutyugin, 2021-01-25:
Yes, the code is not very elaborate and can be found [here](https://github.com/dmbutyugin/klipper/commit/ecf3e26d0c6d2f261d7f8bf1658f3a2106be14d9#diff-d8440e932c64928944cea8b5205daf6170a60d4b845e63577f3329e979b29b68R149-R167).

mhier, 2021-01-26:
@dmbutyugin Looks good. I think the approach to change the position_endstop is actually the best (see the discussion with Kevin above). I would suggest we create a separate module out of it. I also thought about unifying it with the manual probe implementation, but I think it's not really worth it - it's only sharing the z_endstop_finalize() function, which is only a few lines of code. What do you think? Will you do it and create a separate pull request for it? It's a bit pointless without our probe implementations, but getting this right is mostly what is missing for the load cell based probe, so I would hopefully be able to make the pull request soon after. (I can also take it and move it to a separate module myself, if you prefer...)

dmbutyugin, 2021-01-26:
I think there's no point in reusing the code with manual_probe, there's really too little to share.

For the PR, I don't mind it either way, I can make it, or you can do that too. However, we need a principled agreement from @KevinOConnor to accept it :)

Also, in my opinion, it would be good if this `BED_OFFSET_CALIBRATE` could not just adjust the `z_endstop` position, but also adjust the current position (to correct for the bed offset vs z endstop). Kevin, what would be the best way to go about it? Do it similarly to `SET_KINEMATIC_POSITION` and call `toolhead.set_position()` with appropriately adjusted z position? I see that `z_tilt` does some pretty low-level stepper position adjustments and `bed_titl` instead registers a move transform. However, for the bed offset we likely don't need that complex machinery.

mhier, 2021-01-27:
> Also, in my opinion, it would be good if this `BED_OFFSET_CALIBRATE` could not just adjust the `z_endstop` position, but also adjust the current position (to correct for the bed offset vs z [...]

I the discussion above (before you joined) we basically agreed to split this up into two parts: Firstly we have the `Z_ENDSTOP_CALIBRATE` equivalent (you called it `BED_OFFSET_CALIBRATE`, maybe we should find a name which is closer to the "original", e.g. `Z_ENDSTOP_SELF_CALIBRATE`). Secondly there should be a command which just finds the contact position with a probe and stops there. The user can then use `SET_KINEMATIC_POSITION` to set the Z coordinate as he wishes.

So far I have implemented both functionality within the same command using a move transform, but it doesn't work so well. Typically we want to combine this with a bed mesh, which also does a move transform. This has to be taken into account properly, which is not trivial. At least there will be a lot of special cases to consider to get this right: What happens if people do not have the bed mesh module enabled? Or what happens, if there is a bed mesh enabled, but you have not yet done any `BED_MESH_CALIBRATE`?

Btw: Maybe we could make things even more consistent (for the user), if `BED_OFFSET_CALIBRATE` and our new `BED_OFFSET_CALIBRATE`/`BED_OFFSET_AUTO_CALIBRATE` (or whatever we will call it) will also not run the probe themselves. Instead we would run the probe (manual or automatic) explicitly with a different command, which stops at the contact position. Then we have the choice to either call `SET_KINEMATIC_POSITION Z=0` or something like `CALIBRATE_ENDSTOP_POSITION Z=0`, which will change the endstop position such that the current position will be Z=0 after re-homing. In that case, one can also call both, if needed. This approach would of course require to change existing behaviour (at least if we want to have it consistent).

What do you think?

KevinOConnor, 2021-01-27:
> what would be the best way to go about it? Do it similarly to SET_KINEMATIC_POSITION and call toolhead.set_position() with appropriately adjusted z position?

Great question.  I don't know the answer.  I'm certainly open to adding support of useful functionality and commands that support that.  I think it would also be useful to discuss and come to a conclusion on a high-level Z calibration strategy.

By my count, there are 11 Z calibration tools in Klipper right now:
1. G-Code `G92 Z0` command.  Accounts for a Z offset.  Implemented via a gcode transform.  Applied to current session and there is no auto save/restore (though it may be used in macros or by slicers).
2. G-Code `SET_GCODE_OFFSET` command.  Accounts for a Z offset.  Implemented via a gcode transform.  Applied to current session and there is no auto save/restore (though it may be used in macros).
3. G-Code `Z_ENDSTOP_CALIBRATE` command.  Accounts for a Z offset.  Implemented by adjustment to Z `position_endstop` config.  Applied to future sessions only - via SAVE_CONFIG command.
4. The bed_mesh module. Accounts for Z offset, Z tilt, and bed/rail "bumps".  (If `relative_reference_index` is not set then the Z offset is adjusted and it is based on the probe's z_offset.)  Implemented via a gcode transform.  Can be applied to current session and may be saved/restored to future sessions via SAVE_CONFIG.
5. The skew_correction module.  Can account for a Z tilt.  Implemented via a gcode transform.  Applied to current session (users are expected to enable it via macros).
6. The bed_tilt module.  Accounts for a Z offset and Z tilt.  (Z offset adjustment is based on the probe z_offset.)  Implemented via a gcode transform.  Applied to current session and may be saved/restored to future sessions via SAVE_CONFIG.
7. The z_tilt module.  Accounts for a Z offset and Z tilt.  (Z offset adjustment is based on the probe z_offset, but is only applied until the next `G28 Z0` command.)  Implemented by physically adjusting Z stepper motors.  Applied to current session and following sessions (as long as the Z steppers aren't physically moved out of alignment).
8. The quad_gantry_level module.  Accounts for a "Z gantry tilt".  Implemented by physically adjusting Z stepper motors.  (Corrupts current Z offset - users are expected to run `G28 Z0` after running tool.)  Applied to current and following sessions (as long as the Z steppers aren't physically moved out of alignment).
9. The bed_screws module.  Accounts for a Z offset and Z tilt.  Implemented by a user that physically adjusts bed screws.  Applied to current session and following sessions (as long as the bed screws aren't physically altered).
10. The screws_tilt_adjust module.  Accounts for a Z tilt.  Implemented by a user that physically adjusts bed screws.  (It uses the probe, but does not adjust Z offset nor utilize the probe's z_offset.)  Applied to current session and following sessions (as long as the bed screws aren't physically altered).
11. The delta_calibrate module.  Accounts for Z offset and "Z flatness". Implemented by adjusting kinematic settings in the config.  Applied to future sessions only - via SAVE_CONFIG command.

In addition to the above, there are also the following related Klipper modules: endstop_phases, manual_probe, probe, bltouch, safe_z_home, homing_override, homing, query_endstops.

As you can imagine, I find the above to be daunting.  I expect it will be a challenge for regular users as well.

I'm not sure if it's possible, but if we could come up with some high-level Z calibration scheme, that may make it easier to document and handle new functionality.

@Arksine - maybe you have some comments on this?

Thoughts?
-Kevin

mhier, 2021-02-16:
> Yes, the code is not very elaborate and can be found [here](https://github.com/dmbutyugin/klipper/commit/ecf3e26d0c6d2f261d7f8bf1658f3a2106be14d9#diff-d8440e932c64928944cea8b5205daf6170a60d4b845e63577f3329e979b29b68R149-R167).

Here is my preliminary version in a stand-alone module:
https://github.com/RF1000community/klipper/blob/master/klippy/extras/z_endstop_set_calibration.py

The problem I currently have is that it does not work well when including the config file. If the `position_endstop` value is in the included file, `SAVE_CONFIG` will fail:
```
SAVE_CONFIG section 'stepper_z' option 'position_endstop' conflicts with included value.
```

mhier, 2021-02-20:
I just noticed, we must not use `SET_KINEMATIC_POSITION Z=0` after running the `PROBE` command, but `G92 Z0` instead. This is important if a bed mesh calibration is used as well, which likely has a non-zero compensation at the scanned position. We want the GCODE coordinate system to have Z=0 when the nozzle has contact with the bed, and G92 does exactly that. Using SET_KINEMATIC_POSITION instead will create an error of the size of the bed mesh value at the scan position.

mhier, 2021-03-15:
Still working on it, sorry some things are going a bit slow... :-)

mhier, 2021-04-07:
This bot is a tad fast for my taste... ;-)

KevinOConnor, 2021-04-07:
FYI, I increased the stale bot from 3 weeks to 5 weeks (commit 57291097).

-Kevin

mhier, 2021-05-17:
I am still stuck at the problem that saving the config file is not possible if a value in an included file needs to be overridden. Any hints how to tackle this issue best?

KevinOConnor, 2021-05-17:
In my opinion, the SAVE_CONFIG tool has to "walk a careful balance" between being transparent and being easy to use.  We don't want to require a user to cut-and-paste generated values into their config, but we also don't want to make changes to a user's config in such a way that they can't easily identify their current config.

As a result, the SAVE_CONFIG tool has some limitations.  I don't see a good way forward other than: don't use SAVE_CONFIG (manually save the values); or limit the use of the config include tool to only those sections/settings that don't change.

-Kevin

mhier, 2021-05-17:
Thanks for the reply.

> As a result, the SAVE_CONFIG tool has some limitations. I don't see a good way forward other than: don't use SAVE_CONFIG (manually save the values); or limit the use of the config include tool to only those sections/settings that don't change.

Well, the idea is to automatically determine:
```
[stepper_z]
position_endstop = ***
```
This value is a config value for good and obvious reasons, and it makes very much sense to have a default value for this in each printer config file shipped with Klipper.

Hence I see two solutions:
* Comment this value in the config file for the printer models using the load-cell based probe (same will apply for the accelerometer-based probe) and put an instruction into the header of the file to un-comment it if the file is used stand-alone, resp. put a default value into the including printer.cfg. Without action of the user, the file would then be unusable.
* Alternatively abandon the approach of modifying the position_endstop value and instead introduce again another coordinate transform. I had this approach before, but you talked me out if it ;-)

I would currently tend towards the first approach (commenting value in shipped cfg file). Do you agree with this approach? @dmbutyugin What about you?

mhier, 2021-06-23:
Still working on it, slowly... :-)

mhier, 2021-08-02:
sorry, I am still slow... My printer is currently not in a condition to do the tests I need to do and needs to be fixed first :-(

mhier, 2021-09-07:
My printer is fixed now, hence I will start preparing a first PR this evening (automatic bed offset calibration as discussed above).

mhier, 2021-09-07:
So finally, a first PR is made: #4670 :-)

mhier, 2021-09-29:
Next pull request is there, #4738, introducing finally the module for the load cell-based probe. I am quite certain, @KevinOConnor will request some changes, although I tried to do my best to avoid foreseeable issues :-) It's a lot of code, but I added documentation as well.

mhier, 2021-11-04:
Well, the PR #4738 is still pending... Any news about this? Anything I can do to speed things up?

afriess, 2021-12-09:
Is PR #4738 still pending, and under review ??

afriess, 2022-01-14:
No review for PR #4738 in the meanwhile. No remark.

jakep82, 2022-01-14:
> No review for PR #4738 in the meanwhile. No remark.
> 
> All checkes are viewed as passing.

Passing the automated checks doesn't mean the code is without issues.  Klipper currently has more than 100 open PRs, and a limited number of developers so it could be a while longer until this is reviewed.  Kevin responded in October and said as much.

yzheka, 2022-02-05:
Waiting for this feature in klipepr too

mhier, 2022-02-08:
> Waiting for this feature in klipepr too

Feel free to test it from my branch (see the PR). Let me know if you need help to set it up etc. What hardware do you have?

yzheka, 2022-02-08:
@mhier I have anycubic kossel linear plus and 3 load cells 1kg rated. I will try your branch once i am ready with cells mounted.

mhier, 2022-02-08:
@yzheka Question is also how you read out the load cells. Currently, my branch only supports the specific hardware setup of my printer. Luckily, the schematics of my printer's electronics is openly available:

https://asset.conrad.com/media10/add/160267/c1/-/en/001197630CD01/circuit-diagramm-1197630-main-board-i851ctc-v21-completely-suitable-for-3d-printer-renkforce-rf1000.pdf

It is in German language, sorry about that :-) The important part is on the second page top left ("DMS Verstärker und A/D").

I am also working on a support for the widely known HX711, but it seems to be not really trivial...

yzheka, 2022-02-08:
@mhier I thought using your branch I can wire amplifier board digital pin and use it as probe.

mhier, 2022-02-08:
Just digital on-off information is not sufficient, the result would be too imprecise. We need the analog signal. Also the onboard ADCs in the MCUs are typically not good enough, hence you likely need to use an external ADC.

yzheka, 2022-02-08:
@mhier hmm, i guess use low cost Arduino to convert analog weight value to digital signal (like probe) will be easier.

mhier, 2022-02-08:
> @mhier hmm, i guess use low cost Arduino to convert analog weight value to digital signal (like probe) will be easier. 

As I said, the on-chip ADCs in most microcontrollers are not good enough. You will need at least 16 bit resolution and differential input. Speed on the other hand does not matter. Not sure if you find this on a low-cost arduino... (if yes, let me know!)

garethky, 2022-02-17:
If I understand the PR correctly what you are doing is moving the toolhead to a new position and then taking a measurement from the load cell. This isn't really ideal but I understand why you had to do it that way. You cant use the endstop system on the MCU because that's digital, the 12 bit ADCs on the MCU aren't workable, and the code on the Pi takes too long to run to be interactive with the MCU. So move in small increments and take measurements after each movement is completed.

How long does one of these probe cycles take? What about doing a 5x5 bed mesh with 3 probes per point?

It seems like we need a board that can do the 16 bit ADC, run the detection algorithm and send a digital output signal to the MCU to trigger the endstop pin. The detector settings can be tuned from Klipper but the detection code doesn't run in Python inside Klippy, it happens on the board in C. The existing probing code based on endstops would just work as-is. (My guess is this is how touch probes integrate with a CNC's PLC).

mhier, 2022-02-17:
> If I understand the PR correctly what you are doing is moving the toolhead to a new position and then taking a measurement from the load cell.

Correct.

> the 12 bit ADCs on the MCU aren't workable

Even if they were, the converted value still would have to go through some more or less complicated algorithm, at which end a decision is made what to do next. Applying a simple threshold on the measured ADC value is far too imprecise and unreliable.

> How long does one of these probe cycles take?

I don't have an exact time right now (I can measure it this evening if it helps). It is in the order of half a minute or so, depending how far one choses the initial distance from the bed. The total scan of a 5x5 mesh hence takes maybe 15 minutes or so. This is long compared to other probe systems, but I do not consider this a big problem, as one does not have to do this scan very often.

> It seems like we need a board that can do the 16 bit ADC

I think it does not matter whether the ADC is integrated into the MCU or it is attached via I2C or SPI or so. If we do not want to put complex algorithms into the MCU code, there are some speed limitations to live with. Still, the original firmware of my printer was Repetier-based and had already support for such load-cell-based probe (this is a feature my printer was shipped with by the manufacturer). The scan with that firmware was not faster. The time per point was a bit shorter, but it had to do more points, since the mesh interpolation is merely bilinear (due to limitations of what you can put into the small MCU - we have an ATMega2560 only).

I see one potential for improvement though: If we could schedule the I2C/SPI communication with the external ADC (or internal ADC measurements if the MCU has a good enough one), the algorithm could avoid waiting for the result. It could schedule alternating movements and measurements and do the computations a bit delayed. This would of course mean that especially during the first "fast" approach the head moves further "into" the bed until the Python part can react, so the speed needs to be slow enough such that this is acceptable.

garethky, 2022-02-17:
I want to get the probing time down to where it is with other methods so we can bed mesh at print start. We have found this to be more reliable than storing the mesh. I see load cell based probing as the way forward in FDM. First layer reliability is the biggest hurdle for new users and load cell probes on the tool head can virtually eliminate the machine side of the problem (surface prep and temperature are still an issue).

I guess we will see later this year when Prusa releases the ECAD files and code for the XL exactly how they are doing it. They claimed it would work with Klipper. In the meantime I feel your pain on trying to find a ready made dev board that would do the 16+ bit ADC, provide some compute and I/O pins. Seems like the simplest solutions is a HX711 breakout board + Arduino. Maybe later someone can make that into a RPI hat.

mhier, 2022-02-18:
In my experience, it is sufficient to make a single point scan rather frequently (e.g. before every print). My printer is build rather solid, so maybe this is different for other printers, but I think it would be worth trying. On my printer, I have no second guesses about the first layer at all any more, it just works even without the slightest manual adjustment. The single point scan is possible with my PR already. I have described this here:

https://github.com/RF1000community/klipper/blob/master/docs/LoadCellProbe.md#best-practises

My approach would be to get this PR through first and then see how to improve the speed. The integration of an HX711 is also not so trivial, see my forum post here: https://klipper.discourse.group/t/hx711-support-load-cell-sensor/2057

I am confident that all this can be done. My PR is waiting for quite some time already, because it is already quite complex. I don't want to increase complexity by even adding more right now...

garethky, 2022-02-19:
Ok, some hard stuff is left to do. The right end state seems to be:

* ADC connected to an MCU over SPI (seems like the hard part right now)
* A module is added to the MCU code to interact with the ADC, providing the following functionality:
  * Apply ADC settings from klippy (gain, sample rate etc.)
  * Send raw ADC data back to klippy from the MCU, possibly at some reduced sample rate
  * Configure a Trigger to occur when the next value from the ADC is sampled above the trigger setpoint.
  * Disable/Reset the Trigger

Then in Klippy we can:
* Consume the raw ADC data and make it available to klippy modules for all the use cases that don't require instant stepper respnse. (weight of the print, extruder pressure/blockage detection, nozzle collision detection, live print tuning etc.)
* For probing, set up a specific trigger setpoint for an individual probing move. This can take into account all factors klippy is aware of (speed and direction of the move, current correction for temperature drift etc.). Then we can use `stepper_stop_on_trigger` to perform the homing move and trigger from the ADC on the MCU to stop the move.

Job 1 is getting some ADC to work directly with the MCU over SPI. The HX711 has been brought up numerous times but you have found its SPI communications to be an issue. I trust your investigation. What about using one of the TI chips like ADS1220 or ADS ADS1262? (e.g. https://protocentral.com/product/protocentral-ads1262-32-bit-precision-adc-breakout-board/). If we pick a chip that makes out lives easier to implement the SPI communication that's a win. Programmer time and code complexity cost more than hardware.

mhier, 2022-02-21:
There is already such module in my branch:

https://github.com/RF1000community/klipper/blob/master/klippy/extras/ads1100.py

I though I will send this change as a PR first as it is the more generic part, while the ADS1100 support does not make really much sense without this module. I have also even created a temporary hx711 module with the same interface, but more work is required to get a reliable communication.

Btw: some posts above I posted a link to the schematics of my printer's mainboard including the load cell amplifier and digitiser circuit using an ADS1100. If you reproduce that exactly, you should be able to use my code right away.

>     * ADC connected to an MCU over SPI (seems like the hard part right now)

So this part is done for my hardware, but still needs to be done for more generic hardware (like the HX711).

> * For probing, set up a specific trigger setpoint for an individual probing move. This can take into account all factors klippy is aware of (speed and direction of the move, current correction for [...]

As I said, this is generally insufficient for probing. The algorithm needs to be a lot more complex than a simple threshold. Otherwise you would at least lose the big benefit of an absolute measurement (without offset to calibrate manually), if not degrade the precision beyond usability. I have tried that out already.

I have another idea how to improve the speed:

* Write an MCU driver for whatever ADC is used (ADS1100, HX711 whatever) which ships a stream of ADC data with timestamps to the Python side.
* Python code receives stream. Since data is timestamped, it can understand at which Z position the printer was and if it was moving or standing (readouts are MUCH more reliable when everything is at rest)
* Algorithm needs to be changed such that it can plan measurements a few moves and ADC reads "ahead".

This will probably blow up the complexity of implementation by a lot. That's why I would prefer doing this in a second step only. I consider speed not so problematic, as this is mostly a matter of organising the workflow. If I require a new bed leveling (which is not very often), I will start it before I slice the model. When I am done with slicing, the leveling is done as well usually. The speed is really not that much of an issue here.

garethky, 2022-02-21:
Ok, I will order a selection of these ADC boards that I can easily get (ADS1115, ADS1232, ADS1263 & HX711) and a couple of load cells. Once I get my test rig I hope I can help with this effort.

Because of the 25ms inter-MCU delay I'm really focused on getting this to work as an endstop inside the MCU driving the steppers. Its a great idea to do post-processing of the collision event in Klippy for more accuracy using your `linear_regression` method. That's very similar to the spirit of what what multi-mcu homing does. I believe we are thinking along similar lines but its hard to explain with just words. But bottom line, that I think we both can see, is the MCU needs to stop the move on its own to speed things up. If we can get this solved I'm sure we will have a lot of contributors that will write the Python side for probing, auto-calibration, jam detection etc.

I think I'll start a discussion on to the forum, I'll tag you when I get around to posting (might take me 2-3 weeks to get there).

Heltzen, 2022-02-21:
I have three factory build 3Dgence printers running a load cell to do mesh bed leveling.
3DGence One, 3DGence P255 and 3DGence Industry F340.

I am new to klipper, but not 3D printning (doing it professional for 5 years)

If you need to do some testing i am willing to help.
The printers need a firmware update, but am hesitating to change the firmware since I do not know if I can continue to use the auto Calibration feature. The printers ar so rigid build, and well assembled that there is no manual bed leveling.

mhier, 2022-02-22:
> Its a great idea to do post-processing of the collision event in Klippy for more accuracy using your `linear_regression` method. 

You may miss one point here: The linear regression is not done on direct measurements, because this is way to inaccurate. Instead every single measurement point going into the regression has to be zero-compensated with another measurement without bed contact, and this other measurement has to be right before or right after the measurement itself. To do the linear regression, the algorithm is hence "hopping" in z direction multiple times up and down. This cannot be done with a simple homing move. On top of this, the motors should stand still while taking the measurement (both the actual measurement and the zero-compensation measurement). Hence movement needs to be controlled in synchronisation with the load cell measurements.

Sill I see potential to use your idea: The algorithm currently consists of 4 phases:

1. "fast" rough approach
2. verify bed contact (if not, go back to 1)
3. backup head a bit to a position as close as possible above the bed
4. measurements for linear regression

Step 1 could be replaced with a homing move, since this is essentially a plain threshold. We still need to subtract a zero measurement from the measurements, but for this rough approach it is sufficient to take a single zero measurement before starting the move. For the second step we have to take another zero measurement to confirm that our measurements are not just drifting away (e.g. due to mechanical tension or thermal effects), but this is a quick single measurement (-> can be done fully from Python).

Maybe also Step 3 could be replaced with an inverted homing move with slower speed. Here it is only important that we end up reliably at a position without bed contact but as close as possible to the bed to safe time in the last step.

For Step 4 a homing move cannot be used, but scheduled ADC measurements could help. So if we can make our MCU ADC driver(s) capable of providing a virtual endstop based on a threshold and also to perform ADC measurements synchronised to the movements, we probably can speed this algorithm up a by a lot.

One more thing comes to my mind now: A simple threshold on the ADC measurement for the virtual endstop should be fine, because the direction in which the ADC measurement should change is known (if the load cells are correctly mounted it will go into the negative direction, as the force is moving against the weight - we could make the expected direction configurable). Hence the threshold can be computed to take into account the zero measurement. So we actually need no zero compensation in the MCU.

Btw: Some things in my algorithm may be motivated by my particular hardware setup. The ADS1100 can only do 8 Hz conversion rate (at full 16 bit resolution). This is a strong limit. Even with the original Repetier-based firmware the probing was not really faster. If the ADC gets swapped with an 80 Hz capable HX711, things will be very different. But even for the ADS1100 I see potential, since resolution can be traded for speed. So I think it is definitively worth improving this!

> If we can get this solved I'm sure we will have a lot of contributors that will write the Python side for probing, auto-calibration, jam detection etc.

Indeed, there are many nice things one can do with a load cell sensor, if it is properly mounted. In my printer's case it can measure the extrusion force, so we can do interesting things like compensating for the thermal elongation of the hotend during the first layer. I have implemented this already:

https://github.com/RF1000community/klipper/blob/master/klippy/extras/z_sense_offset.py

In the old Repetier-based firmware for my printer, our community as developed many other ideas which could be re-implemented on Klipper as well. For most of these ideas an asynchronous readout (w.r.t. the movements) is sufficient, but for many a synchronisation would help a lot.

> I think I'll start a discussion on to the forum, I'll tag you when I get around to posting (might take me 2-3 weeks to get there).

Perfect! I am happy if this project finally gets some movement :-) I will try to make a proper HX711 driver in the MCU first. I have already done some investigation in this direction, and the HX711 anyway needs special MCU code (see https://klipper.discourse.group/t/hx711-support-load-cell-sensor/2057/4). Hence I think this is a good way to dive into writing an MCU driver for me - with the aim to support the above mentioned features in the end.

mhier, 2022-02-22:
> I have three factory build 3Dgence printers running a load cell to do mesh bed leveling. 3DGence One, 3DGence P255 and 3DGence Industry F340.

Do you know what ADC chip the printers are using to read out the load cell? The first thing to do is to write a driver for it. Once that is done you can try this algorithm - you will at least have to tune a number of parameters to get it working reliably with your printer setup.

D4SK, 2022-03-05:
@mhier I recently wrote a module using mcu timers to communicate to the hx711 for this purpose. It just needs some integration to work with the probing code. Maybe we can collaborate on this. My discord is Konstantin#3476.

See https://github.com/D4SK/klippo/commit/e376340849dac9b8636a408a195464a2d2d2f8be
This reliably obtains adc values at 80hz, and provides calibration options for gain etc.

mhier, 2022-03-07:
@D4SK That looks great, thanks! I will try to get this working with my probing code as soon as possible (after my vacation ;-)). I will contact you (probably not via discord, since I am not really active there - are you in the discourse forum? My nick is mhier there, too) once I have done the first steps.

Does this module already report the correct time stamp of the read out to the Python side?

D4SK, 2022-03-07:
@mhier It does report a time stamp, maybe not the correct one though. It just uses the #sent_time parameter from the mcu message, and converts it with estimated_print_time()

I think a continuous single probing approach is also very interesting. If the timestamps exist, the only problem left should 
be finding the past stepper position for the timestamps.

mhier, 2022-04-12:
There was some discussion in the PR  #4738. I will have to split up the PR in smaller, more handleable pieces.

D4SK, 2022-04-15:
@mhier I recently found out the #sent_time parameter is probably not usable. Seems it is indicating the time the last host request was sent. So its probably best to attach a proper timestamp in the mcu code instead.

closed 2022-05-28 by github-actions[bot]