Klipper3d/klipper #3497: Error when overloaded ACTIVATE_EXTRUDER

elpopo-eng, 2020-11-03:
I would like to overload `ACTIVATE_EXTRUDER` in case of simulate multi-extruder with susie.
But this macro generate an error when i call `ACTIVATE_EXTRUDER`
```
[gcode_macro ACTIVATE_EXTRUDER]
rename_existing: BASE_ACTIVATE_EXTRUDER
gcode: 
	BASE_ACTIVATE_EXTRUDER EXTRUDER=extruder
```
But not this one:
```
[gcode_macro ACTIVATE_EXTRUDER]
rename_existing: BASE_ACTIVATE_EXTRUDER
gcode: 

```

Error:
```
Internal error on command:"BASE_ACTIVATE_EXTRUDER"
Traceback (most recent call last):
  File "/home/pi/klipper/klippy/gcode.py", line 177, in _process_commands
    handler(gcmd)
  File "/home/pi/klipper/klippy/gcode.py", line 115, in <lambda>
    func = lambda params: origfunc(self._get_extended_params(params))
  File "/home/pi/klipper/klippy/gcode.py", line 115, in <lambda>
    func = lambda params: origfunc(self._get_extended_params(params))
  File "/home/pi/klipper/klippy/gcode.py", line 273, in _cmd_mux
    key, values = self.mux_commands[gcmd.get_command()]
KeyError: 'BASE_ACTIVATE_EXTRUDER'
```
[klippy.zip](https://github.com/KevinOConnor/klipper/files/5482013/klippy.zip)

Regards

KevinOConnor, 2020-11-05:
It does look like Klipper could have done better here.

-Kevin

KevinOConnor, 2021-11-22:
PR #4843 should fix this.

-Kevin

closed 2021-11-30 by github-actions[bot]