Klipper3d/klipper #3501: Neopixel Macro with 2 Arrays Klipper Crashed

EraserFX, 2020-11-03:
Klipper crashes after I've used this macro
[gcode_macro FLASH_PART_RED1]
gcode:
	SET_LED LED=part_neopixel RED=1.0 GREEN=1.0 BLUE=1.0 TRANSMIT=1
	SET_LED LED=bed_neopixel RED=1.0 GREEN=1.0 BLUE=1.0 TRANSMIT=1

[klippy.log](https://github.com/KevinOConnor/klipper/files/5483978/klippy.log)

zellneralex, 2020-11-04:
We tested it. Daniel has 2 neopixel Array defined. One contains 1 LED the other contains 18 LEDs. Since the last comits he can not use 2 macros in a row controlling both arrays after each other. This was working in the past.

KevinOConnor, 2020-11-04:
Looks like an error.  I think this should be fixed now (commit 82156170).

-Kevin

KevinOConnor, 2020-11-05:
Alas, that commit did not work as intended, but it should now be fixed as of commit 123908ac.

-Kevin

EraserFX, 2020-11-05:
The bug is fixed. Great Work ! Thank You

closed 2020-11-06 by EraserFX