Klipper3d/klipper #3502: Z_Tilt_Align - Newbie Config Issue

mdframe, 2020-11-03:
I am sure this is my configuration causing this but I can't figure out why.  I have a corexy printer with dual z steppers.  The printer homes on all axis and the correct direction properly.  The z endtops work correctly when homing z as it waits until both are triggered before stopping.  All of my moves for x,y and z go to the correct location however when I perform a z_tilt alignment the right stepper, z1, continues to move away from the probe for each iteration.  

Please let me know thoughts or recommendations based on my klippy.log. 

Thank you, 

Matt

[klippy.log](https://github.com/KevinOConnor/klipper/files/5484030/klippy.log)

KevinOConnor, 2020-11-09:
Alas, it doesn't seem like anyone here has enough knowledge of your hardware to help - you're probably better asking in a general 3d printing forum.

-Kevin

mdframe, 2020-11-09:
Thanks Kevin.  This works perfectly under Marlin, same machine, so I am sure something in my printer.cfg is incorrect.  I will keep researching and searching. 

Thanks again, 

Matt

closed 2020-11-09 by mdframe

jgagnon5541, 2021-02-02:
I have the exact same issue also on a corexy configuration of an ender 5 plus frame with kay3d kit.  The Z1 stepper just keeps getting further and further but I can print fine and all the stepper buzz show proper configuration. Z left stepper, Z1 right stepper.
Z tilt configuration:
[z_tilt]
z_positions:
    -23,-185
    -393,185
points:
    65,180
    365,180
horizontal_move_z: 5
retries: 5
retry_tolerance: 0.015

Z_TILT_ADJUST output:
Send: Z_TILT_ADJUST
Recv: // probe at 65.000,180.000 is z=-0.618750
Recv: // probe at 65.000,180.000 is z=-0.615000
Recv: // probe at 365.000,180.000 is z=-0.585000
Recv: // probe at 365.000,180.000 is z=-0.588750
Recv: // Making the following Z adjustments:
Recv: // stepper_z = -0.094617
Recv: // stepper_z1 = -0.160812
Recv: // Retries: 0/5 Probed points range: 0.030000 tolerance: 0.015000
Recv: // probe at 65.000,180.000 is z=-0.504188
Recv: // probe at 65.000,180.000 is z=-0.504188
Recv: // probe at 365.000,180.000 is z=-0.439188
Recv: // probe at 365.000,180.000 is z=-0.439188
Recv: // Making the following Z adjustments:
Recv: // stepper_z = -0.046112
Recv: // stepper_z1 = -0.097374
Recv: // Retries: 1/5 Probed points range: 0.065000 tolerance: 0.015000
Recv: // probe at 65.000,180.000 is z=-0.452626
Recv: // probe at 65.000,180.000 is z=-0.451376
Recv: // probe at 365.000,180.000 is z=-0.355126
Recv: // probe at 365.000,180.000 is z=-0.352626
Recv: // Making the following Z adjustments:
Recv: // stepper_z = -0.027106
Recv: // stepper_z1 = -0.092428
Recv: // Retries: 2/5 Probed points range: 0.098125 tolerance: 0.015000
Recv: !! Retries aborting: Probed points range is increasing.

[klippy.log](https://github.com/KevinOConnor/klipper/files/5910706/klippy.log)