Klipper3d/klipper #3503: Blink Macro with Neopixel works no more

EraserFX, 2020-11-03:
My Neopixels blinks no more after Update.
[gcode_macro FLASH_BED_RED]
gcode:
	SET_LED LED=bed_neopixel RED=0.0 GREEN=0.0 BLUE=0.0 TRANSMIT=1
       G4 P250
	SET_LED LED=bed_neopixel RED=1.0 GREEN=0.0 BLUE=0.0 TRANSMIT=1
	G4 P250
	SET_LED LED=bed_neopixel RED=0.0 GREEN=0.0 BLUE=0.0 TRANSMIT=1
	G4 P250
	SET_LED LED=bed_neopixel RED=1.0 GREEN=0.0 BLUE=0.0 TRANSMIT=1
	G4 P250
	SET_LED LED=bed_neopixel RED=0.0 GREEN=0.0 BLUE=0.0 TRANSMIT=1
	G4 P500
	SET_LED LED=bed_neopixel RED=1.0 GREEN=0.0 BLUE=0.0 TRANSMIT=1
[klippy.log](https://github.com/KevinOConnor/klipper/files/5484107/klippy.log)

KevinOConnor, 2020-11-04:
As of #3380 the code changed to just send the latest values on an update.  I can see where this would not be ideal though.  I'd have to look closer to see how it could be fixed.

-Kevin

KevinOConnor, 2020-11-05:
This should be fixed now (commit 9e169778).

-Kevin

EraserFX, 2020-11-05:
Yes Fixed too. Nice. Thanks Man

closed 2020-11-06 by EraserFX