Klipper3d/klipper #3508: Neopixel Error with last commit (8215617)

elpopo-eng, 2020-11-05:
Hello with this [commit ](https://github.com/KevinOConnor/klipper/commit/821561708d0973ba2dbc93fabff381075c0b2aed)we have this error in the initialization (line 1038) :
```
MCU error during connect
Traceback (most recent call last):
  File "/home/pi/klipper/klippy/klippy.py", line 151, in _connect
    cb()
  File "/home/pi/klipper/klippy/extras/neopixel.py", line 107, in send_data
    reqclock=BACKGROUND_PRIORITY_CLOCK)
  File "/home/pi/klipper/klippy/mcu.py", line 378, in send
    raise error(str(e))
error: Unable to obtain 'neopixel_result' response
Build file /home/pi/klipper/klippy/../.config(1067): Mon Oct  5 11:37:14 2020
```
Here under the kliiper log
[klippy.zip](https://github.com/KevinOConnor/klipper/files/5494594/klippy.zip)

Regards

vvhh2002, 2020-11-05:
I encountered the same problem and the previous version worked fine.

KevinOConnor, 2020-11-05:
It should be fixed now (commit 123908ac).  The latest code will require flashing the micro-controller again.

-Kevin

closed 2020-12-05 by github-actions[bot]