Klipper3d/klipper #3511: Print corners issue

Blind-Joe, 2020-11-05:
[klippy.log](https://drive.google.com/file/d/1ybxk1sBWPnTZu2tlulQnYS4IMSNEUq5J/view?usp=sharing)

My problem is to print the corners correctly, one side is too rounded but on the other side you can see too much plastic.
Printer: Voron 2.4

[Photo1](https://drive.google.com/file/d/1TqmT3as3ERL1Ldf5UW-WVIjcloMY67Xd/view?usp=sharing)
[Photo2](https://drive.google.com/file/d/1JFvK5bi88jY5poBHmiuth-1YUvmvNuRH/view?usp=sharing)
Photo1 and Photo2 shows same corner
[Photo3](https://drive.google.com/file/d/1v3t4zZPg6_ON4zQy6BDWwDUj0gkpnNYa/view?usp=sharing)
[Photo4](https://drive.google.com/file/d/1ZI8PwaWdPrKiOrTui6EJE6qBf0fSyO_8/view?usp=sharing)
Photo4: From the top you can see like too much PA but even with this value there is still too much plastic on one side of the corner 
Printer config: [config](https://drive.google.com/file/d/10msc7YW80ul9AWoHRJa3M-zs30g4E77L/view?usp=sharing)

So far I tested:
-input shaper ( tried with and without)
-printer config (tried couple include default)
-printer hardware (motors plastic parts, gantry) printer hardware is stock from BOM
-extruder hardware (extruder gears filament path, different nozzle etc)
-klipper version (current master branch)
-slicer (superslicer kisslicer cura simplify3d)
-slicer profile ( tested 12 different profiles on 4 slicers)
-PA/preload (tested value from 0 to 0.1)
-filament (tested 4 PLAs, 3 PETG and 2 ABS brands)
-square corner velocity (from 1 to 15)/accelerations (from 100 to 10k)/speeds (from 20 to 300)
Klipper and octoprint was reinstalled 2 times. I even replaced tmc drivers.

**This is not a seam**, I can't calibrate PA because good value doesn't exist, problem occurs on all corners.

Blind-Joe, 2020-11-05:
My bad. Sending correct klippy file
[klippy.log](https://drive.google.com/file/d/1TyIPLYoHP5ffsfyeMtW-2dF2c-gCr9jj/view?usp=sharing)

Frix-x, 2020-11-05:
Have you tried reducing the acceleration (and maybe jerk) ?

I may be wrong but it look like the printer is having hard time to change direction and make an overshoot in the corner. Is your print head "heavy" ?

Blind-Joe, 2020-11-05:
I tried to print with 30mm/s and 200 acceleration but results is exactly same :P Saying jerk mean square corner velocity ? My printhead is standard voron afterburner extruder

KevinOConnor, 2020-11-09:
Alas, this isn't a good location for general support questions. You're probably better off asking your question in a Voron forum.

-Kevin

closed 2020-11-12 by klipper-gitissuebot

Talha909, 2021-06-05:
> [klippy.log](https://drive.google.com/file/d/1ybxk1sBWPnTZu2tlulQnYS4IMSNEUq5J/view?usp=sharing)
>
> My problem is to print the corners correctly, one side is too rounded but on the other side you can [...]

Did you find the solution. ?