Klipper3d/klipper #3512: TMC2209's skipping steps

jameshgolen, 2020-11-06:
I realize that skipping steps can be related to stepper current and the mechanical capabilities of the machine, however this seems repeatable at the same point each time I run this specific gcode file. I e-stopped right after it did the big skip (I'd guess 1-2 inches). This appears to be a x/y stepper issue as I'm not seeing any distinct evidence of skipping on other steppers.

I've repeated this with my x/y steppers at 650, 700 and 750ma.

I can try again at 800 but in reality I think that's too high for these steppers. They're steppers from an Anet A8.
[klippy.log](https://github.com/KevinOConnor/klipper/files/5498293/klippy.log)

Hywelmartin, 2020-11-06:
set higher current and keep track of the temperature of the steppers and/ or check datablade

try to turn off stealtchop

`stealthchop_threshold = 0`

Max acc isn't that high  but it might be to high..  for this current setup

`max_accel = 2500`

1N4148, 2020-11-06:
Also you might check the cooling of your TMCs. 1-2 inches aren't a single missing step. 
If your steppers aren't getting hotter than 50°C there's room for more current. Turning off stealthchop might be a good starting point for debugging.

jameshgolen, 2020-11-06:
Interesting I didn't think stealthchop would make much of a difference (other than being noisy as heck). This is definitely a me problem, not a klipper problem.

closed 2020-11-06 by jameshgolen