Klipper3d/klipper #3513: Resume after controlled power off or restart?

anjoku, 2020-11-06:
Hello, please excuse my not so good English.

I've read a lot that many users want a resume after a sudden powerloss.
That's not what I'm talking about.

pause and resume works for me. Even over several hours.

I just have to remember to homen before resume x y z and to heat the nozzle again.
Thanks to the homen, the printer knows where the nozzle is.

I type in resume and the print then continues without a problem.

i do this because i don't want to leave the printer unattended for long.

unfortunately it doesn't work when I switch off the printer or the OrangePi.
Klipper then no longer has a connection to the printer and I have to restart a host or firmware in octoprint.

Klipper forgets the pause status and reports that no resume is possible.

is it possible that pause saves the data in such a way that resume is also possible after a restart?

I don't care about uncontrolled power loss.

Many thanks.

Andre
[klippy.log](https://github.com/KevinOConnor/klipper/files/5501557/klippy.log)

KevinOConnor, 2020-11-09:
> is it possible that pause saves the data in such a way that resume is also possible after a restart?

It may be possible, but an interested developer would need to implement it and submit it.  It's probably easier to just leave the electronics powered up.

-Kevin

anjoku, 2020-12-01:
Thank you, unfortunately I will not be able to implement it.
Thanks for your help.
best regards

closed 2020-12-01 by anjoku