Klipper3d/klipper #3514: Command format mismatch error

solcer, 2020-11-06:
[klippy.log](https://github.com/KevinOConnor/klipper/files/5503128/klippy.log)
Hello,

just update new version of klipper then i get this error. 

Command format mismatch: schedule_soft_pwm_out oid=%c clock=%u on_ticks=%u off_ticks=%u vs schedule_soft_pwm_out oid=%c clock=%u on_ticks=%u
This type of error is frequently caused by running an older
version of the firmware on the micro-controller (fix by
recompiling and flashing the firmware).
Once the underlying issue is corrected, use the "RESTART"
command to reload the config and restart the host software.
Protocol error connecting to printer
Klipper state: Not ready

Command format mismatch: schedule_soft_pwm_out oid=%c clock=%u on_ticks=%u off_ticks=%u vs schedule_soft_pwm_out oid=%c clock=%u on_ticks=%u

I tried RESTART command and install new klipper instance but result is same. I couldn't solve problem. pls help.

1N4148, 2020-11-06:
You must build & flash a newer firmware for the controller board in the same way as you had done on initial installation.

solcer, 2020-11-06:
I did as you said. i followed this: https://github.com/KevinOConnor/klipper/blob/master/docs/FAQ.md#how-do-i-upgrade-to-the-latest-software

Btw my board is skr 1.3. I upload firmware by sd card.

1N4148, 2020-11-06:
Was the firmware.bin renamed to firmware.cur? Please check again that you've build a new firmware and rename the klipper.bin to firmware.bin. I've did the update some days ago (I've got the same message) and it works.

solcer, 2020-11-07:
I changed file name and it worked. Thank you for your help.

closed 2020-11-07 by solcer