Klipper3d/klipper #3516: Feature request - include heater name in ADC out of range error

vladbabii, 2020-11-07:
Current error looks like this
```
Transition to shutdown state: MCU 'mcu' shutdown: ADC out of range
This generally occurs when a heater temperature exceeds its configured min_temp or max_temp.
```

but i think it should include heater_bed or extruder or some name of the heater that cause the issue, and that would be more important than the mcu name, since you can track back the mcu from the heater.

Thank you!

vladbabii, 2020-11-07:
Also, the logs look like this

```
Receive: 92 823990.063098 823990.035360 13: seq: 16, analog_in_state oid=10 next_clock=54400000 value=8163
Receive: 93 823990.159039 823990.035360 13: seq: 16, analog_in_state oid=19 next_clock=55840000 value=7832
Receive: 94 823990.367056 823990.035360 13: seq: 16, analog_in_state oid=10 next_clock=59200000 value=8165
Receive: 95 823990.450477 823990.035360 13: seq: 16, analog_in_state oid=19 next_clock=60640000 value=7835
Receive: 96 823990.623062 823990.610769 10: seq: 17, clock clock=58635874
Receive: 97 823990.654974 823990.610769 13: seq: 17, analog_in_state oid=10 next_clock=64000000 value=8159
Receive: 98 823990.750918 823990.610769 13: seq: 17, analog_in_state oid=19 next_clock=65440000 value=7827
Receive: 99 823990.959102 823990.610769 11: seq: 17, shutdown clock=64128182 static_string_id=ADC out of r                                                                ange
Transition to shutdown state: MCU 'mcu' shutdown: ADC out of range
This generally occurs when a heater temperature exceeds
its configured min_temp or max_temp.
Dumping gcode input 0 blocks
gcode state: absolute_coord=True absolute_extrude=True base_position=[0.0, 0.0, 0.0, 0.0] last_position=[0                                                                .0, 0.0, -0.002017, 0.0] homing_position=[0.0, 0.0, 0.0, 0.0] speed_factor=0.0166666666667 extrude_factor=                                                                1.0 speed=25.0
Reactor garbage collection: (823988.116978, 0.0, 0.0)
Stats 823992.1: gcodein=0 mcu: mcu_awake=0.038 mcu_task_avg=0.000140 mcu_task_stddev=0.000295 bytes_write=                                                                1832 bytes_read=4665 bytes_retransmit=9 bytes_invalid=8 send_seq=168 receive_seq=168 retransmit_seq=2 srtt                                                                =0.011 rttvar=0.002 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=15989350  heater_bed: target=0 temp=-26.6                                                                 pwm=0.000 print_time=0.001 buffer_time=0.000 print_stall=0 extruder: target=0 temp=23.8 pwm=0.000 sysload                                                                =0.79 cputime=-475.849 memavail=315404
```

in the gcode state (or another log line) it would be useful to also have heaters with current temperature

KevinOConnor, 2020-11-09:
As with all  "feature requests", an interested developer would need to implement it and submit it.  That said, the micro-controller shutdown mechanism only currently supports static strings (not dynamic strings) so this is not a simple change.  The log (and/or future temperature updates) usually indicate the faulty sensor.

-Kevin

JustAnother1, 2020-11-22:
Just had the same issue. You can issue a "M105" command. (Octoprint does this anyway) This works in shutdown mode and it shows the read temperatures. So one of the reported temperatures will be out of the limits defined in the printer.cfg file.

closed 2020-12-21 by github-actions[bot]