Klipper3d/klipper #3518: Runtime Error on clicking SD card menu item

dushyantahuja, 2020-11-10:
Hi, I've recently updated klipper and it seems something has gone wrong. Whenever I click on the SD Card menu I get "Unhandled exception during run". At first I thought it might be because of some of my menu customisations (I had changed the menu_root to Main, instead of __main. 

However, when I reverted the menus to default, I got the same error. 

Attaching two log files - one with my customisations, and one without. 
[klippy.log](https://github.com/KevinOConnor/klipper/files/5518615/klippy.log)
[klippy - after reseting the menu items.log](https://github.com/KevinOConnor/klipper/files/5518617/klippy.-.after.reseting.the.menu.items.log)

KevinOConnor, 2020-11-13:
Those log file looks truncated as it seems to be missing important version information.  Without the full log, there's little we can do.

-Kevin

dushyantahuja, 2020-11-14:
Please find attached full klippy.log
[klippy.log](https://github.com/KevinOConnor/klipper/files/5540300/klippy.log)

dushyantahuja, 2020-11-27:
Do you need anything else to check the issue?

KevinOConnor, 2020-12-01:
I'm not sure what's going on.  Is it possible you've got a file in the virtual_sdcard directory with strange characters in the file name?  Perhaps @mcmatrix has an idea?

-Kevin

mcmatrix, 2020-12-02:
Hi, actually I was able to replicate this error with the current master branch code.
Made a file with a name `test_öäüõž.gcode`, having non-Ascii Estonian characters and I was getting the same error.
```
Error loading template 'menu __menu_75a77588:name': UnicodeDecodeError: 'ascii' codec can't decode byte 0xc3 in position 6:
UnicodeDecodeError: 'ascii' codec can't decode byte 0xc3 in position 6: ordinal not in range(128)
```
I'll check how to solve it.

mcmatrix, 2020-12-02:
When using my PR #3262 then it won't throw an error.
The output is 
[image]

Need to investigate a little more why it happens, that part looks the same as in master code.

mcmatrix, 2020-12-02:
@dushyantahuja could you give a try for the menu code in PR #3262 
-Janar

closed 2021-01-06 by github-actions[bot]