Klipper3d/klipper #3519: Installation fails on latest Ubuntu Server 20.04

bepstein111, 2020-11-10:
Upon running install-ubuntu OR install-debian, the following error appears:

> ###### Installing packages...
> Reading package lists... Done
> Building dependency tree       
> Reading state information... Done
> Note, selecting 'python-dev-is-python2' instead of 'python-dev'
> Note, selecting 'libusb-1.0-0' for regex 'libusb-1.0'
> Note, selecting 'libusb-1.0-0-dev' for regex 'libusb-1.0'
> Note, selecting 'libusb-1.0-doc' for regex 'libusb-1.0'
> Package python-virtualenv is not available, but is referred to by another package.
> This may mean that the package is missing, has been obsoleted, or
> is only available from another source
> 
> E: Package 'python-virtualenv' has no installation candidate

Coffee0297, 2020-11-11:
thats because ubuntu server don't support python2 anymore from repos.

bepstein111, 2020-11-11:
> thats because ubuntu server don't support python2 anymore from repos.

Ahhh okay. No plans to update to Python 3?

KevinOConnor, 2020-11-13:
It's certainly possible to install python2 on Ubuntu (and on every major linux distribution).  However, we don't support "custom installations" of Klipper, so you'll need to figure out the steps necessary.  If someone wishes to contribute an update to scripts/install-XXX.sh that would be great.

-Kevin

jpnielsen, 2020-11-24:
Same error on Raspbian bullseye:
 ./klipper/scripts/install-octopi.sh

###### Running apt-get update...
Hit:1 http://raspbian.raspberrypi.org/raspbian bullseye InRelease
Hit:2 http://archive.raspberrypi.org/debian buster InRelease
Reading package lists... Done

###### Installing packages...
Reading package lists... Done
Building dependency tree
Reading state information... Done
Note, selecting 'python-dev-is-python2' instead of 'python-dev'
Note, selecting 'libusb-1.0-0' for regex 'libusb-1.0'
Note, selecting 'libusb-1.0-0-dev' for regex 'libusb-1.0'
Note, selecting 'libusb-1.0-doc' for regex 'libusb-1.0'
Package python-virtualenv is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source

E: Package 'python-virtualenv' has no installation candidate

chron0, 2020-12-14:
Hey, just jumping in quickly here, as I want to give klipper a go as well, but python2 is EOL and although it may still be possible to install python2 on some distros, it will get harder and more uncomfortable very soon. I'm currently playing with a buildroot env for odroid c2 as a host for klipper and I have to chose python2 or python3 (mutually exclusive). Octoprint is python3 rdy, so the question still stands: What are the plans for klipper regarding python3?

closed 2021-01-05 by bepstein111