Klipper3d/klipper #3520: Ender 3 pro not work

Deymos42, 2020-11-10:
Hi, I flash the atmega AVR klipper firmware and default config (with correct usb port) and printer can't connect.

I try to follow steps in config file named printer-creality-ender3pro-2020.cfg to flash with this:

 This file contains pin mappings for the stock 2020 Creality Ender 3
 Pro with the 32-bit Creality 4.2.2 board. To use this config, during
 "make menuconfig" select the STM32F103 with a "28KiB bootloader" and
 with "Use USB for communication" disabled.

if use:
make flash FLASH_DEVICE=/dev/serial/by-id/usb-1a86_USB_Serial-if00-port0 I get that error:

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2016 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/

dfu-util: Invalid DFU suffix signature
dfu-util: A valid DFU suffix will be required in a future dfu-util release!!!
dfu-util: No DFU capable USB device available

Failed to flash to /dev/serial/by-id/usb-1a86_USB_Serial-if00-port0: Error running dfu-util

If the device is already in bootloader mode it can be flashed with the
following command:
  make flash FLASH_DEVICE=1eaf:0003
  OR
  make flash FLASH_DEVICE=1209:beba

If attempting to flash via 3.3V serial, then use:
  make serialflash FLASH_DEVICE=/dev/serial/by-id/usb-1a86_USB_Serial-if00-port0

src/stm32/Makefile:67: recipe for target 'flash' failed
make: *** [flash] Error 255

And if i try to copy .bin file into SD card and put it to ende3 pro nothings happend.

Please help. 
[klippy.log](https://github.com/KevinOConnor/klipper/files/5519123/klippy.log)

Deymos42, 2020-11-10:
Nothing, is already working. i reflash avr firmware and select the correct baud and port.

closed 2020-11-10 by Deymos42

Ducati87, 2021-02-12:
> Nothing, is already working. i reflash avr firmware and select the correct baud and port.

Hello, I have the same problem. How do you fixed it? Thank you.

Deymos42, 2021-02-12:
Reflash avr firmware and select the correct baud and port, the default baud for the avr firware now is 250000, and you have to make sure the processor model is correct before flashing (atmega1284p).  After flashing you just have to select the port tmp/printer and 250000 baudrate in octoprint

Ducati87, 2021-02-12:
> Reflash avr firmware and select the correct baud and port, the default baud for the avr firware now is 250000, and you have to make sure the processor model is correct before flashing (atmega1284p). After flashing you just have to select the port tmp/printer and 250000 baudrate in octoprint

The creality 4.2.2 doesn't have the avr CPU but it has the stm32

Deymos42, 2021-02-12:
Oh, my processor is atmega, I have never tried with an stm, sorry I can't help you.