Klipper3d/klipper #3524: layer shift issues delta printer

goosst, 2020-11-12:
Dear Klippers,

I have a delta printer which seems to experience layer shifts, the weird thing is they only start after hours of printing (like 4-5 hours into the print). This is happening practically in every longer printer now.

Since I find it strange printer-hardware issues would occur so late in the print, I'm wondering if:
1. Something in the log files / firmware of klipper can help me debug this issue?
2. Other klipper users experience similar behaviour on delta printers

[image]
[image]
[image]

[klippy.log](https://github.com/KevinOConnor/klipper/files/5531729/klippy.log)

goosst, 2020-11-15:
or to make it a bit more concrete, is there anything in klipper that helps monitoring any of the temperatures of the (stepper)motor (extruder, actuation, ... ) or controllerboard ?

thanks!

Hywelmartin, 2020-11-15:
https://github.com/KevinOConnor/klipper/blob/bdaca327072598452c14aa6716fa292446cc8086/config/example-extras.cfg#L1179

goosst, 2020-11-15:
I do understand how to hook up external temp sensor, I was mainly wondering if there were temperatures available from internal components like from common controlboards etc.

closed 2020-12-15 by github-actions[bot]