Klipper3d/klipper #3526: Over extrusion

gabba80, 2020-11-13:
[klippy.log](https://github.com/KevinOConnor/klipper/files/5534007/klippy.log)

Hi I am having over extrusion troubles using klipper, using the equivalent settings in marlin works perfectly. Im sure im doing something wrong.

any help would be muchly appreciated

Chris

1N4148, 2020-11-13:
Check your step_distance in the [extruder] section. The value in your config looks like a geared Bondtech/BMG extruder with 415 Steps/mm. Whats your E-Step value in Marlin?

gabba80, 2020-11-13:
Its set to 415 in marlin. Im thinking it could be a slicer issue maybe? I'm using simplify3d.

gabba80, 2020-11-15:
It was my error not klipper. I sent the incorrect uart pin for the extruder driver so was stepping at the wrong rate.

closed 2020-11-15 by gabba80