Klipper3d/klipper #3530: Adding section [mcu rpi] disrupts possibility to connect

cicinovec, 2020-11-14:
Hello, I have issues with adding Raspberry pi MCU to the config file.
I would like to enable measurement with ADXL345 following the manual https://www.klipper3d.org/Measuring_Resonances.html
But adding this section:

[mcu rpi]
serial: /tmp/klipper_host_mcu

causes that I am not able to connect via Octoprint to klipper anymore.

16:31:38 Klipper state: Disconnect
Printer is not ready
The klippy host software is attempting to connect. Please
retry in a few moments.
Klipper state: Not ready
16:31:38 Printer is not ready 

Commenting of this section returns all back to normal.

I am running RPi 4, 4gb, SPI is enabled (checked)
[klippy.log](https://github.com/KevinOConnor/klipper/files/5541093/klippy.log)

BlackStump, 2020-11-14:
[did you follow the instructions to install?](https://github.com/KevinOConnor/klipper/blob/master/docs/RPi_microcontroller.md)

cicinovec, 2020-11-15:
Hi @BlackStump, yes, I did partially. Forgot sudo with second step.

closed 2020-11-15 by cicinovec