Klipper3d/klipper #3532: How to find pin numbers?

sonnius, 2020-11-15:
Hi all,

I can't seem to figure out how to get the pinout values for my printer motherboard, I have:

- Ender3 V2
- 4.2.2 motherboard (stock)

I have used the example config which works, but now I'd like to connect a BL-Touch sensor and I don't know how to find the pin numbers..
The wiring is the same as in this guide: https://smith3d.com/ender-3-v2-bltouch-firmware-installation-guide-by-smith3d-com/

Thanks!

Sineos, 2020-11-15:
https://github.com/MarlinFirmware/Marlin/blob/2.0.x/Marlin/src/pins/stm32f1/pins_CREALITY_V4.h

The Marlin pins file for this board seems pretty comprehensive. BLtouch and servo pins are contained. Maybe this helps

sonnius, 2020-11-16:
Thank you, that solved my issue.

For completeness I include my working bltouch section below:

[bltouch]
sensor_pin = ^PB1
control_pin = PB0
x_offset = -45.5
y_offset = -10.0
z_offset = 2.6
speed = 5.0
sample_retract_dist = 8.0
samples = 2

closed 2020-11-16 by sonnius