Klipper3d/klipper #3534: Thermocoupler Type K + AD8495 + SKR 1.4 Turbo = wrong values

issic, 2020-11-15:
Issue:
Temperature from Adufruit AD8495 amplifier reads 406.7C which is 27.6x higher then actual temperature.

Detail:
According to this [Adufruit equation ](https://www.adafruit.com/product/1778) temperature should be 14.7C, measuring the output voltage (Vo) gives: 1.3235 Volts which equals to 14.7C, but Klipper shows 406.7C. So the thermocouple and the amplifier board works correctly.

Wiring:
AD8495 has 3 wires connected to EXP1 +5V, GND and to pin P1.30 which is Analog To Digital AD0.4 pin according to [this scheme](https://github.com/bigtreetech/BIGTREETECH-SKR-V1.3/blob/master/BTT%20SKR%20V1.4/Hardware/BTT%20SKR%20V1.4-SCH.pdf)

Config:
sensor_type: AD8495
sensor_pin: P1.30
voltage_offset: 0

Variation:
if we apply voltage_offset to -1.25 as that is the calculation in adafruit, the output temperature is 156.3C still 10x bigger then the actual temperature of 14.7

I have 2 AD8495 boards from TriangleLabs and 2 boards from Adufruit, 2 type K thermocouples ... all shows the same behavior.

Possibility:
according to [this ](https://github.com/MarlinFirmware/Marlin/issues/16993#issuecomment-592287975) Marlin issue, it's possible that due the fact that LPC1768/9 has 12Bit Analog to Digital the calculation by klipper is wrong? Here is details for LPC1768 ADC [LPC1768 ADC](https://www.electronicshub.org/how-to-use-adc-in-lpc1768/)

[klippy.log](https://github.com/KevinOConnor/klipper/files/5541960/klippy.log)

issic, 2020-11-15:
Update:

We getting closer to real temp if I set:
adc_voltage: 3.33
voltage_offset: -1.25

But then is confusing is what "The ADC comparison voltage (in Volts). The default is 5 volts." refer's to... 

V(in) is definitely 5V (4.97V) from EXP1

[image]

Hywelmartin, 2020-11-15:
LPC 17xx has 3.3v as logic voltage.. and can't withstand as high as 5v on analog pins ... so try feeding it with 3.3v as a start

issic, 2020-11-15:
Ok, getting closer. with

voltage_offset: 0 it's 0.03 C 

with voltage_offset: 0.08 = 17.1C which is what it suppose to be. OK Case closed? Or it's something weird happening in the calculation anyway. Cos based on the output power of Vo it's correct.

KevinOConnor, 2020-11-15:
@RadicalZephyr - Any thoughts on this?  Where did the AD859x tables currently in klippy/extras/adc_temperature.py originate from?

-Kevin

KevinOConnor, 2020-11-15:
Ah, on closer inspection I see the tables originated from: https://www.analog.com/en/app-notes/an-1087.html

I'm not sure how the table in that doc is supposed to be used though.

-Kevin

KevinOConnor, 2020-11-15:
Okay - looking closer at the tables, it appears one just needs to set the voltage_offset to the Vref value - which is 1.25 for the adafruit board.

-Kevin

issic, 2020-11-15:
For anyone trying this on his SKR1.4Turbo or SKR1.4 

- U can use only P1.30 and P1.31 on the EXP1 and EXP2, these two has direct Analog to Digital others has some kind of pullup resistors. U can use the TH0,TH1 or THB, but you need to remove the 4.7 Ohm resistor from board.

- You must use only 3.3V as logic in LPC176x is 3.3V

in config you must use:
sensor_type: AD8495
sensor_pin: P1.30 or P1.31
adc_voltage: 3.3
voltage_offset: -1.24 (if it's a Adafruit AD8495 board) if it's Triangle lab then I think it's 0

issic, 2020-11-16:
I still have interesting problem, higher the temperature of bed goes UP (blue line) the red line of extruder (which is off) with Thermocouple and AD8495 goes down, until it hits under 10C and the printer stops (even inside the chamber is 40C) That's weird...

[image]

RadicalZephyr, 2020-11-16:
@issic The comment in the code about the default for AD8495 being 5v is just that it is the default value in the code, not that this is what should be used.  You should connect the Vin to 3.3V power line, not 5V power.  Since it is an ADC measurement you should preferably use a noise filtered voltage source, preferably the same one that is connected to the ADC voltage reference on the MCU. In your last comment it sounds like you've connected the right voltage level, but the sources seem a bit questionable.

Also, you should check the polarity of your thermocouple wires. If it's trending slightly downward as the enclosure heats up, it could be that the extruder and thermocouple are heating slightly and you just have the wires connected backwards.  Finally, the voltage offset should not be negative in the klipper config, it is already subtracted from the voltage value, so just put in a positive value of 1.24

RadicalZephyr, 2020-11-17:
Actually, I forgot that the AD8495 has an inductor on the board for suppressing noise on the Vin line, so the 3.3V source you are using should be fine.

issic, 2020-11-17:
@RadicalZephyr 

> it is already subtracted from the voltage value, so just put in a positive value of 1.24

voltage_offset= 1.25
That doesn't work. That gives temperature of 896C.

I still think there is somewhere error in interpreting the voltage from the board by SKR or Klipper.

> Also, you should check the polarity of your thermocouple wires. <

Polarity is right, as if I turn the heater on, the temperature goes the right way up. But when something else (like the bed heating is on) the temperature goes other way.

Ill try to measure input and output voltage, if the error is by fluctuation of voltage to the A8495 board or other hw issue to rule that out. Ill try to feed it directly 3.3V from convertor or from lab source, let see if that helps.

issic, 2020-11-17:
So you can't connect the GND from AD8495 to Analog GND (AGND) as someone suggested. That will throw the whole reading to mayhem. When you touch the nozzle the temperature rapidly fluctuate. I think actually it destroyed one of the thermocouples cos it shows on the output 3V constantly...

Hywelmartin, 2020-12-10:
Isn't this pins also usable 
[image]

closed 2021-01-08 by github-actions[bot]