Klipper3d/klipper #3541: Progress bar not working ender 3 pro

Deymos42, 2020-11-16:
Hello,

 I am new to klipper. I just installed and configured it for my ender 3 pro, and everything works fine, except that while printing the progress bar does not show (allways at 0%)  but the time and the rest do.

I search for a solution but I haven't found anything. 

I don't think the log is helpful in this case but:

[klippy.log](https://github.com/KevinOConnor/klipper/files/5548974/klippy.log)

jakep82, 2020-11-16:
Klipper requires M73 commands to update the progress bar if you're printing through Octoprint.  You can enable M73 commands with an Octoprint plugin.  Some slicers also offer M73 status updates as an option.

https://plugins.octoprint.org/plugins/m73progress/

Deymos42, 2020-11-17:
Thanks! 

Its working now.

Install ocotprint M73 plugin and works.

Thank you for the quick response. Greetings.

closed 2020-11-17 by Deymos42