Klipper3d/klipper #3542: Debugging macro evaluation/execution?

theophile, 2020-11-16:
Is there a way to have Klipper spit out in human readable format the gcode it would process upon execution of a macro? I've got a rather lengthy macro that I use for filament changes that works great, except it appears to result in the part cooling fan being disabled when the print resumes. Here's the macro:

```
[gcode_macro FILAMENT_CHANGE]
gcode:
    M400
    M107; Cut the fan
    G91; Relative Coordinates
    G92 E0; Zero Extruder
    M104 S{TOOLCHANGE_TEMP} ;Toolchange temp
    G1 E-{RETRACT_LENGTH} F10000; Fast Retract 
    G90; Absolute Coordinates
    G1 X249 F9000 ; Move to bucket
    G1 E{RETRACT_LENGTH} F10000; Unretract
    G92 E0; Zero Extruder
    G91; Relative Coordinates
    ;M109 S{(printer["extruder"].target) - (printer["gcode_macro GLOBALS"].cooling_amount)}
    G1 E{(printer["gcode_macro GLOBALS"].ramming_distance)} F{(printer["gcode_macro GLOBALS"].ramming_speed)}
    G1 E-{(printer["gcode_macro GLOBALS"].ramming_distance)} F3000
    G1 E{COOLING_RETRACT_LENGTH} F3000
    G1 E12 F200
    G1 E-12 F200
    G1 E12 F255
    G1 E-12 F255
    ;G1 E12 F300
    ;G1 E-12 F300
    ;G1 E12 F400
    ;G1 E-12 F400
    M400
    M104 S{PURGE_TEMP}
    G1 E{TIP_SHAPE_DISTANCE} F2000 ;move stringy tip into melt zone
    ;G4 P50 ;pause in melt zone
    G1 E-{TIP_SHAPE_DISTANCE} F4000 ;extract clean tip from melt zone
    G1 E-{(printer["gcode_macro GLOBALS"].hotend_length)} F4000
    M400
    SYNC_STEPPER_TO_EXTRUDER STEPPER=my_extra_stepper EXTRUDER=extruder6
    G1 E-{(printer["gcode_macro GLOBALS"].bowden_length) - 26 } F4000
    G90; Absolute Coordinates
    G92 E0; Zero Extruder
    ACTIVATE_EXTRUDER EXTRUDER={NEXT_EXTRUDER}
    SET_STEPPER_ENABLE STEPPER={(printer.toolhead.extruder)} ENABLE=0
    G92 E0; Zero Extruder
    G91; Relative Coordinates
    G1 E25 F200; Medium insert
    G1 E25 F300
    G1 E25 F400
    G1 E{(printer["gcode_macro GLOBALS"].bowden_length) - 75 } F10000
    M400
    SYNC_STEPPER_TO_EXTRUDER STEPPER=my_extra_stepper EXTRUDER={NEXT_EXTRUDER}
    G1 E{(printer["gcode_macro GLOBALS"].hotend_length)} F10000
    M400
    M109 S{PURGE_TEMP}
    M106 S256 ;Comment out this line to make purge pellets
    G1 E30 F150; Feed 10mm
    ;G1 E15 F75; Feed 15mm
    G1 E{INITIAL_PURGE_VOLUME} F70; Initial purge
    M400
    M104 S{NEXT_EXTRUDER_TEMP}
    G1 E{SECONDARY_PURGE_VOLUME} F70; Secondary purge
    M400
    M107 ;Cut the fan
    M109 S{NEXT_EXTRUDER_TEMP}
    M106 S256
    G1 E10 F50; purge 10mm more
    ;G1 E-0.5 F1000; premove retraction
    G4 P5000
    M106 S{ printer.fan.speed * 255}
    G1 X-15 Z1 F10000
    G1 X15 F10000
    G1 X-15 F10000
    G92 E0; Zero Extruder
    G90; Absolute Coordinates
    G1 X{(printer.toolhead.position.x)} Y{(printer.toolhead.position.y)} Z{(printer.toolhead.position.z) + 1} F18000
```

The undefined variables are passed from PrusaSlicer as part of the toolchange gcode insertion. As you can see, I have various M107 commands to disable the part cooling fan at certain points, and M106 commands to turn it on full blast at other points. I use a purge bucket so the idea is to run the fan full speed while purging so that the purged filament comes out in strings rather that pooling up around the nozzle. I also have to cut the fan before M109 commands because I found that the fan messes up the PID tune and causes the printer to take way too long to settle at the target temperature.

Anyway, after all this on-off-on-off for the fan, I want to reset the fan to whatever it was set to before the filament change gcode command. Seven lines from the end of the macro I have this: `M106 S{ printer.fan.speed * 255}`

I understand that the macro is fully evaluated before it is executed, which I _think_ means that when klipper evaluates that M106 command above, it will replace `printer.fan.speed` with whatever that value is at the time the macro is called. But in practice, after the toolchange is complete, the fan runs at either 0% or 1%. 

Is there some way to find out exactly how klipper is evaluating that last M106 command? Or is there some error in my macro logic?

Here are the klippy.log and gcode files for a recent print that showed this issue:
[klippy.zip](https://github.com/KevinOConnor/klipper/files/5549004/klippy.zip)

[Quad_3_board_(SS-project).aw.gcode.zip](https://github.com/KevinOConnor/klipper/files/5549011/Quad_3_board_.SS-project.aw.gcode.zip)

KevinOConnor, 2020-11-19:
I'd add `ECHO` statements to debug the output - for example `ECHO WOULD_RUN="{M106 S{ printer.fan.speed * 255}"`.  The results of the echo should show up in the log and in the g-code terminal window.

Separately, you could simplify your macros a bit by defining `{% set globals = printer["gcode_macro GLOBALS"] %}` at the top of your macros and then referencing variables in that macro with just ` { globals.bowden_length }`.

-Kevin

theophile, 2020-12-10:
Thanks @KevinOConnor , that's a helpful tip about the macro simplification. I ended up figuring out the problem with my fan by watching it from the beginning and monitoring each toolchange. Per the gcode, the fan was supposed to run at 100% the whole time, but after the first toolchange, it ran at 50%. Then after the next, it ran at 25%, then 12%, then 6%, and so on, until it reached 0 and stayed there the rest of the print.

The culprit was my "max_power: 0.5" line in the config. Each time I tried to send a M106 command based on the _current_ speed of the fan, it would cut it in half. I fixed it by changing the corresponding line in the macro to:
`M106 S{ printer.fan.speed * 510}`
Of course, I'll have to remember to change this if I ever use a different max_power value, but it's behaving as expected now!

closed 2021-01-08 by github-actions[bot]