Klipper3d/klipper #3547: Trying to setup printer.cfg...error......No option 'step_pin' in section: 'stepper_x'

yannvautrin, 2020-11-17:
No option 'step_pin' in section: 'stepper_x'
Once the underlying issue is corrected, use the "RESTART"
command to reload the config and restart the host software.
Printer is halted

I am running a skr pro V1.1 + tmc2209 UART, so I don't know why I am asked step_pin...

Here is my printer.cfg file

`########################################
#     Hypercube EVO - Yann VAUTRIN     #
########################################

# 08/11/2020
# Config V1.0
# SKR Pro V1.1
# TMC 2209 V1.2
# Single Extruder

########################################
#          Steppers Settings           #
########################################

########################################
# TMC2209 configuration
########################################

[stepper_x] 
uart_pin: PC13
microsteps: 16
run_current: 0.800
hold_current: 0.500
#stealthchop_threshold: 250

[stepper_y] 
uart_pin: PE3
microsteps: 16
run_current: 0.800
hold_current: 0.500
#stealthchop_threshold: 250

[stepper_z]
uart_pin: PE1
microsteps: 16
run_current: 0.650
hold_current: 0.450
#stealthchop_threshold: 30
position_max: 350

[extruder]
uart_pin: PD4
microsteps: 16
run_current: 0.800
hold_current: 0.500
stealthchop_threshold: 5

[extruder]
step_pin: PE14
dir_pin: PA0
enable_pin: !PC3
step_distance: .002
nozzle_diameter: 0.400
filament_diameter: 1.750
heater_pin: PB1 # Heat0 - Extruder Heater
sensor_pin:  PF4 # T1 Header - Extruder Sensor
sensor_type: NTC 100K beta 3950

################################
#   NTC 100K might be the one  #
################################

#   Type of sensor - common thermistors are "EPCOS 100K B57560G104F",
#   "ATC Semitec 104GT-2", "NTC 100K beta 3950", "Honeywell 100K
#   135-104LAG-J01", "NTC 100K MGB18-104F39050L32",
#   "SliceEngineering 450", and "TDK NTCG104LH104JT1". 
control: pid
pid_Kp: 22.2
pid_Ki: 1.08
pid_Kd: 114
#pid_integral_max:
min_temp: 0
max_temp: 250

[heater_bed]
heater_pin: PD12 # Bed Heater
sensor_pin: PF3 # T0 - Bed Sensor
sensor_type: ATC Semitec 104GT-2
control: watermark
min_temp: 0
max_temp: 130

[fan]
pin: PC8 # Nozzle Fan

[heater_fan fan1]
pin: PE5 #Hotend Fan

[heater_fan fan2]
pin: PE6 # X/Y Carriage Blower Fan

[mcu]
serial: /dev/serial/by-id/usb-Klipper_stm32f407xx_2E00290003504D4E48333620-if00

[printer]
kinematics: corexy
#   This option must be "corexy" for corexy printers.
max_velocity: 300
max_accel: 3000
max_z_velocity: 25
max_z_accel: 30

[bltouch]
sensor_pin: PA1
control_pin: PA2
set_output_mode: 5V #Not sure about this one...
x_offset: -47.0
y_offset: -5.0
z_offset: 1.65
speed: 5.0
samples: 2
sample_retract_dist: 8.0

[safe_z_home]
home_xy_position: 155,155 # Coordinate of bed center
speed: 50
z_hop: 10 # Move up 10mm
z_hop_speed: 5

######################################################################
# "RepRapDiscount 128x64 Full Graphic Smart Controller" type displays
######################################################################

[display]
lcd_type: st7920
cs_pin: EXP1_4
sclk_pin: EXP1_5
sid_pin: EXP1_3
encoder_pins: ^EXP2_3, ^EXP2_5
click_pin: ^!EXP1_2
#kill_pin: ^!EXP2_8

[output_pin beeper]
pin: EXP1_1`

yannvautrin, 2020-11-17:
Sorry for my mistake, here is the log file!
[klippy.log](https://github.com/KevinOConnor/klipper/files/5552097/klippy.log)

Arksine, 2020-11-17:
All steppers need a `step_pin` .  See [here](https://github.com/KevinOConnor/klipper/blob/master/config/generic-bigtreetech-skr-pro.cfg) for an SKR Pro configuration example.

Keep in mind that trinamic driver configuration is separate from the stepper configuration.  For example, your configuration should include both `[tmc2209 stepper_x]` and `[stepper_x]` sections.  This applies to all steppers, including the extruder.

ianc1215, 2020-11-17:
The pins listed in the TMC2209 configuration are supplemental. What the error is tell you is it does not know what pins to use to drive the actual stepper motor. Each motor driver need an ENABLE pin, STEP pin to pulse and DIR pin to let it know which direction it needs to move. All the UART pin does is tells Klipper how to talk to the driver to enable things like Streathchop, VREF adjustments, etc. It has nothing to do with getting the stepper moving.

Here is my config from my Ender 3.

```
[stepper_x]
step_pin: PC2
dir_pin: PB9
enable_pin: !PC3
step_distance: 0.0125220
endstop_pin: ^PA5
position_endstop: 0
position_max: 235
homing_speed: 50

[stepper_y]
step_pin: PB8
dir_pin: PB7
enable_pin: !PC3
step_distance: 0.0123725
endstop_pin: ^PA6
position_endstop: -9
position_min: -9
position_max: 235
homing_speed: 50

[stepper_z]
step_pin: PB6
dir_pin: !PB5
enable_pin: !PC3
step_distance: 0.0025061
endstop_pin: probe:z_virtual_endstop
position_min: -2
position_max: 250
```

yannvautrin, 2020-11-19:
Thanks all for your help, I am almost there now!
One more question, when in the printer.cfg a pin is set as (for example): !PC3, the "!" stands for "Low", right?

theopensourcerer, 2020-11-19:
Yes. Please see the descriptive text in example.cfg:

https://github.com/KevinOConnor/klipper/blob/master/config/example.cfg#L15

closed 2020-12-18 by github-actions[bot]