Klipper3d/klipper #3549: BTT SKR e3Turbo

kinggrave, 2020-11-17:
Hello
I have problems getting the K
[klippy(5).log](https://github.com/KevinOConnor/klipper/files/5555288/klippy.5.log)
lipper to work on the new SKR e3 turbo.
please help me

Hywelmartin, 2020-11-17:
#tx_pin:
#   If using separate receive and transmit lines to communicate with
#   the driver then set uart_pin to the receive pin and tx_pin to the
#   transmit pin. The default is to use uart_pin for both reading and
#   writing.

kinggrave, 2020-11-17:
thank you but how do i do that?

Hywelmartin, 2020-11-17:
Remove the line tx_pin:xxx on all tmc

kinggrave, 2020-11-17:
I'm sorry, that was just an attempt, it doesn't work even without this entry.

kinggrave, 2020-11-17:
#### AUTOCREATED BY KIAUH #####
[include kiauh.cfg]

[stepper_x]
step_pin: P1.4
dir_pin: P1.8
enable_pin: !P1.0
step_distance: .0125
#step_distance: 0.00625
endstop_pin: ^P1.29
position_endstop: 0
position_max: 235
homing_speed: 50

[tmc2209 stepper_x]
uart_pin: P1.1
microsteps: 16
run_current: 0.580
hold_current: 0.500
stealthchop_threshold: 250

[stepper_y]
step_pin: P1.14
dir_pin: P1.15
enable_pin: !P1.9
step_distance: .0125 
#step_distance: 0.00625
endstop_pin: ^P1.28
position_endstop: 0
position_max: 235
homing_speed: 50

[tmc2209 stepper_y]
uart_pin: P1.10
microsteps: 16
run_current: 0.580
hold_current: 0.500
stealthchop_threshold: 250

[stepper_z]
step_pin: P4.29
dir_pin: !P4.28
enable_pin: !P1.16
step_distance: .0025
#step_distance: 0.00125
endstop_pin: ^P1.27
position_endstop: 0.0
position_min: -0.5
position_max: 250

[tmc2209 stepper_z]
uart_pin: P1.17
microsteps: 16
run_current: 0.580
hold_current: 0.500
stealthchop_threshold: 5

[extruder]
step_pin: P2.6
dir_pin: !P2.7
enable_pin: !P0.4
step_distance: .007143
#step_distance: .0035715
nozzle_diameter: 0.400
filament_diameter: 1.750
Pressure_advance: 0.3
#max_extrude_only_accel: 3000
heater_pin: P2.3
sensor_type: EPCOS 100K B57560G104F
sensor_pin: P0.24
control: pid
pid_Kp: 26.772
pid_Ki: 1.803
pid_Kd: 99.390
min_temp: 0
max_temp: 250

[tmc2209 extruder]
uart_pin: P0.5
microsteps: 16
run_current: 0.650
hold_current: 0.500
stealthchop_threshold: 0

[heater_bed]
heater_pin: P2.5
sensor_type: ATC Semitec 104GT-2
sensor_pin: P0.25
control: pid
pid_Kp: 54.027
pid_Ki: 0.770
pid_Kd: 948.182
min_temp: 0
max_temp: 130

[fan]
pin: P2.1

[mcu]
serial: /dev/serial/by-id/usb-Klipper_lpc1769_06100005A49869AFEADE3E5EC22000F5-if00

[printer]
kinematics: cartesian
max_velocity: 300
max_accel: 3000
max_z_velocity: 5
max_z_accel: 100

[board_pins]
aliases:
    # EXP1 header
    EXP1_1=P2.8, EXP1_3=P0.19,   EXP1_5=P0.20, EXP1_7=P0.17, EXP1_9=<GND>,
    EXP1_2=P0.16, EXP1_4=<RST>, EXP1_6=P0.15,  EXP1_8=P0.18, EXP1_10=<5V>

[display]
lcd_type: st7920
cs_pin: EXP1_7
sclk_pin: EXP1_6
sid_pin: EXP1_8
encoder_pins: ^EXP1_5, ^EXP1_3
click_pin: ^!EXP1_2

[virtual_sdcard]
path: ~/sdcard

[pause_resume]
[display_status]

kinggrave, 2020-11-17:
[image]

kinggrave, 2020-11-17:
Are the settings for the flash correct?

jakep82, 2020-11-17:
It looks like this board has a new "low power standby" feature which might require some pins to be set at microcontroller startup in order for the TMC drivers to work.  Someone more knowledgeable than me will need to take a look.

https://github.com/MarlinFirmware/Marlin/pull/19500

jakep82, 2020-11-18:
The list of standby pins is [here](https://github.com/MarlinFirmware/Marlin/pull/19500/commits/d3466e8fb3ca9a5a184fe402cc29bf973f8c8d5e#diff-d8e29201516c1da19b6adbc572ef01ffab0c22310a556405d9bfe0059a4a61dfR153).

It looks like they need to be pulled low during startup from what I'm seeing in the Marlin PR.

kinggrave, 2020-11-18:
Thank you 
How can I control the pins? 
Yes, I think the standby feature will trigger this problem

kinggrave, 2020-11-18:
I did it, I entered the GPIO pins of the standby in the make menuconfig and it works!

kinggrave, 2020-11-18:
#####GPIO pins in Menuconfig !P.3.26, !P3.25, !P1.18, !P1.19   #####

[include kiauh.cfg]

[tmc2209 stepper_x]
uart_pin: P1.1
microsteps: 16
run_current: 0.580
hold_current: 0.500
stealthchop_threshold: 250

[tmc2209 stepper_y]
uart_pin: P1.10
microsteps: 16
run_current: 0.580
hold_current: 0.500
stealthchop_threshold: 250

[tmc2209 stepper_z]
uart_pin: P1.17
microsteps: 16
run_current: 0.580
hold_current: 0.500
stealthchop_threshold: 5

[tmc2209 extruder]
uart_pin: P0.5
microsteps: 16
run_current: 0.650
hold_current: 0.500
stealthchop_threshold: 0

[stepper_x]
step_pin: P1.4
dir_pin: !P1.8
enable_pin: !P1.0
step_distance: .0125
#step_distance: 0.00625
endstop_pin: ^P1.29
position_endstop: 0
position_max: 235
homing_speed: 50

#[tmc2209 stepper_x]
#uart_pin: P1.1
#microsteps: 16
#run_current: 0.580
#hold_current: 0.500
#stealthchop_threshold: 250

[stepper_y]
step_pin: P1.14
dir_pin: !P1.15
enable_pin: !P1.9
step_distance: .0125 
#step_distance: 0.00625
endstop_pin: ^P1.28
position_endstop: 0
position_max: 235
homing_speed: 50

#[tmc2209 stepper_y]
#uart_pin: P1.10
#microsteps: 16
#run_current: 0.580
#hold_current: 0.500
#stealthchop_threshold: 250

[stepper_z]
step_pin: P4.29
dir_pin: P4.28
enable_pin: !P1.16
step_distance: .0025
#step_distance: 0.00125
endstop_pin: ^P1.27
position_endstop: 0.0
position_min: -0.5
position_max: 250

#[tmc2209 stepper_z]
#uart_pin: P1.17
#microsteps: 16
#run_current: 0.580
#hold_current: 0.500
#stealthchop_threshold: 5

[extruder]
step_pin: P2.6
dir_pin: !P2.7
enable_pin: !P0.4
step_distance: .007143
#step_distance: .0035715
nozzle_diameter: 0.400
filament_diameter: 1.750
Pressure_advance: 0.3
#max_extrude_only_accel: 3000
heater_pin: P2.3
sensor_type: EPCOS 100K B57560G104F
sensor_pin: P0.24
control: pid
pid_Kp: 26.772
pid_Ki: 1.803
pid_Kd: 99.390
min_temp: 0
max_temp: 250

#[tmc2209 extruder]
#uart_pin: P0.5
#microsteps: 16
#run_current: 0.650
#hold_current: 0.500
#stealthchop_threshold: 0

[heater_bed]
heater_pin: P2.5
sensor_type: ATC Semitec 104GT-2
sensor_pin: P0.25
control: pid
pid_Kp: 54.027
pid_Ki: 0.770
pid_Kd: 948.182
min_temp: 0
max_temp: 130

[fan]
pin: P2.1

[mcu]
serial: /dev/serial/by-id/usb-Klipper_lpc1769_06100005A49869AFEADE3E5EC22000F5-if00

[printer]
kinematics: cartesian
max_velocity: 300
max_accel: 3000
max_z_velocity: 5
max_z_accel: 100

[static_digital_output buzzer]
pins: !P2.8

[board_pins]
aliases:
    # EXP1 header
    EXP1_1=P2.8, EXP1_3=P0.19,   EXP1_5=P0.20, EXP1_7=P0.17, EXP1_9=<GND>,
    EXP1_2=P0.16, EXP1_4=<RST>, EXP1_6=P0.15,  EXP1_8=P0.18, EXP1_10=<5V>

[display]
lcd_type: st7920
cs_pin: EXP1_7
sclk_pin: EXP1_6
sid_pin: EXP1_8
encoder_pins: ^EXP1_5, ^EXP1_3
click_pin: ^!EXP1_2

[virtual_sdcard]
path: ~/sdcard

[pause_resume]
[display_status]

kinggrave, 2020-11-18:
meine config für das skr e3 turbo !! 

[printer.zip](https://github.com/KevinOConnor/klipper/files/5560850/printer.zip)

okkman, 2020-12-05:
This also fixed the communication fault for me.  To summarize, when building the firmware for the E3 Turbo, the following GPIO pins need to be named in "make menuconfig": 

!P.3.26, !P3.25, !P1.18, !P1.19, !P2.13

KevinOConnor, 2020-12-05:
FYI, it's better to not modify "make menuconfig" and instead add the following to printer.cfg:
```ini
[static_digital_output tmc_standby_pins]
pins: !P.3.26, !P3.25, !P1.18, !P1.19, !P2.13
```

-Kevin

okkman, 2020-12-06:
Thanks Kevin I'll give that a shot.  Can you elaborate at all on why that's a better way?  I am still working to get my head around the klipper ecosystem.

KevinOConnor, 2020-12-07:
Just that it's easier to modify and view printer.cfg.

-Kevin

KevinOConnor, 2021-01-05:
FYI, I added an example config for the e3-turbe (commit f79187d7).

-Kevin

closed 2021-01-13 by github-actions[bot]