Klipper3d/klipper #3550: "PID Calibrate Interrupted" on Octoprint

WarriorClub, 2020-11-17:
I have tried to find the Klippy log to post here but unfortunately I don't see the log from this error for some reason. I have been trying to calibrate PID for temp of 220. I have my parts fan on while doing this. I ran the calibrate command for the hot end but keep getting "PID calibrate interrupted" after running for a few minutes. The issue im having is this worked fine before with the fan. But now even with the fan at 50 percent the calibrate keeps getting "interrupted". Without the blower fan on it calibrates and i get the pid values. 

I will try and post more from the klippy log if i can somehow get it (not sure why it doesn't show). 

In the meantime this issue started happening when i tried a temperature tower which went up to 220 in temp. Could this have somehow damaged the heating sensor and elements? Will post some temp graphs later too.

WarriorClub, 2020-11-18:
Here is the Klippy log that I got right after the "interruption" error.  
[klippy.log](https://github.com/KevinOConnor/klipper/files/5557230/klippy.log)

jakep82, 2020-11-18:
heater_config_name isn't a valid heater name.  You're supposed to change that to the heater you want to modify.  In this case it would be extruder.

`[verify_heater extruder]`

closed 2020-12-18 by github-actions[bot]