Klipper3d/klipper #3554: Failed to home probe: Timeout during homing

n3roGit, 2020-11-19:
From time to time I get the error "Failed to home probe: Timeout during homing" with ABL. I cannot explain this at the moment.

The search has not been successful so far.

Below you find my Config and the Log.

I am grateful for every tip =)

https://pastebin.com/raw/E4CHNuNf

[klippy.zip](https://github.com/KevinOConnor/klipper/files/5569548/klippy.zip)

jakep82, 2020-11-19:
The full error is

```
"BLTouch failed to raise probe")
CommandError: BLTouch failed to raise probe
Failed to home z: Timeout during homing
```

Take a look at the [bltouch document](https://github.com/KevinOConnor/klipper/blob/master/docs/BLTouch.md) to see what configuration options you need to set based on the type of bltouch you're using.  Different versions work a little different, and clones also sometimes require different settings.

n3roGit, 2020-11-20:
I use this one:
https://de.aliexpress.com/item/32949450525.html?spm=a2g0s.9042311.0.0.27424c4dHtxBCH
[image]

I have already read the documentation. Unfortunately I do not know where to start.

If you have a hint with which settings I can try next, feel free to send it to me.

Feche, 2020-11-20:
> I use this one:
> https://de.aliexpress.com/item/32949450525.html?spm=a2g0s.9042311.0.0.27424c4dHtxBCH
> [image]
> 
> I have already read the documentation. Unfortunately I do not know where to start.
> 
> If you have a hint with which settings I can try next, feel free to send it to me.

You should re-read the documentation, specially the BLtouch section that addresses what you can do to use those BLtouch clones.

closed 2020-12-19 by github-actions[bot]