Klipper3d/klipper #3555: Possible way to avoid snipping DIAG pins on 2209 and SKR 1.4 and use endstop switches for homing

Angelol125, 2020-11-19:
It's not really an issue, but I think there may be a way to avoid mutilating the DIAG pins and still use endstop switches for homing when using an SKR 1.4 and TMC2209s. 

Over at [BigTreeTech's Github](https://github.com/bigtreetech/BIGTREETECH-SKR-V1.3/issues/335), I found some info saying that we can use endstop switches for homing if:

1) We invert the corresponding pins (!P1.27, !P1.28, !P1.29) at microcontroller startup when building klipper
2) and connect the endstops like this: NO pin to +5V, NC to GND, C to signal

The info I'm talking about more specifically is this:

> I have been searching everywhere about this.
It seems that DIAG pin pulls low all the time except when a stall is detected. On the other hand, mechanical endstops are usually wired to pull low only when triggered, this wiring opposes what the DIAG pin does.
If you connect the endstop to +V instead of GND, the DIAG pin will be pulling low until the endstop pulls high and the triggered state is detected, even more, if the DIAG pin pulls high marlin should also detect it, so detecting with and endstop and a stall at the same time should be possible. My only concern is if the DIAG pin of the tmc2209 can handle the 5V signal, and if the pulldown of the output could be damaged by the pulse of the endstop. I will keep searching and post here for whoever comes after me

and 

> I just confirmed. DIAG Pin is a Digital output pin. It pulls low until triggered, then outputs high, giving it 5V does nothing to the pin. All Digital pins are 3.3 to 5V capable, and pull up/down resistors are typically 166 Kohm for the tmc2209. So if we give it 5V when pulled low it draws less than 40uA which is under the 2mA stated as current of the low state.
So, yes. You can wire switches that pull to 5V when triggered in parallel to DIAG pins safely and even use both detections at the same time (maybe for example as xmin stallguard and xmax endstop, or delta probing) 

I'm still feeling reluctant to try this since I do not understand completely what happens by connecting the endstop like this and thus I am afraid I could damage the board. I believe though that there are people here more knowledgeable than me able enough to judge if that is gonna work or not.

Angelol125, 2020-11-20:
OK, got an update.

First of all, some info, I have an Ender 3, therefore my endstop switch is like [this ](https://www.cytron.io/image/cache/catalog/products/3D-ENDER3-ES/3D-ENDER3-ES%20(1)-800x800.jpg)

I inverted the P1.29 pin (X-stop pin on my SKR 1.4 Turbo), flashed klipper. I then connected the switch as follows

Switch Board              SKR

C -------------------> S
G (NO) ------------->+5V
V(NC) --------------> G

Now when I query the endstops, I get TRIGGERED when the switch isn't pressed, and Open when I press it. I guess I have to reinvert the pins via my printer.cfg but I think it should work. Let me know of your thoughts and suggestions on this. Thank you in advance.

RicVilla78, 2020-11-25:
Have you tried to bend the pin instead of cutting it. I have done it and it works good.

Angelol125, 2020-11-25:
Have you read the title? The whole point is to not cut the DIAG pin.

KevinOConnor, 2020-11-26:
This isn't a good place for general support questions.  You'll probably get better support from a general 3d printing forum or a bigtreetech forum.

-Kevin

Angelol125, 2020-11-26:
Yeah, I know that. The issue can be closed if you want. I just wrote here so other people could test my findings too. Afaik this works so I thought if more people corroborate this, we could include it in the tmc2209 section, that the pins don't have to get messed with.

closed 2020-11-26 by klipper-gitissuebot

andreas-lenze, 2021-01-29:
@Angelol125  Thanks for clarifying! Just wired my end stops like you suggested, and it works like a charm. SKR 1.4 Turbo, TMC2209 drivers.

Angelol125, 2021-01-29:
Glad you didn't mutilate your drivers!