Klipper3d/klipper #3556: Is there any printing benefit for faster MCU, than the SKR Pro 1.2 ?

dreamcat4, 2020-11-20:
There are now being developed support in marlin firmware for a couple of coretex m7 based mcus. In the range of 400-600mhz. And at a reasonably low cost that makes them affordable. For example the teensy 4.1 module is only about $30.

**Teensy 4.1**

* https://github.com/MarlinFirmware/Marlin/pull/19311
* https://www.cnx-software.com/2020/10/13/teensy-4-1-cortex-m7-board-gets-marlin-firmware-and-openpnp-breakout-board/

**STM32H7**

* https://www.reddit.com/r/BIGTREETECH/comments/j4rl7s/skr_sebx_v10_motherboard_of_bx_32_bits_400mhz/?sort=new
* https://github.com/MarlinFirmware/Marlin/issues/19751

So would it make sense to support these in klipper? In the future?

Hywelmartin, 2020-11-20:
isn't the BTT printer using something like that.. skr se-bx 
https://www.youtube.com/watch?v=YuYOQ6SHBAU

dreamcat4, 2020-11-20:
BIQU and BTT are the same people. That is the exact same thing. It is the controller board in the new BIQU B1 printer.

Hywelmartin, 2020-11-20:
sorry didn't look at the links.. taught you condensed the material in your comment

KevinOConnor, 2020-11-23:
The way that Klipper is architected, there isn't much gain to faster micro-controllers.  A faster micro-controller will provide faster stepping rates (see docs/Benchmarks.md for details), but there isn't much real-world gain once one can reach 100K steps per second for each stepper.  So, for a typical real-world printer, a 70mhz or faster micro-controller should be more than sufficient.

-Kevin

dreamcat4, 2020-11-23:
@KevinOConnor well how about this printer then? Is there perhaps some formula or mathematical way to estimate the maximum stepping rates?

https://www.youtube.com/watch?v=m6DoKoESPdg

aerialviews007, 2020-12-11:
The benefit to newer ARM platform is the inclusion of more interface pins which allows for increased number of integrated stepper motor driver connections and other peripherals.

closed 2021-01-09 by github-actions[bot]