Klipper3d/klipper #3558: RESTART and FIRMWARE_RESTART problem with SKR Mini E3 V2

Feche, 2020-11-20:
If I do a M112 and then FIRMWARE_RESTART, I have to manually powercycle the printer in order to reconnect the printer to Klipper, else a 'MCU shutdown' error is displayed on terminal.

RESTART sometimes work, sometimes doesn't - for example - if I make a configuration change, a RESTART is issued but there is a chance that the printer hangs, and I have to do the manual restart by poweroff cycle.
[klipper.log](https://github.com/KevinOConnor/klipper/files/5574505/klipper.log)

wesk-ms, 2020-11-20:
I'm seeing the exact same behavior with an SKR PRO 1.1 and SKR PRO 1.2.  It also happens with the stock bootloader and the HID bootloader.  It seems like it started happening sometime in July, but I haven't had time to determine which PR caused the behavior change.

grigi, 2020-12-06:
Just an extra datapoint. I have an SKR Mini E3 v2, and I have not noticed any such usb reset issues.

Based on your description it sounds more like your Pi installation has an issue. I had some weird issues on my Raspberry Pi, and the issue got resolved by changing to a different SD card, as it was getting corrupted.

ManIkWeet, 2020-12-29:
I also run into this issue on my SKRV1.3, appears to have nothing to do with M112 specifically.
Simply running FIRMWARE_RESTART will cause the board to never reconnect.

I have to restart my RPI to get it working again.
[klippy.log](https://github.com/KevinOConnor/klipper/files/5751991/klippy.log)

Board is visible when running
`ls /dev/serial/by-id/*`
in terminal

davide-molino, 2021-01-10:
I've the same issue with an SKR Pro 1.2

Klaygar, 2021-01-10:
Have same issue with an SKR GTR.
FIRMWARE_RESTART after config changes will cause the board to never reconnect and need powercycle the printer.

Pi and GTR connected via USB

guythnick, 2021-01-19:
Recently updated Klipper and noticed that my UDEV rule was no longer working, which issues a RESTART command when connected.  I have had to restart the Klipper service for it to be recognized.  
Furthermore, Octoprint now requires me to click "Connect" after issuing a FIRMWARE_RESTART.  Prior, it would automatically reconnect.  I don't know if this is related or not.  Also using SKR Mini 2.0

Sineos, 2021-01-20:
Noticed this as well on a SKR 1.4T. Sometimes the `FIRMWARE_RESTART` will leave the Klipper host service in a "confused" state.
On my side restarting the Klipper service is enough. For my own convenience I added a service-restart command to octoprint.

grigi, 2021-01-20:
So, I had a spout of weird issues in my SKR mini v2, and it turned out to be a bad USB cable.
But I still randomly have a `FIRMWARE_RESTART` causing klipper to crash.
Possibly we need to implement a supervisor pattern to restart klipper and retry a few times before giving up?

mattswest, 2021-01-21:
I too am having issues with the klipper service crashing and needing to be manually restarted, also on a BTT SKR Mini E3 V2.

@KevinOConnor I know you're not a fan of user reports like these, but perhaps with some direction we could help contribute.

davide-molino, 2021-01-31:
Update: searching for a solution I've connected my SKR Pro 1.2 via UART (3 wire) and reflashed the board with updated firmware. The issue is still present, so the FIRMWARE_RESTART doesn't work neither via USB or UART.

davide-molino, 2021-02-10:
I think I've resolved the issue. It seems to be realtive to the stepper drivers (TMC2209 UART). I've reinstalled all the os and started from zero, but the skr won't restart. Without the drivers it works fine. After adding one driver at the time now it works.

misterpeppery, 2021-02-15:
> I think I've resolved the issue. It seems to be realtive to the stepper drivers (TMC2209 UART). I've reinstalled all the os and started from zero, but the skr won't restart. Without the drivers it works fine. After adding one driver at the time now it works.

I'm not sure I follow. Are you rebuilding the os and then updating it repeatedly, adding a new driver each time or is there some sort of modified boot procedure that adds the drivers sequentially during reboot? I'm completely new to klipper (just installed it with fluidd last night and this is my first/only experience with raspberry pi) so the clearer the explanation the better.

churd83, 2021-03-01:
having the same issue with my SKR Pro v1.2. Mine also hangs when i press the reset button on the SKR. I have to powercycle to get it to list a serial in ssh

KevinOConnor, 2021-03-05:
As far as we know, the Klipper code works as intended.  My best guess is that the issue here is due to flaky behaviour in the skr bootloader on some boards.  (Many people use the skr-mini boards and do not have this issue.)

It is always possible there is a defect in Klipper, but if so, it will require a developer with a board that has the problem to track it down and identify the root cause of the issue.

Given our updated policy on github tickets (see https://www.klipper3d.org/Contact.html ) I am closing this ticket.

-Kevin

closed 2021-03-05 by KevinOConnor