Klipper3d/klipper #3560: STM32F407 serial timeout

Zamundaaa, 2020-11-20:
I recently upgraded my x axis stepper controllers to a pair of TB6600 as it's quite heavy and needs the torque, just to notice that they need a custom step pulse duration (which appears to be unusably slow on the arduino(s) I've been using so far). As I had a STM32F407 laying around I added that to the printer, flashed it and tried to connect.

This however doesn't appear to be working. Do I need to address the microcontroller in a different way (as there was something about USB instead of serial in the config that I left enabled)?
[klippy.log](https://github.com/KevinOConnor/klipper/files/5574942/klippy.log)

Arksine, 2020-11-22:
It looks as if you are trying to connect to a ST-Link, that won't work.  My guess is that you are trying to connect to a Discovery board that does not have direct USB access to the STM32F4.

If this is the case you need to configure the Klipper build to use UART for communication, then wire up the correct UART pins to the Pis GPIOs (or a USB-TTL converter like the FT232R).  When that is done  you need to set the correct "serial" option for the mcu.  For example, if you are connecting directly to the Pi GPIOs it could be `/dev/ttyS0` or `/dev/ttyAMA0` depending on your Pi's UART configuration.

Zamundaaa, 2020-11-22:
I am indeed using a discovery board but it has two usb ports: one mini usb for the ST-Link / JTAG and one micro usb port that's directly connected to the STM32F4.

I don't particulary care how it's connected though so I'll try to make it work with UART tomorrow. Thanks!

Zamundaaa, 2020-11-26:
I now tried all the Clock Reference options to make sure I didn't just get it wrong but I always get exactly the same result, serial timeout. No matter if I connect over USB or UART.
Do you have any ideas what could be wrong? The exact board name is STM32F407G-DISC1 and I'm out of ideas.

Zamundaaa, 2020-11-27:
Well, so I didn't manage to make the STM connect but I was able to fix the actual problem I had... The step pin just needed to be inverted for the TB6600 to work properly, no custom pulse duration required.

closed 2020-11-27 by Zamundaaa