Klipper3d/klipper #3563: missing example.cfg and example.-extrascfg

bung69, 2020-11-21:
Hi,  

it looks like your latest push deleted example.cfg and example-extras.cfg :)

Sineos, 2020-11-21:
Moved and restructured: https://github.com/KevinOConnor/klipper/blob/master/docs/Config_Reference.md
Ref: https://github.com/KevinOConnor/klipper/commit/15b5dbaf8ffac8e3889b2cc0ba7a61bce1e9fb6f

tobecwb, 2020-11-22:
Some documents still have references to the old sample files, which ended up breaking the links.
For example, https://www.klipper3d.org/RPi_microcontroller.html, where it references the files sample-raspberry-pi.cfg and sample-multi-mcu.cfg

Almost all documentation that makes any reference to a sample file is broken

KevinOConnor, 2020-11-23:
This was discussed in #3545 .  The links in RPi_microcontroller.md is a separate issue, which I've fixed in commit (c6f870a6).  If there are any other links not working please cite them.

-Kevin

closed 2020-11-28 by klipper-gitissuebot