Klipper3d/klipper #3565: Servo angle must be multiplied by 10

tobecwb, 2020-11-21:
Got a cheap servo (tower 9g), and hooked on a ramps 1.6 (ar6 port)
```
[servo my_servo]
pin: ar6
```

using the code `SET_SERVO SERVO=my_servo ANGLE=0` and `SET_SERVO SERVO=pen_servo ANGLE=180`, the servo barely moves.
if I include a `maximum_servo_angle: 1800`, and execute the command `SET_SERVO SERVO=pen_servo ANGLE=1800` the servo moves 180 degrees (apparently).

then, in config, if I use `initial_angle: 1800`, with `maximum_servo_angle: 1800`, I got a error:
`Recv: // Option 'initial_angle' in section 'servo my_servo' must have maximum of [360.0`]

[klippy.log](https://github.com/KevinOConnor/klipper/files/5578322/klippy.log)

jakep82, 2020-11-21:
You need to adjust the maximum_pulse_width.

https://www.klipper3d.org/Config_Reference.html#servo

tobecwb, 2020-11-21:
tks @jakep82 
`maximum_pulse_width: 0.004` resolved the problem

closed 2020-11-21 by tobecwb