Klipper3d/klipper #3567: MAX31856 Cold Junction High Fault

NBouquain, 2020-11-22:
Hi, I've recently fitted my printer with a K type thermocouple and an MAX31856 breakout board from Adafruit, the whole thing is connected to a SKR 1.3 board on the hardware SPI bus ssp0 (available on the EXP2 connector).
Unfortunately, this does not seems to work as Klipper report a Cold Junction High Fault and enter emergency stop, even though the reported temperature before the stop was correct.
Here is the corresponding log file:
[klippy.log](https://github.com/KevinOConnor/klipper/files/5579990/klippy.log)

The problem has been reproduced when using my old MKS GEN L board, this time using SW SPI.
When using the MAX31856 with the Arduino framework on the same MKS board (same MAX31856 breakout board, TC, cables and pins) and sending the same command as Klipper, the problem could not be reproduced.
This rules out a possible MAX31856 defect and an issue as the board/SPI implementation. Patching the MCU code to ignore the fault leads to a working setup and proper extruder temperature reading (at idle) but is not a proper solution in the long run.

I have seen that similar problems (#2326, same issue but with a Duet board) were previously reported but closed without solution and indeed at high level the Klipper implementation seems correct (plus MAX31855 breakout board are cheap and work).

However the issue may be related to the way that Klipper configure the sensor.
Here is the trace of the MAX31856 initialization performed with the Arduino framework:
[image]

And the same performed by Klipper (slight difference for the CR1 content due to a configuration discrepancy but irrelevant):
[image]
[image]
It can be seen that the reading of the Fault register (0x0F) report fault 0x20, that is the Cold Junction High Fault.

While both the Arduino code and Klipper configure the CR0 (0x80), CR1 (0x81) and MASK (0x82) registers, the Arduino code perform three consecutive writes while Klipper write perform a continuous write from address 0x80.
This behavior would be OK if the addresses of the CR1 and MASK registers were not send as they will be written in the registers instead, leading the to the following MAX31856 configuration:
* 0x81 in register CR0 (as expected)
* 0x81 in register CR1 (instead of 0x23) => this is dangerous as the thermocouple is set to type E !
* 0x23 in register MASK (instead of 0x03) => no impact since Klipper relies on the Fault register content and not the Fault pin.
* 0x82 in the Cold Junction High Fault Threshold register (no modification was expected)
* 0x03 in the Cold Junction Low Fault Threshold register (no modification was expected)

Since the Cold Junction fault threshold registers are spuriously overwritten, it may explain why the Fault condition is triggered.

KevinOConnor, 2020-11-23:
I agree that looks like a Klipper bug.  The fix seems to be straight forward:

```diff
--- a/klippy/extras/spi_temperature.py
+++ b/klippy/extras/spi_temperature.py
@@ -174,12 +174,10 @@ class MAX31856(SensorBase):
             "16" : MAX31856_CR1_AVGSEL16
         }
         value |= config.getchoice('tc_averaging_count', averages, "1")
-        cmds.append(0x80 + MAX31856_CR1_REG)
         cmds.append(value)
 
         value = (MAX31856_MASK_VOLTAGE_UNDER_OVER_FAULT |
                  MAX31856_MASK_THERMOCOUPLE_OPEN_FAULT)
-        cmds.append(0x80 + MAX31856_MASK_REG)
         cmds.append(value)
         return cmds
 
```

However, I don't have the hardware to test this.  Would you be able to delete those two lines from klippy/extras/spi_temperature.py and verify it works correctly?  (Be sure to do a full `sudo service klipper restart` after any code changes.)

Thanks,
-Kevin

NBouquain, 2020-11-23:
Thanks Kevin, it works !
Here is the new log, I have purposely send a M112 to dump the transfer queue.
[klippy.log](https://github.com/KevinOConnor/klipper/files/5581187/klippy.log)

And here is the logic analyzer traces for both the init sequence and a temperature read sequence:
[image]
[image]

This time the CR0, CR1 and MASK registers are properly configured, and there is no more fault.

KevinOConnor, 2020-11-23:
Great.  Thanks for reporting.  This should be fixed now (commit 665ec5e9).

-Kevin

NBouquain, 2020-11-23:
Closing as the issue has been solved with commit 665ec5e 
Thanks again Kevin that was fast 👍

closed 2020-11-23 by NBouquain